linrunner/TLP #287: Set to AC mode on suspend event?

SuperFluffy, 2017-08-01:
When a suspend event is triggered on BAT mode (e.g. when the lid is closed), is there a way to have TLP switch to AC mode so that all powersaving features are disabled prior to sleep?

I am trying to rule out some quirks with my Alienware 13R3 laptop (sometimes the screen is not working after resume from suspend), and I am imagining these only occuring when the laptop was on BAT mode (i.e. when various powersaving options were set).

I might be completely wrong, but I would like to see if with having TLP go back to AC mode I still get that behaviour.

linrunner, 2017-08-02:
TLP does it already by default (not configurable) 
--> https://github.com/linrunner/TLP/blob/master/tlp.in#L247

    apply_common_settings 0

means AC.

khfeng, 2017-08-15:
Normally, device driver will bring device back to full functional state (i.e. AC mode) before system suspend.
So I think this is more likely a kernel issue.

SuperFluffy, 2017-08-15:
Since the wake-from-suspend issue happened on AC as well now (screen stays black), I suspect I can rule out TLPs powersaving triggers as well.

I am going to close this issue, but if anyone has pointers as to how I could triangulate this, those would be welcome.

closed 2017-08-15 by SuperFluffy