linrunner/TLP #288: TLP not charging with thinkpad X230 and docking connector

arseto, 2017-08-15:
### Symptom data

1. The problem occurs on AC mode, I am using thinkpad X230 with docking connector. (Yesterday was working fine with docking too)
2. TLP output [here](https://gist.github.com/arseto/db2e0c8d6101b5ece3f339daa429f3fc)

### Expected behavior
The charging threshold is set at  80%, so the battery should charge when the percentage drops to 80%.

### Actual behavior
But the battery continues to drop all the way to 40%, and still dropping, but the tlp stat says 'Charging'.

### Steps to reproduce the problem
1. set the threshold to 80 (charge) and 100 (stop) in `/etc/default/tlp`
2. booted the system while the battery charge is at 99%
3. the battery level continues to drop below 80% (even with charging threshold set to 80)

arseto, 2017-08-15:
To make it charge again (a workaround):
1. undock the laptop
2. connect the charger directly
3. ensure the battery is charging with tlp stat
4. disconnect the charger
5. dock the laptop, and connect the charger to the dock
6. it's charging normally as it should

Do I need specialized configuration for thinkpad in general? (or maybe specifically with docking connector - I use ultrabase series 3).

Thanks!

linrunner, 2017-08-15:
Your configuration is correct and the output shows that the thresholds are set: 

    tpacpi-bat.BAT0.startThreshold = 80 [%]
    tpacpi-bat.BAT0.stopThreshold = 100 [%]

I see no issue with TLP here, just an odd behaviour of your ThinkPad hardware. Please seek advice in a hardware or ThinkPad oriented internet forum.

closed 2017-08-15 by linrunner