linrunner/TLP #290: Failed built of ACPI_CALL kernel module on kernel 4.12.5

SKUndef, 2017-08-15:
### Symptom data

1. Does the problem occur on battery or AC or both?
    Both
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
    https://gist.github.com/SKUndef/f678e83484c8f1427934a11ab6291df6

### Expected behavior
ACPI_CALL kernel module working

### Actual behavior
Error on building ACPI_CALL kernel module

### Steps to reproduce the problem
Install kernel 4.12.5 on Fedora 26, and run `akmods --force`

velichkov, 2017-08-19:
I had the same problem and the fix is quite simple mkottman/acpi_call#67

linrunner, 2017-08-20:
The maintainer provided a patched **akmod-acpi_call** package in the **tlp-updates-testing** repo. Please test and report.

velichkov, 2017-08-21:
Hi @linrunner,

The `akmod-acpi_call-1.1.1-1.fc26.x86_64.rpm` from `tlp-updates` repository compiles successfully on fc26. The only problem is that it is signed with a different key so to install it I had to disable the gpg check

```
$sudo dnf reinstall akmod-acpi_call
Last metadata expiration check: 0:13:57 ago on Mon 21 Aug 2017 04:48:48 PM EEST.
Dependencies resolved.
======================================================================================================================================================
 Package                                Arch                          Version                                Repository                          Size
======================================================================================================================================================
Reinstalling:
 akmod-acpi_call                        x86_64                        1.1.1-1.fc26                           tlp-updates                         29 k

Transaction Summary
======================================================================================================================================================

Total download size: 29 k
Is this ok [y/N]: y
Downloading Packages:
akmod-acpi_call-1.1.1-1.fc26.x86_64.rpm                                                                               191 kB/s |  29 kB     00:00    
------------------------------------------------------------------------------------------------------------------------------------------------------
Total                                                                                                                 114 kB/s |  29 kB     00:00     
warning: /var/cache/dnf/tlp-updates-f3f475529bdb508f/packages/akmod-acpi_call-1.1.1-1.fc26.x86_64.rpm: Header V4 RSA/SHA256 Signature, key ID 1482d93f: NOKEY
Importing GPG key 0xC2FB431C:
 Userid     : "Andreas Roederer (tlp Repository) <tlp@warpnine.de>"
 Fingerprint: 8AD5 9D64 3341 C382 1333 3FBD C89A 1C1A C2FB 431C
 From       : /etc/pki/rpm-gpg/RPM-GPG-KEY-tlp
Is this ok [y/N]: y
Key imported successfully
Import of key(s) didn't help, wrong key(s)?

Public key for akmod-acpi_call-1.1.1-1.fc26.x86_64.rpm is not installedFailing package is: akmod-acpi_call-1.1.1-1.fc26.x86_64
 GPG Keys are configured as: file:///etc/pki/rpm-gpg/RPM-GPG-KEY-tlp

The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: GPG check FAILED
```

```
$rpm -qpi /var/cache/dnf/tlp-updates-f3f475529bdb508f/packages/akmod-acpi_call-1.1.1-1.fc26.x86_64.rpm
warning: /var/cache/dnf/tlp-updates-f3f475529bdb508f/packages/akmod-acpi_call-1.1.1-1.fc26.x86_64.rpm: Header V4 RSA/SHA256 Signature, key ID 1482d93f: NOKEY
Name        : akmod-acpi_call
Version     : 1.1.1
Release     : 1.fc26
Architecture: x86_64
Install Date: (not installed)
Group       : System Environment/Kernel
Size        : 27346
License     : GPL
Signature   : RSA/SHA256, Sun 20 Aug 2017 01:07:22 AM EEST, Key ID 097f64451482d93f
Source RPM  : acpi_call-kmod-1.1.1-1.fc26.src.rpm
Build Date  : Sun 20 Aug 2017 01:07:21 AM EEST
Build Host  : f26dev64.aritgw.de
Relocations : (not relocatable)
URL         : https://github.com/mkottman/acpi_call
Summary     : Akmod package for acpi_call kernel module(s)
Description :
This package provides the akmod package for the acpi_call kernel modules.
```

linrunner, 2017-08-22:
Right. See https://github.com/linrunner/TLP/issues/210#issuecomment-324076054

closed 2017-08-22 by linrunner

apeyser, 2017-09-07:
This is also needed for fc25 -- and I couldn't seem to find the src rpm for the akmod.