linrunner/TLP #291: Elaborate in README what TLP stands for

Hi-Angel, 2017-08-15:
I am sorry for bringing that simple issue, but Jesus Christ, I'm occasionally configuring and checking different things in TLP for a few days on 2 notebooks, and every time I want to look up docs, tutorials, content of package, I can't remember the name. Was it TPL? Or LPT? Or PTL? The number is the 3! arrangements, and it doesn't help that some of them have meaning on their own, e.g. LPT.

Knowing what TLP stands for would really help!

linrunner, 2017-08-16:
I think you should see a doctor for your memory problem ;-).

closed 2017-08-16 by linrunner

Hi-Angel, 2017-08-16:
@linrunner very funny. Believe me, I have no smallest problem with my memory, but I definitely don't want to remember every non-sense abbreviation I see on the Internet. The name should have some meaning.

I'm sure the project will be very popular with such a name.

Hi-Angel, 2017-08-16:
@linrunner as your "won't fix" and the short reply are suggesting, the abbreviation doesn't actually mean anything?

Bert-de-Jong, 2018-08-09:
So, linrunner, is the abbreviation meaningless or not? :expressionless:
I guess the L might stand for Linux and the P for Power(management/er)?

WinEunuuchs2Unix, 2018-08-14:
Best I've deciphered in last 5 minutes is "T" stands for "Thinkpad", "L" stands for "Laptop" and "P" stands for "Power management".

zelmor, 2018-08-16:
https://linrunner.de/en/tlp/tlp.html

Truth be told, three letter abbreviations are not for everyone, but OP could simply set a symlink in /bin to point to tlp and name it whatever he wants to.

LinuxOnTheDesktop, 2019-05-04:
&nbsp;&nbsp;&nbsp;&nbsp;I am reluctant to say it, but I too do not know what 'TLP' stands for. That is the case even after I have read this page, and report #318, and the page linked above (viz., https://linrunner.de/en/tlp/tlp.html). Like one of the other contributors to this report (#291), my best _guess_ is: 

>  "T" stands for "Thinkpad", "L" stands for "Laptop" and "P" stands for "Power management".

&nbsp;&nbsp;&nbsp;&nbsp;Still, I can remember the acronym (if it _is_ an acronym) 'TLP' . . (Also, it's good software.)

linrunner, 2021-05-03:
For those who are still curious: https://linrunner.de/tlp/faq/misc.html#what-does-tlp-stand-for