linrunner/TLP #295: Flickering Display - Solved

BrendonIrwan, 2017-09-07:
### Symptom data

1. Does the problem occur on battery or AC or both? _Battery_
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.

### Expected behavior
Screen not to flicker

### Actual behavior
Screen flickers at an inconsistent rate, though not that sporadic.

### Steps to reproduce the problem
RADEON configs does not solve the issue (no Radeon hardware installed here). Set:
`PCIE_ASPM_ON_BAT=performance`

Remarks: might want to add this to FAQ if similar problem occurs elsewhere. Also note the development build of my distro.

linrunner, 2017-09-09:
Hi, thanks for your hint. I'll add a hint to the FAQ. 

Btw: why did you think a setting for AMD Radeon will help for Nvidia Optimus? ;-)

BrendonIrwan, 2017-09-11:
At first I didn't know that the tweak was for Radeon hardware. But at least I tried and reported the result
=D

BrendonIrwan, 2017-11-21:
This problem disappears in the release version of Ubuntu 17.10 with BIOS updates. Reverted setting:
`PCIE_ASPM_ON_BAT=powersave`

linrunner, 2018-01-25:
FAQ changed.

closed 2018-01-25 by linrunner