linrunner/TLP #296: X220 does not wake from sleep after being removed from dock

thatpixguy, 2017-09-10:
### Symptom data

I've only recently started using TLP though I've been using this X220 for many years. I installed TLP at the same time as installing a replacement battery and hoping to extend its life (so there exists the possibility that the new battery is the cause).

When I remove my charged and sleeping X220 from my dock, it behaves as if the battery has been removed and won't wake from sleep or even power up until AC is applied.

### Steps to reproduce the problem

1. X220 is running on battery
2. I put it to sleep by closing the lid (half-moon sleep light is on)
3. I put the sleeping laptop in the dock to charge overnight (half-moon light stays on, battery light is blinking as it charges)
4. In the morning, just the half-moon light is on but no battery light as it is fully charged
5. When I remove the still-sleeping laptop from the dock, the half-moon light immediately turns off
6. Opening the lid does not wake the laptop, and even pressing the power button has no effect
7. Laptop only turns on when AC is connected

As an experiment, I have also tried:
5. Open lid on laptop while still in dock, waking it up
6. Remove running laptop from dock at which point it loses power

In both cases the laptop is acting as if the battery is dead/disconnected but upon booting the battery is charged (as can be seen in the attached gist taken shortly after booting).

https://gist.github.com/thatpixguy/aca657e01f1174d1013fc437d840a980

### Expected behavior

Expected behaviour in the first case would be that the laptop would remain asleep while being removed from power, and would wake when the lid is opened.

### Actual behavior

Laptop appears to lose power when being removed from the dock, and stops sleeping (loses sleep state). Laptop will not power up until AC is applied.

### Pending research

Since this takes a full night to test, there are some experiments I haven't had time to try yet, such as testing if this also happens when connected with the AC adaptor overnight rather than being placed in the dock.

thatpixguy, 2017-09-10:
I just put my running laptop in thedock to mke another gist of tlp-stat in case it mentioned anything useful about the dock, and the laptop hut off when I plugged it in.

Here's the gist when it's plugged in (after I rebooted): https://gist.github.com/thatpixguy/122b404655b6bb2d76f67f9665983884

thatpixguy, 2017-09-11:
I have tried leaving it connected to the normal AC adaptor over night and then pulling the power in the morning to see if it has the same effect (acting like the battery has been removed) but it hasn't happened. So it seems to be something specific about how it behaves when in the dock.

linrunner, 2017-09-25:
Hi, looks like a problem with your hardware. I own a X220 with Ultrabase for a long time too, never observed such symptoms.

Please uninstall TLP, poweroff and restart your system. Then check if the symptoms persist.

linrunner, 2017-11-23:
No response --> closed.

closed 2017-11-23 by linrunner