linrunner/TLP #297: x86_energy_perf_policy values have changed

jojo4u, 2017-09-16:
x86_energy_perf_policy has been changed in May 2017 for linux-tools 4.13.
https://github.com/torvalds/linux/commits/master/tools/power/x86/x86_energy_perf_policy

old version reports as: x86_energy_perf_policy Nov 24, 2010
new version reports as: x86_energy_perf_policy 17.05.11 (C) Len Brown <len.brown@intel.com>\n

Old values are: 'performance' | 'normal' | 'powersave'
New values are: "normal" | "performance" | "balance-performance" | "balance-power"| "power"

Most portable way is probably to allow all text values and map them to a number - these have not changed.

Old constants from [x86_energy_perf_policy.c](https://github.com/torvalds/linux/blob/84764a415c707b43e751deb579a421776f190a95/tools/power/x86/x86_energy_perf_policy/x86_energy_perf_policy.c)
#define BIAS_PERFORMANCE		0
#define BIAS_BALANCE			6
#define BIAS_POWERSAVE			15

New constants are in [msr-index.h:](https://github.com/torvalds/linux/blob/b1b6f83ac938d176742c85757960dec2cf10e468/arch/x86/include/asm/msr-index.h)
#define ENERGY_PERF_BIAS_PERFORMANCE		0
#define ENERGY_PERF_BIAS_BALANCE_PERFORMANCE	4
#define ENERGY_PERF_BIAS_NORMAL			6
#define ENERGY_PERF_BIAS_BALANCE_POWERSAVE	8
#define ENERGY_PERF_BIAS_POWERSAVE		15

linrunner, 2017-09-25:
Hi, thanks for your hint.

andreyv, 2017-10-12:
Printing of values in `tlp-stat` also seems to be broken with `x86_energy_perf_policy` version 4.13:
```
x86_energy_perf_policy.cpu0                            = EPB
x86_energy_perf_policy.cpu1                            = EPB
x86_energy_perf_policy.cpu2                            = EPB
x86_energy_perf_policy.cpu3                            = EPB
```

linrunner, 2017-11-29:
Done. Could you do some testing please?

Thanks again for your helpful insights.

jojo4u, 2017-12-17:
I tested TLP 1.0.900 and x86_energy_perf_policy 17.05.11

All text values configured in tlp conf file corresponded to correct number as printed by x86_energy_perf_policy. Thanks!

Tested values:
```
* performance --> 0
* balance-performance --> 4
* default|normal --> 6
* balance-power --> 8
* power|powersave --> 15

```

linrunner, 2017-12-19:
Thanks for testing :-).

linrunner, 2018-01-24:
Released.

closed 2018-01-24 by linrunner