linrunner/TLP #301: DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE="bluetooth" not working as expected

aiguofer, 2017-09-28:
### Symptom data

When I'm using a BT device for sound output, BT still turns off when I unplug AC even though I have `DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE="bluetooth"`

1. Does the problem occur on battery or AC or both?

On battery

2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.

> --- TLP 1.0 --------------------------------------------
> 
> +++ Configured Settings: /etc/default/tlp
> TLP_ENABLE=1
> TLP_DEFAULT_MODE=AC
> DISK_IDLE_SECS_ON_AC=0
> DISK_IDLE_SECS_ON_BAT=2
> MAX_LOST_WORK_SECS_ON_AC=15
> MAX_LOST_WORK_SECS_ON_BAT=60
> SCHED_POWERSAVE_ON_AC=0
> SCHED_POWERSAVE_ON_BAT=1
> NMI_WATCHDOG=0
> ENERGY_PERF_POLICY_ON_AC=performance
> ENERGY_PERF_POLICY_ON_BAT=powersave
> DISK_DEVICES="sda"
> DISK_APM_LEVEL_ON_AC="254"
> DISK_APM_LEVEL_ON_BAT="128"
> SATA_LINKPWR_ON_AC=max_performance
> SATA_LINKPWR_ON_BAT=min_power
> AHCI_RUNTIME_PM_TIMEOUT=15
> PCIE_ASPM_ON_AC=performance
> PCIE_ASPM_ON_BAT=powersave
> RADEON_POWER_PROFILE_ON_AC=high
> RADEON_POWER_PROFILE_ON_BAT=low
> RADEON_DPM_STATE_ON_AC=performance
> RADEON_DPM_STATE_ON_BAT=battery
> RADEON_DPM_PERF_LEVEL_ON_AC=auto
> RADEON_DPM_PERF_LEVEL_ON_BAT=auto
> WIFI_PWR_ON_AC=off
> WIFI_PWR_ON_BAT=on
> WOL_DISABLE=Y
> SOUND_POWER_SAVE_ON_AC=0
> SOUND_POWER_SAVE_ON_BAT=1
> SOUND_POWER_SAVE_CONTROLLER=Y
> BAY_POWEROFF_ON_BAT=0
> BAY_DEVICE="sr0"
> RUNTIME_PM_ON_AC=on
> RUNTIME_PM_ON_BAT=auto
> RUNTIME_PM_ALL=1
> RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
> USB_AUTOSUSPEND=1
> USB_BLACKLIST_WWAN=1
> RESTORE_DEVICE_STATE_ON_STARTUP=0
> DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE="bluetooth"
> START_CHARGE_THRESH_BAT0=85
> STOP_CHARGE_THRESH_BAT0=90
> START_CHARGE_THRESH_BAT1=85
> STOP_CHARGE_THRESH_BAT1=90
> 
> +++ System Info
> System         = LENOVO ThinkPad T460s 20FAS1EF0M
> BIOS           = N1CET58W (1.26 )
> Release        = "Fedora release 26 (Twenty Six)"
> Kernel         = 4.12.13-300.fc26.x86_64 #1 SMP Thu Sep 14 16:00:38 UTC 2017 x86_64
> /proc/cmdline  = BOOT_IMAGE=/vmlinuz-4.12.13-300.fc26.x86_64 root=/dev/mapper/fedora-root ro rd.lvm.lv=fedora/root rd.luks.uuid=luks-75bc5e50-eb56-4d63-abcf-96ddf1bb59d1 rd.lvm.lv=fedora/swap rhgb quiet consoleblank=0 LANG=en_US.UTF-8
> Init system    = systemd
> Boot mode      = UEFI
> 
> +++ TLP Status
> State          = enabled
[... 186 lines cut ...]
> /sys/class/power_supply/BAT1/energy_now                     =  20940 [mWh]
> /sys/class/power_supply/BAT1/power_now                      =  11534 [mW]
> /sys/class/power_supply/BAT1/status                         = Discharging
> 
> tpacpi-bat.BAT1.startThreshold                              =     85 [%]
> tpacpi-bat.BAT1.stopThreshold                               =     90 [%]
> tpacpi-bat.BAT1.forceDischarge                              =      0
> 
> Charge                                                      =   89.1 [%]
> Capacity                                                    =   89.3 [%]
> 
> +++ Charge total                                            =   88.8 [%]
> 

### Expected behavior

When I switch to BAT while BT is in use, BT stays on and connected.

### Actual behavior

When I switch to BAT while BT is in use, BT still turns off.

### Steps to reproduce the problem

Connect BT device while plugged in. Unplug.

linrunner, 2017-10-27:
Your outputs are after the event. Does

    tlp-stat -s -r

show bluetooth "connected" while still on AC?

aiguofer, 2017-12-05:
Hey, sorry I don't know why I never got a notification about your reply.

My output while on AC with my headset connected is:

```
--- TLP 1.0 --------------------------------------------

+++ System Info
System         = LENOVO ThinkPad T460s 20FAS1EF0M
BIOS           = N1CET60W (1.28 )
Release        = "Fedora release 27 (Twenty Seven)"
Kernel         = 4.13.16-300.fc27.x86_64 #1 SMP Mon Nov 27 18:19:43 UTC 2017 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-4.13.16-300.fc27.x86_64 root=/dev/mapper/fedora-root ro rd.lvm.lv=fedora/root rd.luks.uuid=luks-75bc5e50-eb56-4d63-abcf-96ddf1bb59d1 rd.lvm.lv=fedora/swap rhgb quiet consoleblank=0 systemd.log_level=debug
Init system    = systemd
Boot mode      = UEFI

+++ TLP Status
State          = enabled
Last run       = 02:52:53 PM,  11408 sec(s) ago
Mode           = AC
Power source   = AC

+++ Wireless
bluetooth = on
wifi      = on
wwan      = none (no device)

hci0(btusb)                   : bluetooth, not connected
wlp4s0(iwlwifi)               : wifi, connected, power management = off
```

linrunner, 2017-12-05:
Thanks. My BT with mouse attached shows similar symptoms ...

aiguofer, 2017-12-05:
Weird... in case it matters, here's my device:

```
$ lspci | grep -i wire
04:00.0 Network controller: Intel Corporation Wireless 8260 (rev 3a)
```
and here's the initialization from dmesg:

```
$ dmesg | grep -i blue
[   13.741237] SELinux:  Class bluetooth_socket not defined in policy.
[   14.498691] thinkpad_acpi: rfkill switch tpacpi_bluetooth_sw: radio is unblocked
[   15.762105] Bluetooth: Core ver 2.22
[   15.762126] Bluetooth: HCI device and connection manager initialized
[   15.762130] Bluetooth: HCI socket layer initialized
[   15.762132] Bluetooth: L2CAP socket layer initialized
[   15.762141] Bluetooth: SCO socket layer initialized
[   15.871187] Bluetooth: hci0: Bootloader revision 0.0 build 2 week 52 2014
[   15.878163] Bluetooth: hci0: Device revision is 5
[   15.878168] Bluetooth: hci0: Secure boot is enabled
[   15.878172] Bluetooth: hci0: OTP lock is enabled
[   15.878176] Bluetooth: hci0: API lock is enabled
[   15.878180] Bluetooth: hci0: Debug lock is disabled
[   15.878186] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[   15.885209] Bluetooth: hci0: Found device firmware: intel/ibt-11-5.sfi
[   17.657228] Bluetooth: hci0: Waiting for firmware download to complete
[   17.658122] Bluetooth: hci0: Firmware loaded in 1745622 usecs
[   17.658264] Bluetooth: hci0: Waiting for device to boot
[   17.669486] Bluetooth: hci0: Device booted in 11013 usecs
[   17.670675] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-11-5.ddc
[   17.674180] Bluetooth: hci0: Applying Intel DDC parameters completed
```
Also, I've been experiencing the bug from https://bugs.launchpad.net/ubuntu/+source/linux-firmware/+bug/1591167 (although this is a different card, and I'm on Fedora 27). No idea if it's related.

aiguofer, 2017-12-13:
Hmm not sure I'm testing the right way... here's what I did:

```
$ sudo systemctl stop tlp
$ git clone git@github.com:linrunner/TLP.git
$ cd TLP
$ make
$ chmod +x tlp
$ ./tlp start
$  sudo ./tlp stat -s -r
--- TLP 1.0 --------------------------------------------

+++ System Info
System         = LENOVO ThinkPad T460s 20FAS1EF0M
BIOS           = N1CET60W (1.28 )
Release        = "Fedora release 27 (Twenty Seven)"
Kernel         = 4.14.3-300.fc27.x86_64 #1 SMP Mon Dec 4 17:18:27 UTC 2017 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-4.14.3-300.fc27.x86_64 root=/dev/mapper/fedora-root ro rd.lvm.lv=fedora/root rd.luks.uuid=luks-75bc5e50-eb56-4d63-abcf-96ddf1bb59d1 rd.lvm.lv=fedora/swap rhgb quiet consoleblank=0 systemd.log_level=debug LANG=en_US.UTF-8
Init system    = systemd
Boot mode      = UEFI

+++ TLP Status
State          = enabled
Last run       = 02:34:02 PM,      6 sec(s) ago
Mode           = AC
Power source   = AC

+++ Wireless
bluetooth = on
wifi      = on
wwan      = none (no device)

hci0(btusb)                   : bluetooth, not connected
wlp4s0(iwlwifi)               : wifi, connected, power management = off
```

This is with my bluetooth connected.

linrunner, 2017-12-13:
You must build a proper rpm.

aiguofer, 2017-12-13:
You have an docs for that? I don't see a spec file in the repo

linrunner, 2017-12-14:
Sorry, no. But hold on, I'll do some beta packages soon.

linrunner, 2017-12-21:
Packages are [up](http://download.linrunner.de/packages/) now. Go ahead :-).

aiguofer, 2017-12-21:
Awesome! It's correctly reporting when I'm connected, which indeed prevents bluetooth from disconnecting when I unplug AC.

I'll keep this version running and will let you know if I encounter any issues.

linrunner, 2017-12-22:
Thanks for your feedback.

linrunner, 2018-01-24:
Released.

closed 2018-01-24 by linrunner