linrunner/TLP #303: Bluetooth enabled automatically on versions >0.8

gitarra, 2017-10-16:
I have reproduced this issue with two laptops. A Thinkpad T61 with Xubuntu 16.04 and an X230 with Ubuntu 17.10.

I have bluetooth set to disabled but after installing TLP 1.0 bluetooth gets enabled automatically after a reboot or suspend.

On Xubuntu 16.04 I didn't have this issue with version 0.8, it only appeared after I upgraded to 1.0 through your PPA.

On Ubuntu 17.10 I didn't have this issue until I installed TLP 1.0 from your PPA.

gitarra, 2017-10-17:
Correction: On Ubuntu 17.10 bluetooth gets enabled only after a reboot. On Xubuntu 16.04 also after suspend.

linrunner, 2017-10-17:
Please show the full output of

    tlp-stat

via https://gist.github.com/ 

16.04: 
* After boot
* Before suspend (with manually disabled bluetooth) and after resume

17.10:
* After boot

gitarra, 2017-10-18:
I was wrong, bluetooth is only enabled after a reboot on both systems, sorry about the noise.

Here are the outputs:

16.04: https://gist.github.com/gitarra/bebb066b701401c45b997a0320e28495

17.10: https://gist.github.com/gitarra/5114a61676acb7f3b9ef48741e4c31b8

linrunner, 2017-10-21:
Both of your active configurations do not contain the required line

   DEVICES_TO_DISABLE_ON_STARTUP="bluetooth" 

Maybe you didn't keep your original config file when upgrading the package to the PPA version.

gitarra, 2017-10-21:
I am using the default configuration on both systems. I have not done any modifications.

gitarra, 2017-10-21:
Is TLP supposed to enable bluetooth on every boot by default?

Ubuntu without TLP remembers the bluetooth state and keeps it disabled so I do not expect it to be enabled on boot.

Xubuntus blueman applet also has a setting to disable bluetooth on startup which worked fine until upgrading from 0.8 in the official repos to 1.0 in your PPA.

linrunner, 2017-10-21:
Default configuration leaves radio states – including bluetooth – untouched in any version of TLP. No change between 0.8 and 1.0.

linrunner, 2017-10-21:
> Ubuntu without TLP remembers the bluetooth state and keeps it disabled so I do not expect it to be enabled on boot.
Package tlp disables Ubuntu's state remembering by masking systemd-rfkill.service. But this wasn't changed between 0.8 and 1.0. The default with TLP installed has always been (from version 0.2) that bluetooth is not touched by TLP but enabled by the kernel upon system start.

gitarra, 2017-10-27:
Interesting. Doesn't that mean TLP overrides OS settings and effectively enables bluetooth by default? Unless one manually edits the TLP config.

I didn't have this issue with Xubuntu 16.04 when I used 0.8 from the official Ubuntu repo's. I had bluetooth "auto power on" disabled from blueman applet and bluetooth was correctly disabled on boot. That stopped working only after I upgraded to 1.0 from your PPA.

So to understand correctly:

In any case, one must manually edit the tlp config and add that line to disable bluetooth on startup, as otherwise TLP overrides the users OS settings.

linrunner, 2017-10-27:
Just the opposite: "the OS" does the following upon system start
* Kernel enables bluetooth 
* Some userspace tools (blueman) enable or disable bluetooth

TLP with default config does not touch (= not override) bluetooth state set by "the OS". You have to edit config for TLP to change (disable) bluetooth state. Same behaviour applies to any version of TLP (including wifi, wwan).

gitarra, 2017-11-12:
It would be nice if it was stated in the documentation that one must manually disable bluetooth in the config after installing TLP.

As a user I do not expect bluetooth to be enabled by default after installing TLP, as Ubuntu already has bluetooth disabled by default, and TLP is supposed to save battery.

I guess this can be closed.

closed 2017-11-19 by linrunner