linrunner/TLP #305: TLP prevents VMWare from locking memory

nrapopor, 2017-10-25:
### Symptom data

1. Does the problem occur on battery or AC or both?

    * tested only on AC

2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
    * [tlp-stat](https://gist.github.com/nrapopor/6c379f18018e0468c8fa4bab331d4efe "https://gist.github.com/nrapopor/6c379f18018e0468c8fa4bab331d4efe")

### Expected behavior
vmware workstation launches virtual machine

### Actual behavior
vmware refuses to start virtual machine with "can't reserve memory"
see below for relevant log snippet.

### Steps to reproduce the problem
* install vmware 14.0.0 on ubuntu 17.10
* create/or use existing VM and  launch it. (my vm is using 6gig of memory from 64gig available) 
* suspend the vm  
* install/start tlp 
* attempt to restart the VM -- get the error (log snippet below)
* stop tlp service
* attempt to restart the VM -- no error

#### Relevant log snippet 
```sh
2017-10-25T15:51:30.500-04:00| vmx| I125: OvhdMem_PowerOn: initial admission: paged   573856 nonpaged     6975 anonymous    12364
2017-10-25T15:51:30.500-04:00| vmx| I125: VMMEM: Initial Reservation: 2317MB (MainMem=6144MB)
2017-10-25T15:51:30.500-04:00| vmx| I125: MemSched_PowerOn: balloon minGuestSize 13107 (80% of min required size 16384)
2017-10-25T15:51:30.500-04:00| vmx| I125: MemSched: reserved mem (in MB) min 128 max 56336 recommended 56336
2017-10-25T15:51:30.500-04:00| vmx| I125: MemSched: pg 573856 np 6975 anon 12364 mem 1572864
2017-10-25T15:51:30.500-04:00| vmx| I125: MemSched: numvm 1 locked pages: num 0 max 0
2017-10-25T15:51:30.500-04:00| vmx| I125: MemSched: locked Page Limit: host 0 config 13107200 dynam -1
2017-10-25T15:51:30.500-04:00| vmx| I125: MemSched: minmempct 100 minalloc 0 admitted 0
2017-10-25T15:51:30.500-04:00| vmx| I125: Msg_Post: Error
2017-10-25T15:51:30.500-04:00| vmx| I125: [msg.memsched.preNotEnoughMem] Not enough physical memory is available to power on this virtual machine with its configured settings.
2017-10-25T15:51:30.500-04:00| vmx| I125: [msg.memsched.notEnoughMemHostMemVal] To fix this problem, increase the amount of physical memory for all virtual machines to 6252 MB or adjust the additional memory
2017-10-25T15:51:30.500-04:00| vmx| I125: [msg.memsched.hostUnderDuress] It is possible that native applications and/or services have locked down memory which could be preventing the virtual machine from laun
2017-10-25T15:51:30.500-04:00| vmx| I125: [msg.memsched.postNotEnoughMem] If you were able to power on this virtual machine on this host computer in the past, try rebooting the host computer. Rebooting may al
2017-10-25T15:51:30.500-04:00| vmx| I125: ----------------------------------------
2017-10-25T15:51:34.327-04:00| vmx| I125: Module 'MemSched' power on failed.
```

TIA

Nick

linrunner, 2017-10-27:
Despite TLP triggering the problem, it's most likely a power saving setting of some kernel driver that causes VMware to fail.

I suggest your try to isolate the offending setting --> see [Troubleshooting](http://linrunner.de/en/tlp/docs/tlp-troubleshooting.html).

nrapopor, 2017-10-27:
Yes you were right -- it's vmware fault all the way down. I just rebuilt vmmon with the 4.13 kernel awareness and the problem is gone. I've re-enabled TLP and will reopen the issue if I see it is the cause. sorry for the bandwidth ... I was blaming "last change" and doing the root-cause analysis ...

closed 2017-10-27 by nrapopor