linrunner/TLP #306: Dell XPS 13 9360: Error: ThinkPad battery features not available.

Hallelouia, 2017-11-17:
### Symptom data

1. Does the problem occur on battery or AC or both?
Happens on recalibrate on AC.

2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
https://gist.github.com/Hallelouia/cd78256d178751f2ec55183b56f96bc8
### Expected behavior
recalibrate

### Actual behavior
outputs 2 times: Error: ThinkPad battery features not available.

### Steps to reproduce the problem
sudo tlp recalibrate

linrunner, 2017-11-17:
Dear, this is a "ThinkPad only" feature as stated in the docs which implies that it doesn't work on other laptop brands. Please refrain from opening issues for the obvious. Thanks.

closed 2017-11-17 by linrunner

Hallelouia, 2017-11-17:
My bad, Sorry for the inconvenience, I thought the error message was telling me otherwise, that it detected my battery like a thinkpad one and that and that it wasnl't supported.