linrunner/TLP #308: tlp auto becomes uninterruptible

real-or-random, 2017-12-02:
### Symptom data

1. Does the problem occur on battery or AC or both?
Certainly on battery, I'm pretty sure it occurs also on AC (I'll report back)

2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
https://gist.github.com/real-or-random/717d5f0018176b45b2611d3fa7961a23

### Actual behavior
`tlp auto` becomes uninterruptible on boot.
Then, `tlp init start` fires and becomes uninterruptible, too.
This does not happen on every boot but it happens often.

Running another instance of `tlp auto` manually works fine.

I wonder how to debug this. I could try to replace tlp by a wrapper that uses strace...

real-or-random, 2017-12-02:
I don't know if it's related but 2 minutes and 50 seconds after the hanging `tlp auto` was started I get the following in the syslog:
```
Dez 02 10:26:39 tonno systemd-udevd[309]: seq 2490 '/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:00/PNP0C0A:01/power_supply/BAT1' killed
Dez 02 10:26:39 tonno systemd-udevd[309]: worker [412] terminated by signal 9 (KILL)
Dez 02 10:26:39 tonno systemd-udevd[309]: worker [412] failed while handling '/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:00/PNP0C0A:01/power_supply/BAT1'
```

linrunner, 2017-12-02:
Hi,

I suppose by "uninterruptible" you mean even kill -9 cannot end the process.

I would enable [trace mode](http://linrunner.de/en/tlp/docs/tlp-configuration.html#tracemode) in the configuration and comment the line for "tlp auto" in /lib/udev/rules.d/85-tlp.rules (path may differ in Arch) to have only one tlp process.

Atm I see two possible causes:

1. A kernel call resp. write to a sysfile that hangs indefinitely --> kernel problem
2. SELinux throws access violations (AVCs): something like that happened in Fedora [Bug 1408434](https://bugzilla.redhat.com/show_bug.cgi?id=1408434), but I don't know –and doubt – if SELinux is active in Arch

Good luck!

real-or-random, 2017-12-08:
I went the other way and disabled the tlp service, not the udev rule.

Do you see anything suspicious here? The uninterruptible process had pid 575, which does not even exist here.
```
Dez 08 10:44:59 tonno tlp[407]: udev_usb.auto: Bus 1 Dev 1 ID 1d6b:0002 /sys/devices/pci0000:00/0000:00:14.0/usb1
Dez 08 10:44:59 tonno tlp[406]: udev_usb.auto: Bus 2 Dev 1 ID 1d6b:0003 /sys/devices/pci0000:00/0000:00:14.0/usb2
Dez 08 10:44:59 tonno tlp[577]: udev_usb.auto: Bus 4 Dev 1 ID 1d6b:0003 /sys/devices/pci0000:00/0000:00:1d.0/0000:07:00.0/0000:08:02.0/0000:3f:00.0/usb4
Dez 08 10:44:59 tonno tlp[595]: +++ mode=auto (1.0) ++++++++++++++++++++++++++++++++++++++++
Dez 08 10:44:59 tonno tlp[596]: PATH=/usr/local/sbin:/usr/local/bin:/usr/bin
Dez 08 10:44:59 tonno tlp[597]: power_mode=bat
Dez 08 10:44:59 tonno tlp[598]: power_source=bat
Dez 08 10:44:59 tonno tlp[602]: udev_usb.auto: Bus 3 Dev 1 ID 1d6b:0002 /sys/devices/pci0000:00/0000:00:1d.0/0000:07:00.0/0000:08:02.0/0000:3f:00.0/usb3
Dez 08 10:44:59 tonno tlp[606]: lock_tlp_nb.locked
Dez 08 10:44:59 tonno tlp[609]: compare_and_save_power_state(1).different: old=
Dez 08 10:44:59 tonno tlp[614]: set_laptopmode(1): 2
Dez 08 10:44:59 tonno tlp[615]: set_dirty_parms(1): 6000
Dez 08 10:44:59 tonno tlp[626]: set_cpu_hwp_pref(1): balance_power
Dez 08 10:44:59 tonno tlp[629]: set_cpu_perf_pct(1).min.not_configured
Dez 08 10:44:59 tonno tlp[630]: set_cpu_perf_pct(1).max.not_configured
Dez 08 10:44:59 tonno tlp[631]: set_cpu_boost_all(1).not_configured
Dez 08 10:44:59 tonno tlp[637]: set_sched_powersave(1).not_available
Dez 08 10:44:59 tonno tlp[641]: set_nmi_watchdog: 0
Dez 08 10:44:59 tonno tlp[642]: set_phc_controls.not_configured
Dez 08 10:44:59 tonno tlp[662]: set_energy_perf_policy(1): powersave -- unsupported cpu
Dez 08 10:44:59 tonno tlp[681]: +++ mode=auto (1.0) ++++++++++++++++++++++++++++++++++++++++
Dez 08 10:44:59 tonno tlp[683]: PATH=/usr/local/sbin:/usr/local/bin:/usr/bin
Dez 08 10:44:59 tonno tlp[686]: power_mode=bat
Dez 08 10:44:59 tonno tlp[687]: set_disk_apm_level(1): sda [] 128 -- not supported
Dez 08 10:44:59 tonno tlp[690]: power_source=bat
Dez 08 10:44:59 tonno tlp[694]: +++ mode=auto (1.0) ++++++++++++++++++++++++++++++++++++++++
Dez 08 10:44:59 tonno tlp[697]: PATH=/usr/local/sbin:/usr/local/bin:/usr/bin
Dez 08 10:44:59 tonno tlp[696]: set_disk_apm_level(1): sdb [] 128 -- missing
Dez 08 10:44:59 tonno tlp[699]: power_mode=bat
Dez 08 10:44:59 tonno tlp[700]: lock_tlp_nb.failed
Dez 08 10:44:59 tonno systemd-udevd[312]: Process '/usr/bin/tlp auto' failed with exit code 4.
Dez 08 10:44:59 tonno tlp[704]: power_source=bat
Dez 08 10:44:59 tonno tlp[705]: set_disk_apm_level(1): nvme0n [] 128 -- missing
Dez 08 10:44:59 tonno tlp[707]: set_sata_link_power(1).not_available
Dez 08 10:44:59 tonno tlp[709]: set_ahci_runtime_pm(1).not_configured
Dez 08 10:44:59 tonno tlp[710]: lock_tlp_nb.failed
Dez 08 10:44:59 tonno systemd-udevd[313]: Process '/usr/bin/tlp auto' failed with exit code 4.
Dez 08 10:44:59 tonno tlp[711]: set_pcie_aspm(1).disabled_by_kernel
Dez 08 10:44:59 tonno tlp[712]: set_radeon_profile(1).no_card
Dez 08 10:44:59 tonno tlp[717]: set_wifi_power_mode(1, wlp4s0).iw: on; rc=0
Dez 08 10:44:59 tonno tlp[720]: disable_wake_on_lan: enp0s31f6
Dez 08 10:44:59 tonno tlp[723]: set_sound_power_mode(1).hda: 1 controller=Y
Dez 08 10:44:59 tonno tlp[729]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:00.0 [0x060000]
Dez 08 10:44:59 tonno tlp[736]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:02.0 [0x030000]
Dez 08 10:44:59 tonno tlp[741]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:08.0 [0x088000]
Dez 08 10:44:59 tonno tlp[747]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:14.0 [0x0c0330]
Dez 08 10:44:59 tonno tlp[751]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:14.2 [0x118000]
Dez 08 10:44:59 tonno tlp[755]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:16.0 [0x078000]
Dez 08 10:44:59 tonno tlp[759]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1c.0 [0x060400]
Dez 08 10:45:00 tonno tlp[769]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1c.6 [0x060400]
Dez 08 10:45:00 tonno tlp[773]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1d.0 [0x060400]
Dez 08 10:45:00 tonno tlp[777]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1d.2 [0x060400]
Dez 08 10:45:00 tonno tlp[781]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.0 [0x060100]
Dez 08 10:45:00 tonno tlp[786]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.2 [0x058000]
Dez 08 10:45:00 tonno tlp[794]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.3 [0x040300]
Dez 08 10:45:00 tonno tlp[799]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.4 [0x0c0500]
Dez 08 10:45:00 tonno tlp[806]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.6 [0x020000]
Dez 08 10:45:00 tonno tlp[810]: set_runtime_pm(1).black_nvgpu: /sys/bus/pci/devices/0000:02:00.0 [0x030200]
Dez 08 10:45:00 tonno tlp[817]: udev_usb.auto: Bus 1 Dev 5 ID 8087:0a2b /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7 [control]
Dez 08 10:45:00 tonno tlp[818]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:04:00.0 [0x028000]
Dez 08 10:45:00 tonno tlp[825]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:07:00.0 [0x060400]
Dez 08 10:45:00 tonno tlp[829]: udev_usb.auto: Bus 1 Dev 6 ID 04f2:b5ab /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8 [control]
[... cut ...]
Dez 08 10:50:28 tonno tlp[2471]: get_devs(wwan) = 0
Dez 08 10:50:28 tonno tlp[2472]: device_switch(wwan, off).no_change: rc=4
Dez 08 10:50:29 tonno tlp[2493]: +++ mode=auto (1.0) ++++++++++++++++++++++++++++++++++++++++
Dez 08 10:50:29 tonno tlp[2494]: PATH=/usr/local/sbin:/usr/local/bin:/usr/bin
Dez 08 10:50:29 tonno tlp[2495]: power_mode=bat
Dez 08 10:50:29 tonno tlp[2496]: power_source=bat
Dez 08 10:50:29 tonno tlp[2499]: lock_tlp_nb.failed
Dez 08 10:50:29 tonno systemd-udevd[2477]: Process '/usr/bin/tlp auto' failed with exit code 4.
```

linrunner, 2017-12-08:
By your way we've got a indecipherable mess of multiple events in the logs. Please do it as I suggested.

real-or-random, 2017-12-08:
Okay, I see, thanks! I'll report back.

(I used the other way because then the system shutdowns more smoothly. But it still hangs in the very end -- this is the reason why I'm interested in solving this issue by the way.)

real-or-random, 2017-12-09:
Hm, no idea whether this is more useful.
```
Dez 09 01:35:55 tonno systemd[1]: Starting TLP system startup/shutdown...
Dez 09 01:35:55 tonno tlp[1360]: +++ mode=init start (1.0) ++++++++++++++++++++++++++++++++++++++++
Dez 09 01:35:55 tonno tlp[1361]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin
Dez 09 01:35:55 tonno tlp[1362]: power_mode=bat
Dez 09 01:35:55 tonno tlp[1363]: power_source=bat
Dez 09 01:35:55 tonno tlp[1369]: lock_tlp.locked
Dez 09 01:35:55 tonno tlp[1372]: compare_and_save_power_state(1).different: old=
Dez 09 01:35:55 tonno tlp[1376]: set_laptopmode(1): 2
Dez 09 01:35:55 tonno tlp[1377]: set_dirty_parms(1): 6000
Dez 09 01:35:55 tonno tlp[1378]: set_cpu_hwp_pref(1): balance_power
Dez 09 01:35:55 tonno tlp[1379]: set_cpu_perf_pct(1).min.not_configured
Dez 09 01:35:55 tonno tlp[1380]: set_cpu_perf_pct(1).max.not_configured
Dez 09 01:35:55 tonno tlp[1381]: set_cpu_boost_all(1).not_configured
Dez 09 01:35:55 tonno tlp[1382]: set_sched_powersave(1).not_available
Dez 09 01:35:55 tonno tlp[1387]: set_energy_perf_policy(1): powersave -- unsupported cpu
Dez 09 01:35:55 tonno tlp[1394]: set_disk_apm_level(1): sda [] 128 -- not supported
Dez 09 01:35:55 tonno tlp[1398]: set_disk_apm_level(1): sdb [] 128 -- missing
Dez 09 01:35:55 tonno tlp[1403]: set_sata_link_power(1).not_available
Dez 09 01:35:55 tonno tlp[1404]: set_ahci_runtime_pm(1).not_configured
Dez 09 01:35:55 tonno tlp[1405]: set_pcie_aspm(1).disabled_by_kernel
Dez 09 01:35:55 tonno tlp[1406]: set_radeon_profile(1).no_card
Dez 09 01:35:55 tonno tlp[1412]: set_wifi_power_mode(1, wlp4s0).iw: on; rc=0
Dez 09 01:35:55 tonno tlp[1415]: disable_wake_on_lan: enp0s31f6
Dez 09 01:35:55 tonno tlp[1417]: set_sound_power_mode(1).hda: 1 controller=Y
Dez 09 01:35:55 tonno tlp[1425]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:02.0 [0x030000]
Dez 09 01:35:55 tonno tlp[1429]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:08.0 [0x088000]
Dez 09 01:35:55 tonno tlp[1441]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:16.0 [0x078000]
Dez 09 01:35:55 tonno tlp[1445]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1c.0 [0x060400]
Dez 09 01:35:55 tonno tlp[1449]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1c.6 [0x060400]
Dez 09 01:35:55 tonno tlp[1453]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1d.0 [0x060400]
Dez 09 01:35:55 tonno tlp[1457]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1d.2 [0x060400]
Dez 09 01:35:55 tonno tlp[1461]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.0 [0x060100]
Dez 09 01:35:55 tonno tlp[1465]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.2 [0x058000]
Dez 09 01:35:55 tonno tlp[1473]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.4 [0x0c0500]
Dez 09 01:35:55 tonno tlp[1477]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.6 [0x020000]
Dez 09 01:35:55 tonno tlp[1481]: set_runtime_pm(1).black_nvgpu: /sys/bus/pci/devices/0000:02:00.0 [0x030200]
```

linrunner, 2017-12-10:
Execution seems to stall in function [set_runtime_pm()](https://github.com/linrunner/TLP/blob/1.0/tlp-functions.in#L1326) right after the nvidia graphics (which is blacklisted and therefore not touched) at PCI address 02:00.0:

    Dez 09 01:35:55 tonno tlp[1481]: set_runtime_pm(1).black_nvgpu: /sys/bus/pci/devices/0000:02:00.0 [0x030200]

When you compare the list of devices in the tlp-stat output with the trace output, there are 8 more PCIe devices after the nvidia. From the sequence my first suspect would be your wifi:

    /sys/bus/pci/devices/0000:04:00.0/power/control = auto (0x028000, Network controller, iwlwifi) 

I'd try to blacklist it with

    RUNTIME_PM_BLACKLIST="04:00.0"

or disable runtime pm altogether by commenting the setting:

    #RUNTIME_PM_ON_AC=on
    #RUNTIME_PM_ON_BAT=auto

Good luck!

real-or-random, 2017-12-10:
Thanks. Your comment about my tlp stat made me check lspci again:
```
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th Gen Core Processor Gaussian Mixture Model
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 (rev f1)
00:1c.6 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #7 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1d.2 PCI bridge: Intel Corporation Device 9d1a (rev f1)
00:1f.0 ISA bridge: Intel Corporation Device 9d4e (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (4) I219-LM (rev 21)
02:00.0 3D controller: NVIDIA Corporation GM108M [GeForce 940MX] (rev ff)
04:00.0 Network controller: Intel Corporation Wireless 8265 / 8275 (rev 78)
40:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM961/PM961
```

Wtf, a few devices are missing after the NVIDIA card. It turns out that these are related to the  Thunderbolt port. Then I checked system logs again (when it failed) and it turns out that I've overlooked a kernel stack trace there related to thunderbolt. 

Investigating that revealed that the system is probably effected by a BIOS issue:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1708043/ (see there for a similar track trace)
https://forums.lenovo.com/t5/ThinkPad-T400-T500-and-newer-T/ThinkPad-T470s-BIOS-bug-WOL-and-Thunderbolt-3/m-p/3707059

So I updated my BIOS. Let's first try this.

As far as I understand, the Thunderbolt PCI devices are supposed to appear only when Thunderbolt is actually used (and they indeed appear when I use this port as a USB-C port). I've never used the port before in my life but still the devices were apparently up sometimes, which is consistent with the issue in the Lenovo thread and seems to be the case whenever tlp hangs (as far as my logs go back).

real-or-random, 2017-12-19:
It seems to work now with the new BIOS... At least I couldn't observe the issue anymore. Thanks for your help!

closed 2017-12-19 by real-or-random

linrunner, 2017-12-19:
Thanks for your feedback.