linrunner/TLP #309: Disable Bluetooth on System boot

ralpheeee88, 2017-12-04:
### Symptom data
am trying to disable Bluetooth on system boot. I always have to manually switch it off `systemctl stop bluetooth.services`.

[http://ix.io/CM6](url)

1. Does the problem occur on battery or AC or both?
have only tested on power
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.

### Expected behavior

### Actual behavior

### Steps to reproduce the problem

linrunner, 2017-12-05:
Hi,

support is not possible without the required outputs. Your link doesn't work. Use https://gist.github.com/ please.

ralpheeee88, 2017-12-05:
thxs for your help!
https://gist.github.com/ralpheeee88/31cb2fe8ae853f208028fb7efb938e7e

linrunner, 2017-12-05:
I'm still missing the tlp-stat ouput.

ralpheeee88, 2017-12-05:
@linrunner am really sorry...i have amended the file

not sure if i should ask here or in a new case regarding the suggestions / warnings

linrunner, 2017-12-05:
You configured TLP correctly to disable bluetooth upon boot. So probably your desktop environment's bluetooth applet – or some other part of Arch Linux – re-enables  BT after TLP disabled it. I suggest you ask for advice in the Arch Linux forum.

Suggestions/warnings: the issue tracker is for bug reports against TLP only, it is _not_ intended as a support forum or to give general advice.  

Please check the docs ([installation instructions](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#arch), [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#warnings)). You may ask in the Arch Linux forum as well.

closed 2017-12-08 by linrunner