linrunner/TLP #310: Threshold setting not supported on Thinkpad Edge 130

Ottaviocr, 2017-12-06:
### Symptom data

Environment: 
Thinkpad Edge E130, running Linux Mint Debian Edition
Packages installed: tlp tp-smapi-dkms acpi-call-dkms

The problem:
$ sudo tlp-stat -b
--- TLP 0.9 --------------------------------------------
+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (unsupported hardware)
tpacpi-bat = inactive (unsupported hardware)

This happened despite the suggestion coming from "sudo tlp-stat" was to install the acpi-call modules. So either the suggestion was wrong or this Thinkpad module is blacklisted. 

This was the output before installing tp-smapi-dkms acpi-call-dkms:

tp-smapi   = inactive (kernel module 'tp_smapi' not installed)
tpacpi-bat = inactive (kernel module 'acpi_call' not installed)

1. Does the problem occur on battery or AC or both?
Applies only to AC mode (as long as I know)

2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
https://gist.github.com/Ottaviocr/600041a0da98c6dc180becb0b05e6366#file-edgee130-txt

### Expected behavior
To have battery thresholds working

### Actual behavior
$ sudo tlp setcharge
Error: ThinkPad extended battery functions not available.
### Steps to reproduce the problem
Install tlp tp-smapi-dkms acpi-call-dkms on Thinkpad E130 and run "sudo tlp-stat -b"

Ottaviocr, 2017-12-08:
I've created a second gist with more details.

https://gist.github.com/Ottaviocr/d8637650d768037b4f9f7630c0a71a23

linrunner, 2017-12-08:
You don't need to install tp-smapi-dkms. acpi-call-dkms is sufficient but it seems not to work properly on your ThinkPad.

I suggest to update your BIOS. The installed version 2.05 is *way outdated*, newest is [2.58](https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-edge-laptops/thinkpad-edge-e130/downloads/ds0294319).

Ottaviocr, 2017-12-10:
Unfortunately even after updating the bios, the problem persists:
Edited gist:
https://gist.github.com/Ottaviocr/600041a0da98c6dc180becb0b05e6366

One more thing: (see error at the end)

oc@mint ~ $ dmesg|grep -i acpi
[    0.000000] BIOS-e820: [mem 0x00000000dae9f000-0x00000000daf9efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000daf9f000-0x00000000daffefff] ACPI data
[    0.000000] efi:  ACPI=0xdaffe000  ACPI 2.0=0xdaffe014  SMBIOS=0xdae3b000 
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000DAFFE014 000024 (v02 LENOVO)
[    0.000000] ACPI: XSDT 0x00000000DAFFE170 000094 (v01 LENOVO TP-H4    00002580 PTL  00000002)
[    0.000000] ACPI: FACP 0x00000000DAFEC000 00010C (v05 LENOVO TP-H4    00002580 PTL  00000002)
[    0.000000] ACPI: DSDT 0x00000000DAFEE000 00EDE3 (v02 LENOVO IVB-CPT  00000000 INTL 20061109)
[    0.000000] ACPI: FACS 0x00000000DAF77000 000040
[    0.000000] ACPI: SLIC 0x00000000DAFFD000 000176 (v01 LENOVO TP-H4    00002580 PTL  00000001)
[    0.000000] ACPI: ASF! 0x00000000DAFED000 0000A5 (v32 LENOVO TP-H4    00002580 PTL  00000002)
[    0.000000] ACPI: HPET 0x00000000DAFEA000 000038 (v01 LENOVO TP-H4    00002580 PTL  00000002)
[    0.000000] ACPI: APIC 0x00000000DAFE9000 000098 (v01 LENOVO TP-H4    00002580 PTL  00000002)
[    0.000000] ACPI: MCFG 0x00000000DAFE8000 00003C (v01 LENOVO TP-H4    00002580 PTL  00000002)
[    0.000000] ACPI: FPDT 0x00000000DAFE7000 000064 (v01 LENOVO TP-H4    00002580 PTL  00000002)
[    0.000000] ACPI: SSDT 0x00000000DAFE6000 0008CA (v01 PmRef  Cpu0Ist  00003000 INTL 20061109)
[    0.000000] ACPI: SSDT 0x00000000DAFE5000 000A7E (v01 PmRef  CpuPm    00003000 INTL 20061109)
[    0.000000] ACPI: UEFI 0x00000000DAFE4000 00003E (v01 LENOVO TP-H4    00002580 PTL  00000002)
[    0.000000] ACPI: UEFI 0x00000000DAFE3000 000042 (v01 PTL    COMBUF   00000001 PTL  00000001)
[    0.000000] ACPI: MSDM 0x00000000DAFE2000 000055 (v03 LENOVO TP-H4    00002580 PTL  00000002)
[    0.000000] ACPI: UEFI 0x00000000DAFE1000 0002BA (v01 LENOVO TP-H4    00002580 PTL  00000002)
[    0.000000] ACPI: DBG2 0x00000000DAFE0000 00006B (v00 LENOVO TP-H4    00002580 PTL  00000002)
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] ACPI: PM-Timer IO Port: 0x408
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] ACPI: LAPIC (acpi_id[0x01] lapic_id[0x00] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x02] lapic_id[0x01] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x03] lapic_id[0x02] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x04] lapic_id[0x03] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x05] lapic_id[0x00] disabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x06] lapic_id[0x00] disabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x07] lapic_id[0x00] disabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x08] lapic_id[0x00] disabled)
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x00] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x01] high edge lint[0x1])
[    0.000000] ACPI: IOAPIC (id[0x02] address[0xfec00000] gsi_base[0])
[    0.000000] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[    0.000000] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 high level)
[    0.000000] ACPI: IRQ0 used by override.
[    0.000000] ACPI: IRQ2 used by override.
[    0.000000] ACPI: IRQ9 used by override.
[    0.000000] Using ACPI (MADT) for SMP configuration information
[    0.000000] ACPI: HPET id: 0x8086a301 base: 0xfed00000
[    0.000060] ACPI: Core revision 20140424
[    0.016609] ACPI: All ACPI Tables successfully acquired
[    0.167834] PM: Registering ACPI NVS region [mem 0xdae9f000-0xdaf9efff] (1048576 bytes)
[    0.169376] ACPI: bus type PCI registered
[    0.169378] acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
[    0.180548] ACPI: Added _OSI(Module Device)
[    0.180551] ACPI: Added _OSI(Processor Device)
[    0.180553] ACPI: Added _OSI(3.0 _SCP Extensions)
[    0.180554] ACPI: Added _OSI(Processor Aggregator Device)
[    0.184684] ACPI: Executed 1 blocks of module-level executable AML code
[    0.189174] [Firmware Bug]: ACPI: BIOS _OSI(Linux) query ignored
[    0.189762] ACPI: Dynamic OEM Table Load:
[    0.189774] ACPI: SSDT 0xFFFF88011D83E000 00096A (v01 PmRef  Cpu0Cst  00003001 INTL 20061109)
[    0.199025] ACPI: Dynamic OEM Table Load:
[    0.199035] ACPI: SSDT 0xFFFF8801191DA400 000303 (v01 PmRef  ApIst    00003000 INTL 20061109)
[    0.208138] ACPI: Dynamic OEM Table Load:
[    0.208148] ACPI: SSDT 0xFFFF880119282C00 000119 (v01 PmRef  ApCst    00003000 INTL 20061109)
[    0.672409] ACPI: Interpreter enabled
[    0.672417] ACPI Exception: AE_NOT_FOUND, While evaluating Sleep State [\_S1_] (20140424/hwxface-580)
[    0.672424] ACPI Exception: AE_NOT_FOUND, While evaluating Sleep State [\_S2_] (20140424/hwxface-580)
[... 61 lines cut ...]
[    7.655913] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[    7.655915] ACPI Warning: SystemIO range 0x0000000000000530-0x000000000000053F conflicts with OpRegion 0x0000000000000500-0x0000000000000563 (\GPIO) (20140424/utaddress-254)
[    7.655918] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[    7.655919] ACPI Warning: SystemIO range 0x0000000000000500-0x000000000000052F conflicts with OpRegion 0x0000000000000500-0x0000000000000563 (\GPIO) (20140424/utaddress-254)
[... cut ...]
[  343.646831] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[  343.679675] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[  441.922166] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[  447.445198] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[  447.486575] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[ 1746.938976] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[ 1779.356850] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[ 1779.391935] acpi_call: Cannot get handle: Error: AE_NOT_FOUND

Ottaviocr, 2017-12-10:
I'm running around in circles. I'm now running Fedora 27 Workstation live. I've installed  tlp tlp-rdw  akmod-tp_smapi akmod-acpi_call kernel-devel .

Initial tlp-stat:
https://gist.github.com/Ottaviocr/f2a91dda1b59df899e1ce4f779333b90

However modprobe revealed that acpi_call was not installed.

So I copied acpi_call.ko  to /lib/modules/`uname -r`/kernel/:

```
$ sudo cp /lib/modules/4.13.16-302.fc27.x86_64/extra/acpi_call/acpi_call.ko /lib/modules/`uname -r`/kernel/
$ sudo depmod
$ sudo modprobe -v acpi_call
insmod /lib/modules/4.13.9-300.fc27.x86_64/kernel/acpi_call.ko 
modprobe: ERROR: could not insert 'acpi_call': Exec format error
$ sudo tlp-stat -b
--- TLP 1.0 --------------------------------------------

+++ ThinkPad Battery Features
tp-smapi   = inactive (unsupported hardware)
tpacpi-bat = inactive (kernel module 'acpi_call' load error)
```

I'm ready to give up, but I'd rather leave it to @linrunner what to do with this issue.

linrunner, 2017-12-10:
There's no benefit in changing the distro while analyzing a problem. Just adds confusion ... ;-)

I suggest we continue with Debian resp. LMDE. Please show the output of (**as root**):

    /usr/share/tlp-pm/tpacpi-bat -v -g SP 1

Ottaviocr, 2017-12-10:
```
mint mint # /usr/share/tlp-pm/tpacpi-bat -v -g SP 1
Error: AE_NOT_FOUND for ASL base: \_SB.HKEY
Error: AE_NOT_FOUND at /usr/share/tlp-pm/tpacpi-bat line 409.
mint mint # dmesg
[  791.517113] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[  791.550238] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[  791.580137] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[  820.299844] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[  875.106609] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[ 1101.119564] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
mint mint # 
```

Ottaviocr, 2017-12-10:
I have update the battery firmware (on Windows), rebooted to LMDE and performed the above steps, unfortunately with the same negative result.

linrunner, 2017-12-11:
Please show the output of:

    for ps in /sys/class/power_supply/*; do grep -s '.*' $ps/type $ps/online $ps/present $ps/device/path; done

Ottaviocr, 2017-12-11:
```
mint mint # for ps in /sys/class/power_supply/*; do grep -s '.*' $ps/type $ps/online $ps/present $ps/device/path; done
/sys/class/power_supply/ACAD/type:Mains
/sys/class/power_supply/ACAD/online:1
/sys/class/power_supply/ACAD/device/path:\_SB_.ACAD
/sys/class/power_supply/BAT1/type:Battery
/sys/class/power_supply/BAT1/present:1
/sys/class/power_supply/BAT1/device/path:\_SB_.BAT1
```

linrunner, 2017-12-12:
The output doesn't show a ASLBase need for communication with the battery controller, so tpapci-bat cannot autodetect it and won't work.

If you absolutely want to use thresholds you have to research the ASLBase yourself (see [here](https://github.com/teleshoes/tpacpi-bat/issues/3#issuecomment-13996233)) and provide an additional pull request adding a line for your model in the table shown in the first commit [here](https://github.com/linrunner/TLP/pull/307/commits/42323f01f160c8687e6d498df400bceb001ace33). Your model string is "Edge E130" (from tlp-stat output)

Ottaviocr, 2017-12-12:
@linrunner thanks for your effort. I'm planning to rebuild my system from sources (using pkgsrc probably on Centos 7). If I made a pull request, should it be on the upstream tpacpi-bat from @teleshoes or yours? I'm new to Github, so I don't know if the two projects are automatically synced/

linrunner, 2017-12-12:
Upstream please.

Ottaviocr, 2017-12-12:
Thanks. One final question. If/when the pull request is made upsteam. If it is accepted, would I reopen/update this issue or start a new one?

linrunner, 2017-12-12:
Just give an update here.

linrunner, 2017-12-12:
I found an old commit before autodetect was invented: https://github.com/teleshoes/tpacpi-bat/commit/1ef9ad025b72ce3f32b2bb033867a721555ec9b9

Seems that the ASLBase for the E130 is the same as for the ThinkPad 13. I suggest to try the additional line

    'ThinkPad Edge E130' => '\_SB.PCI0.LPCB.EC.HKEY',

Ottaviocr, 2017-12-12:
In my installation tpacpi-bat is not the same as [upstream](https://github.com/teleshoes/tpacpi-bat) : 

```
mint mint # find / -name *tpacpi-bat*
/usr/share/tlp-pm/tpacpi-bat
mint mint # wc -l /usr/share/tlp-pm/tpacpi-bat 
496 /usr/share/tlp-pm/tpacpi-bat
```
So I uploaded the upstream file and added the line, as above:
`'ThinkPad Edge E130' => '\_SB.PCI0.LPCB.EC.HKEY',`

And it works:
```
mint ~ # tlp start
TLP started in AC mode.
mint ~ # dmesg
[  713.555141] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[  713.590749] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[  730.845239] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[  730.878599] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[  730.910967] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
mint ~ # tlp-stat -b
--- TLP 0.9 --------------------------------------------

+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (unsupported hardware)
tpacpi-bat = active

+++ ThinkPad Battery Status: BAT1 (Main / Internal)
/sys/class/power_supply/BAT1/manufacturer                   = SANYO
/sys/class/power_supply/BAT1/model_name                     = 45N1057
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  62160 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  58310 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  34110 [mWh]
/sys/class/power_supply/BAT1/power_now                      =      0 [mW]
/sys/class/power_supply/BAT1/status                         = Unknown (threshold effective)

tpacpi-bat.BAT1.startThreshold                              =     40 [%]
tpacpi-bat.BAT1.stopThreshold                               =     40 [%]
tpacpi-bat.BAT1.forceDischarge                              =      0

Charge                                                      =   58.5 [%]
Capacity                                                    =   93.8 [%]

mint ~ #  tlp setcharge 40 60 BAT1
Setting temporary charge thresholds for BAT1:
  stop  = 60
  start = 40
mint ~ # 
```

Diff:
```
mint tmp # diff tpacpi-bat /usr/share/tlp-pm/tpacpi-bat 
25c25
< # Tested and works on: W520, T420, X220
---
> # Tested and works on: W520, T420, X220, E130
396a397
> 	'ThinkPad Edge E130' => '\_SB.PCI0.LPCB.EC.HKEY',

```

@linrunner  you did it, mate!
[
New patched tpacpi-bat for Thinkpad E130](https://gist.github.com/Ottaviocr/d1c438389001433cec3ffea4bd78423d)

linrunner, 2017-12-13:
Merged.

linrunner, 2018-01-24:
Released.

closed 2018-01-24 by linrunner