linrunner/TLP #312: Thinkpad T470 BAT1 would not discharge

quoderatd2, 2017-12-11:
### Symptom data

<script src="https://gist.github.com/quoderatd2/a6d1ac688d87c3612b097e5e4590975f.js"></script>

### Expected behavior

1) I did `sudo tlp discharge BAT0` and I expected it to discharge.
2) I did `sudo tlp discharge BAT1` and I expected it to discharge.

### Actual behavior
1) BAT0 did discharge, but
2) BAT1 did not discharge. It is stuck at maximum capacity and percentage. 
It records voltage and power, but the capacity is staying the same. 
Even on tlp stat, it says BAT1 is discharging but the capacity stays the same. 
### Steps to reproduce the problem
Need further help

linrunner, 2017-12-11:
1. Please show the terminal output while `tlp discharge BAT1` is running.

2. While `tlp discharge BAT1` is still running, open a second terminal and show the output of (as root)

       tlp-stat -b

smclt30p, 2017-12-18:
On the XX70 series of laptops from Lenovo the firmware has been changed regarding battery discharging. It's now dynamic, meaning that batteries get discharged based on health and other unknown factors. 

In the past the external battery would discharge first, then the internal. On XX70 series, the order is not constant, which means the internal battery coult be discharged before the external one.

People have been complaining, a lot. See:

https://forums.lenovo.com/t5/ThinkPad-T400-T500-and-newer-T/T470-power-bridge-wired-battery-discharging-order-always/m-p/3687964

linrunner, 2017-12-22:
@smclt30p : you're talking about automatic battery selection performed by the BIOS and/or EC firmware while running on battery power.

This issue is about forced discharging – while running on AC power – not working. 

@quoderatd2 : any news?

smclt30p, 2017-12-24:
@linrunner I know, what I'm saying is maybe the EC is refusing to force-discharge because of the selective discharging.

smclt30p, 2018-01-01:
> @smclt30p : you're talking about automatic battery selection performed by the BIOS and/or EC firmware while running on battery power.

Could it be that the EC/Firmware is refusing to discharge the battery
because of exacly that selective dischage on those models?

linrunner, 2018-01-29:
@quoderatd2 : is your problem solved? How?

closed 2018-02-07 by linrunner