linrunner/TLP #314: [Feature request] Except/select some interfaces from triggering/to trigger tlp-rdw disconnect

marmistrz, 2018-01-02:
I enabled tlp-rdw to disable WLAN when I connect Ethernet (interface `enpXsY`) to my laptop. That works great... well, usually.

My SailfishOS smartphone has a developer mode. It means it's running a DHCP server and the phone is available at a predefined IP address, e.g. 192.168.2.15. When connecting the phone via a USB cable, a new interface is created: `enpXsABuC`, where `X, A, B, C` are digits. Then tlp-rdw detects a LAN connect, drops my WLAN and I'm without Internet access, have to reconnect manually.

This means: I'd like to blacklist `enpXsABuC` from triggering tlp-rdw or even make only `enpXsY` trigger do that. A more advanced version: disable/enable only ifaces matching a regex.

The relevant part of my config: 
```
# Radio devices to disable on connect.
DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi"
#DEVICES_TO_DISABLE_ON_WIFI_CONNECT="wwan"
#DEVICES_TO_DISABLE_ON_WWAN_CONNECT="wifi"

# Radio devices to enable on disconnect.
DEVICES_TO_ENABLE_ON_LAN_DISCONNECT="wifi"
#DEVICES_TO_ENABLE_ON_WIFI_DISCONNECT=""
#DEVICES_TO_ENABLE_ON_WWAN_DISCONNECT=""

# Radio devices to enable/disable when docked.
#DEVICES_TO_ENABLE_ON_DOCK=""
#DEVICES_TO_DISABLE_ON_DOCK=""

# Radio devices to enable/disable when undocked.
#DEVICES_TO_ENABLE_ON_UNDOCK="wifi"
#DEVICES_TO_DISABLE_ON_UNDOCK=""
```

linrunner, 2020-06-07:
Reconsidered and rejected.

closed 2020-06-07 by linrunner

marmistrz, 2020-06-07:
What's the rejection reason?