linrunner/TLP #315: Strange noise produced with SATA_LINKPWR_ON_BAT option on Thinkpad T470

ema-pe, 2018-01-08:
### Symptom data

1. Does the problem occur on battery or AC or both? Battery.
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1. [Here](https://gist.github.com/anonymous/84ae0034d42beb338c20c2900bd5a6be). I use the 0.9 version because I found this on Debian Stable.

### Expected behavior

The default configuration should work without problems on my **Thinkpad T470**.

### Actual behavior

The computer produce a strange noise with the default option `SATA_LINKPWR_ON_BAT=min_power`; the same noise is produced with `medium_power` but not with `max_performance`.

### Steps to reproduce the problem

Simply use the default configuration and check that battery mode is enabled.

_p.s. I have a small off-topic question. When I run `bluetooth off` why it shows `(software)` note?_

linrunner, 2018-01-14:
What you describe could be a common hardware issue called "CPU Whining":
> When the current flow rises or falls sharply, coil wires and other conductors can become slightly deformed. If the current fluctuates in an unfavourable rhythm, audible vibrations occur.

([Source](https://www.heise.de/ct/hotline/Jaulender-Prozessor-CPU-Whining-3032248.html))

Recent Intel processors have a tight relationship between ALPM and low power states of the CPU ([Source](https://mjg59.dreamwidth.org/34868.html)).

You could either disable ALPM in TLP's configuration or open a warranty claim against Lenovo. Excessive CPU whine is not acceptable on a hardware that new.

ema-pe, 2018-01-14:
Thanks for the reply. I think that I'll keep the option `SATA_LINKPWR_ON_BAT` on `max_performance`, because it is the only value that doesn't produce any noise.

There are problems to keep this option? Will my laptop can be damaged?

linrunner, 2018-01-14:
Don't worry, there's no danger for your hardware.

closed 2018-01-24 by linrunner