linrunner/TLP #317: Error in upgrading tlp-rdw 1.0-1~trusty to 1.1-1~trusty: "dpkg-maintscript-helper: error: last version is missing"

adjl, 2018-01-27:
Hello,

I encountered an error in upgrading the `tlp-rdw` package from version `1.0-1~trusty` to `1.1-1~trusty` (latest according to Synaptic).

After running the `sudo apt-get update && sudo apt-get upgrade` commands, I got the following message:
```
Preparing to unpack .../tlp-rdw_1.1-1~trusty_all.deb ...
dpkg-maintscript-helper: error: last version is missing
dpkg: error processing archive /var/cache/apt/archives/tlp-rdw_1.1-1~trusty_all.deb (--unpack):
 subprocess new pre-installation script returned error exit status 1
dpkg-maintscript-helper: error: last version is missing
dpkg: error while cleaning up:
 subprocess new post-removal script returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/tlp-rdw_1.1-1~trusty_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
$ _
```

Interestingly, the `tlp` package was successfully upgraded to `1.1-1~trusty`.

Looking at [TLP on Launchpad](https://launchpad.net/~linrunner/+archive/ubuntu/tlp), the latest version for Trusty is `1.1-2~trusty` although its build status is still pending. Would this be a probable cause?

Sorry, it is possible that the problem may be on my end (e.g. dpkg, broken package). I just want to run this by you first, to be sure.

---
I use the PPA `ppa:linrunner/tlp`.
Running on Linux Mint 17.3 (64-bit).
Kernel 3.19.0-32-generic

Cheers

linrunner, 2018-01-27:
Hi,

thanks for your report. The 1.1-1~trusty packages were in fact broken.

I've uploaded 1.1-2~trusty to the PPA. Please update again:

> apt-get update && apt-get dist-upgrade

Sorry for the inconvenience.

adjl, 2018-01-27:
Cheers, that fixed it.

closed 2018-01-27 by adjl