linrunner/TLP #319: get_disk_dev logic is not compatible with NVMe devices

hexchain, 2018-02-04:
NVMe device names are like "nvme0n1" (with its partitions being "nvme0n1p1", "nvme0n1p2", ...). [This line](https://github.com/linrunner/TLP/blob/ec1803accb01211c7bdfeb3f34268c2fa4443688/tlp-functions.in#L1036) removes the trailing number, resulting in a name like "nvme0n" which is invalid.

Using disk IDs does not help, either.

linrunner, 2018-02-07:
Hi, 

thanks for your report. I will look into that.

Any suggestions how to fix the regex for NVMe?

hexchain, 2018-02-09:
Can't think of a better regex. Maybe either hardcode a drive prefix list (like in udev rules), or just stop trying to infer drive from partition?

linrunner, 2018-04-03:
Done. Could you please test?

hexchain, 2018-04-07:
Looks good!

hexchain, 2018-10-08:
Just found another problem (not quite related to `get_disk_dev`): `tlp diskid` only shows ATA devices, as they are explicitly filtered at https://github.com/linrunner/TLP/blob/master/tlp.in#L371.

On my device, `/dev/disk/by-id` looks like this:

```
$ /usr/bin/ls -1 /dev/disk/by-id/
dm-name-cryptroot
dm-uuid-CRYPT-LUKS2-xxxxxxxxxxxxx-cryptroot
nvme-eui.00080d03000fec29
nvme-eui.00080d03000fec29-part1
nvme-eui.00080d03000fec29-part2
nvme-THNSN5512GPUK_NVMe_TOSHIBA_512GB_373B535IKSJU
nvme-THNSN5512GPUK_NVMe_TOSHIBA_512GB_373B535IKSJU-part1
nvme-THNSN5512GPUK_NVMe_TOSHIBA_512GB_373B535IKSJU-part2
```

linrunner, 2018-12-13:
Fixed and added support in tlp-stat too. Thanks again for your suggestion.

closed 2018-12-16 by linrunner

linrunner, 2019-03-11:
Released with TLP 1.2.