linrunner/TLP #320: TLP not switching to AC when charging

parkerlreed, 2018-02-06:
### Symptom data

1. AC - TLP doesn't detect when battery is charging
2. https://gist.github.com/parkerlreed/2ca76a19e180e79db41f608b2de3f5b7

### Expected behavior

TLP should switch to and stay in AC mode when charging

### Actual behavior

TLP doesn't switch to AC when plugged in.

I have a GPD Pocket which has a slightly weird battery circuit. Not sure if it's just not exposing something that TLP expects or if something may be wrong.

````
[parker@gpdpocket ~]$ upower -e
/org/freedesktop/UPower/devices/battery_max170xx_battery
/org/freedesktop/UPower/devices/line_power_bq24190_charger
/org/freedesktop/UPower/devices/mouse_hidpp_battery_1
/org/freedesktop/UPower/devices/DisplayDevice
[parker@gpdpocket ~]$ upower -i /org/freedesktop/UPower/devices/battery_max170xx_battery #Unplugged
  native-path:          max170xx_battery
  power supply:         yes
  updated:              Tue 06 Feb 2018 02:38:45 PM EST (36 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               discharging
    warning-level:       none
    energy:              10.3093 Wh
    energy-empty:        0 Wh
    energy-full:         27.7725 Wh
    energy-full-design:  24.0555 Wh
    energy-rate:         5.17826 W
    voltage:             3.64547 V
    time to empty:       2.0 hours
    percentage:          37%
    temperature:         22 degrees C
    capacity:            100%
    technology:          lithium-ion
    icon-name:          'battery-good-symbolic'
  History (rate):
    1517945925  5.178   discharging
    1517945925  4.466   discharging

[parker@gpdpocket ~]$ upower -i /org/freedesktop/UPower/devices/line_power_bq24190_charger #Unplugged
  native-path:          bq24190-charger
  power supply:         yes
  updated:              Tue 06 Feb 2018 02:38:45 PM EST (47 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              no
    icon-name:          'ac-adapter-symbolic'

[parker@gpdpocket ~]$ upower -i /org/freedesktop/UPower/devices/battery_max170xx_battery #Plugged in 
  native-path:          max170xx_battery
  power supply:         yes
  updated:              Tue 06 Feb 2018 02:39:36 PM EST (11 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:       none
    energy:              10.234 Wh
    energy-empty:        0 Wh
    energy-full:         27.7725 Wh
    energy-full-design:  24.0555 Wh
    energy-rate:         3.02921 W
    voltage:             3.69937 V
    time to full:        5.8 hours
    percentage:          36%
    temperature:         22 degrees C
    capacity:            100%
    technology:          lithium-ion
    icon-name:          'battery-good-charging-symbolic'
  History (charge):
    1517945976  36.000  charging
  History (rate):
    1517945976  3.029   charging
    1517945976  2.834   charging
    1517945925  5.178   discharging
    1517945925  4.466   discharging

[parker@gpdpocket ~]$ upower -i /org/freedesktop/UPower/devices/line_power_bq24190_charger #Plugged in
  native-path:          bq24190-charger
  power supply:         yes
  updated:              Tue 06 Feb 2018 02:39:36 PM EST (23 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              yes
    icon-name:          'ac-adapter-symbolic'
````

linrunner, 2018-02-07:
Hi,

AC mode depends on proper detection of the power supply e.g. charger. 

I need the following output when plugged in:

    tlp-stat --psup

To check the events that TLP sees when

1. plugging in 
1. unplugging

show 

    tlp-stat --pev

after both actions.

parkerlreed, 2018-02-08:
````
[parker@gpdpocket ~]$ tlp-stat --psup
--- TLP 1.1 --------------------------------------------

+++ Power supply diagnostic
/sys/class/power_supply/bq24190-charger/type:USB
/sys/class/power_supply/bq24190-charger/online:1
/sys/class/power_supply/max170xx_battery/type:Battery
/sys/class/power_supply/max170xx_battery/present:1
````

Unplug
````
[parker@gpdpocket ~]$ tlp-stat --pev
--- TLP 1.1 --------------------------------------------

+++ Monitor power supply events -- cancel with ^C

monitor will print the received events for:
UDEV - the event which udev sends out after rule processing

UDEV  [180.365456] change   /devices/pci0000:00/808622C1:00/i2c-0/i2c-MAX17047:00/power_supply/max170xx_battery (power_supply)
ACTION=change
DEVPATH=/devices/pci0000:00/808622C1:00/i2c-0/i2c-MAX17047:00/power_supply/max170xx_battery
POWER_SUPPLY_CAPACITY=53
POWER_SUPPLY_CHARGE_COUNTER=12531000
POWER_SUPPLY_CHARGE_FULL=7935000
POWER_SUPPLY_CHARGE_FULL_DESIGN=6873000
POWER_SUPPLY_CHARGE_NOW=4221500
POWER_SUPPLY_CURRENT_AVG=83772
POWER_SUPPLY_CURRENT_NOW=103584
POWER_SUPPLY_CYCLE_COUNT=1471
POWER_SUPPLY_HEALTH=Good
POWER_SUPPLY_NAME=max170xx_battery
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SCOPE=System
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_TEMP=220
POWER_SUPPLY_TEMP_ALERT_MAX=1270
POWER_SUPPLY_TEMP_ALERT_MIN=-1280
POWER_SUPPLY_TEMP_MAX=700
POWER_SUPPLY_TEMP_MIN=0
POWER_SUPPLY_VOLTAGE_AVG=3971171
POWER_SUPPLY_VOLTAGE_MAX=4380000
POWER_SUPPLY_VOLTAGE_MIN=3040000
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=3500000
POWER_SUPPLY_VOLTAGE_NOW=3981718
POWER_SUPPLY_VOLTAGE_OCV=3944765
SEQNUM=2909
SUBSYSTEM=power_supply
USEC_INITIALIZED=180168732

UDEV  [180.414228] change   /devices/pci0000:00/808622C1:06/i2c-6/i2c-INT34D3:00/cht_wcove_ext_chgr/i2c-15/i2c-bq24190/power_supply/bq24190-charger (power_supply)
ACTION=change
DEVPATH=/devices/pci0000:00/808622C1:06/i2c-6/i2c-INT34D3:00/cht_wcove_ext_chgr/i2c-15/i2c-bq24190/power_supply/bq24190-charger
POWER_SUPPLY_CHARGE_TERM_CURRENT=256000
POWER_SUPPLY_CHARGE_TYPE=Fast
POWER_SUPPLY_CONSTANT_CHARGE_CURRENT=2816000
POWER_SUPPLY_CONSTANT_CHARGE_CURRENT_MAX=4544000
POWER_SUPPLY_CONSTANT_CHARGE_VOLTAGE=4368000
POWER_SUPPLY_CONSTANT_CHARGE_VOLTAGE_MAX=4400000
POWER_SUPPLY_HEALTH=Good
POWER_SUPPLY_INPUT_CURRENT_LIMIT=500000
POWER_SUPPLY_MANUFACTURER=Texas Instruments
POWER_SUPPLY_MODEL_NAME=bq24190
POWER_SUPPLY_NAME=bq24190-charger
POWER_SUPPLY_ONLINE=0
POWER_SUPPLY_PRECHARGE_CURRENT=256000
POWER_SUPPLY_SCOPE=System
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_TEMP_ALERT_MAX=1200
SEQNUM=2912
SUBSYSTEM=power_supply
USEC_INITIALIZED=180212952
````

Plug
````
[parker@gpdpocket ~]$ tlp-stat --pev
--- TLP 1.1 --------------------------------------------

+++ Monitor power supply events -- cancel with ^C

monitor will print the received events for:
UDEV - the event which udev sends out after rule processing

UDEV  [229.177632] change   /devices/pci0000:00/808622C1:00/i2c-0/i2c-MAX17047:00/power_supply/max170xx_battery (power_supply)
ACTION=change
DEVPATH=/devices/pci0000:00/808622C1:00/i2c-0/i2c-MAX17047:00/power_supply/max170xx_battery
POWER_SUPPLY_CAPACITY=53
POWER_SUPPLY_CHARGE_COUNTER=12526500
POWER_SUPPLY_CHARGE_FULL=7935000
POWER_SUPPLY_CHARGE_FULL_DESIGN=6873000
POWER_SUPPLY_CHARGE_NOW=4217500
POWER_SUPPLY_CURRENT_AVG=-284076
POWER_SUPPLY_CURRENT_NOW=114192
POWER_SUPPLY_CYCLE_COUNT=1471
POWER_SUPPLY_HEALTH=Good
POWER_SUPPLY_NAME=max170xx_battery
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SCOPE=System
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_TEMP=220
POWER_SUPPLY_TEMP_ALERT_MAX=1270
POWER_SUPPLY_TEMP_ALERT_MIN=-1280
POWER_SUPPLY_TEMP_MAX=700
POWER_SUPPLY_TEMP_MIN=0
POWER_SUPPLY_VOLTAGE_AVG=3943281
POWER_SUPPLY_VOLTAGE_MAX=4380000
POWER_SUPPLY_VOLTAGE_MIN=3040000
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=3500000
POWER_SUPPLY_VOLTAGE_NOW=3970625
POWER_SUPPLY_VOLTAGE_OCV=3944531
SEQNUM=2923
SUBSYSTEM=power_supply
USEC_INITIALIZED=228983388

UDEV  [229.251889] change   /devices/pci0000:00/808622C1:06/i2c-6/i2c-INT34D3:00/cht_wcove_ext_chgr/i2c-15/i2c-bq24190/power_supply/bq24190-charger (power_supply)
ACTION=change
DEVPATH=/devices/pci0000:00/808622C1:06/i2c-6/i2c-INT34D3:00/cht_wcove_ext_chgr/i2c-15/i2c-bq24190/power_supply/bq24190-charger
POWER_SUPPLY_CHARGE_TERM_CURRENT=256000
POWER_SUPPLY_CHARGE_TYPE=Fast
POWER_SUPPLY_CONSTANT_CHARGE_CURRENT=2816000
POWER_SUPPLY_CONSTANT_CHARGE_CURRENT_MAX=4544000
POWER_SUPPLY_CONSTANT_CHARGE_VOLTAGE=4368000
POWER_SUPPLY_CONSTANT_CHARGE_VOLTAGE_MAX=4400000
POWER_SUPPLY_HEALTH=Good
POWER_SUPPLY_INPUT_CURRENT_LIMIT=500000
POWER_SUPPLY_MANUFACTURER=Texas Instruments
POWER_SUPPLY_MODEL_NAME=bq24190
POWER_SUPPLY_NAME=bq24190-charger
POWER_SUPPLY_ONLINE=1
POWER_SUPPLY_PRECHARGE_CURRENT=256000
POWER_SUPPLY_SCOPE=System
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_TEMP_ALERT_MAX=1200
SEQNUM=2924
SUBSYSTEM=power_supply
USEC_INITIALIZED=229044111
```

linrunner, 2018-02-08:
Thanks. 

    /sys/class/power_supply/bq24190-charger/type:USB

is the culprit, TLP expects type:Mains.

I suggest you change line 339 in **/usr/share/tlp/tlp-functions** by hand as shown in the [commit](https://github.com/linrunner/TLP/commit/11223cb63309eb135292abe6d41e7cf40fa07637#diff-d121ba8b07515d3eeeed0ef44369373d) and report if it works now.

parkerlreed, 2018-02-08:
That worked great! Thanks for the help.

parkerlreed, 2018-02-08:
Oh sorry for closing too son :)

linrunner, 2018-02-08:
No problem. Will be closed when released with 1.2.

parkerlreed, 2018-12-11:
Cheers!

linrunner, 2019-03-11:
Released with TLP 1.2.

parkerlreed, 2019-03-11:
@linrunner Thanks! Don't have that hardware anymore, but I'm sure that'll help quite a few people.