linrunner/TLP #321: RFC: use upcoming kernel interface for battery threshold configuration

t-8ch, 2018-02-08:
The upcoming 4.17 kernel (maybe even 4.16) will contain a new in-kernel API to expose information and controls like the battery threshold to userspace. [0]
It also comes with a reference implementation for new thinkpads (using the same logic as tpacpi-bat)

This could be used by TLP to replace tpacpi-bat.
(Maybe even a driver for the batteries supported by tp_smapi emerges)

I am willing to contribute a driver.
Should the driver be in addition to tpacpi-bat or will one implementation be enough?
(If the new logic in TLP is released after the kernel there shouldn't be a problem with dropping support for tpacpi-bat)

[0] https://patchwork.kernel.org/patch/10205339/
[1] https://patchwork.kernel.org/patch/10205347/
[2] https://patchwork.kernel.org/patch/10205353/
[3] https://patchwork.kernel.org/patch/10205359/

linrunner, 2018-02-08:
Looks interesting, but it's only infrastructure. The comment in [0] says

> The thinkpad_acpi module can use this API to expose
sysfs attributes that it controls inside the ACPI battery driver
sysfs tree, under /sys/class/power_supply/BATN/.

So someone must actually add this to thinkpad_acpi. I remember Julian Andres Klode attempted to integrate thresholds in thinkpad_acpi years ago, but never finished the task. 

tpacpi-bat is not a [kernel] driver. With sysfiles in place below /sys/class/power_supply/BATN/ there is no need for additional tools. TLP can write them itself.

linrunner, 2018-02-08:
OK, OK. I missed the thinkpad_acpi implementation in [1] :-)

t-8ch, 2018-02-08:
With "driver"to implement I meant "logic inside TLP to talk to the new API".

linrunner, 2018-02-09:
You're welcome of course :-).

t-8ch, 2018-02-09:
Hm, the new API is still missing the force discharge and inhibit charge methods.
Maybe we wait for them and then throw out tpacpi-bat?

linrunner, 2018-02-09:
A threefold selection logic ( tp-smapi - tpapci-bat - new API ) is needed anyway, not all people will have the new kernel at once. Maybe you could ask the kernel devs about force_discharge?

t-8ch, 2018-02-09:
Urgh :-)

We could split it:
* I will get the other methods into the kernel.
* You do the logic inside TLP.

linrunner, 2018-02-09:
Fair enough :-)

smclt30p, 2018-02-09:
Developer of the kernel driver here.

I implemented a whole new generic extension system for the ACPI battery driver inside the kernel, and once that gets sanitized and merged into mainline, I then will move onto the other stuff, like force discharge and inhibit charge as a separate patch. Those patches will be small as most of the infrastructure is inside this patch series, mainly 1/4.

Keep your heads up guys, the future is bright :+1:

t-8ch, 2018-02-10:
@smclt30p  Great! Thanks for your work.
Does your source of official documentation by any chance include hardware errata?
(Asking [for a friend](https://bugzilla.kernel.org/show_bug.cgi?id=198019) ...)
Sorry for being offtopic.

smclt30p, 2018-02-10:
> @smclt30p  Great! Thanks for your work.
> Does your source of official documentation by any chance include hardware errata?

No its only ACPI Battery stuff. 
Shoot an email to Peter FP1 Zhang <zhangfp1@lenovo.com> and he maybe will assist you guys.

smclt30p, 2018-02-27:
Update guys: 

The patch has been merged and is coming in Linux 4.17:
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/drivers/platform/x86/thinkpad_acpi.c?h=next-20180226#n9216

Following the release of 4.17, I will build the inhibit_charge and force_discharge methods.
Stay tuned!

linrunner, 2018-03-02:
Great :-)

linrunner, 2018-03-12:
@smclt30p : I need a "nickname" for the new API to distinguish it from tp-smapi and tpacpi-bat. Any thoughts?

smclt30p, 2018-03-12:
"native"

linrunner, 2018-03-12:
Sounds a bit ... generic.

smclt30p, 2018-03-12:
Well it's what it basically is. And this obsoletes tpacpi-bat basically, so do tpacpi-native.

linrunner, 2018-03-13:
Well, what about "acpibat-native" or abbreviated "acpibat"?

linrunner, 2018-04-05:
@smclt30p :
> Following the release of 4.17, I will build the inhibit_charge and force_discharge methods.
Will the files actually be called  **inhibit_charge** and **force_discharge**?

smclt30p, 2018-04-06:
Yes.

smclt30p, 2018-04-06:
But we have hit a brick wall, so far the "status" attribute is broken because one of the patches got reverted because UPower is broken. And no upower maintainer is active?

See: https://bugs.freedesktop.org/show_bug.cgi?id=105482

t-8ch, 2018-04-06:
The linked issue is assigned to @hughsie, who is definitively still around.

hughsie, 2018-04-06:
Can you define a bit what you mean by "upower is broken"?

smclt30p, 2018-04-06:
See the above bug report I attached on fd.o

linrunner, 2018-04-09:
Just for my own reference: above patches as commits in Linus' tree (4.17)

[0] [battery: Add the battery hooking API](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/acpi?id=fa93854f7a7ed63d054405bf3779247d5300edd3)
[1] [pm: add to_power_supply macro to the API](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/power/supply?id=285995d15d3b1725d021a8a274e55f2ce30ccfa0)
[2] [thinkpad_acpi: Add support for battery thresholds](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/platform/x86/thinkpad_acpi.c?id=2801b9683f740012863f7f0b1f0bc770c417fe72)
[3] [battery: Add the ThinkPad "Not Charging" quirk](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/acpi?id=91eea70e5e5ce12eb1c7cd922e561fab43e201bd)
[Revert](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/acpi?id=7a4ea10c01cdda3d66aa98af258f150d863aee24) of [3]

@smclt30p : doesn't the whole thing work even without the last patch for the status? I'm used to reading "unknown" when thresholds are effective.

smclt30p, 2018-04-09:
Yes that patch has been reverted because it breaks upower and the userspace. It needs a new revision. I will look into it later this week.

smclt30p, 2018-04-10:
I have submitted a new version of the status quirk: https://patchwork.kernel.org/patch/10332631/

Let's see how this goes. Do you have any comments about the above patch?

linrunner, 2018-04-17:
@smclt30p : Patch works for me (T450s, 4.17-rc1).

Nevertheless the above upower bug report should be pursued further. I'd prefer having a distinct state like "not charging" when a threshold is effective.

smclt30p, 2018-04-18:
> @smclt30p <https://github.com/smclt30p> : Patch works for me (T450s,
> 4.17-rc1).
>
> Nevertheless the above upower bug report should be pursued further.

Yeah good luck with that. I tried it all.

I'd prefer having a distinct state like "not charging" when a threshold is

smclt30p, 2018-05-13:
The kernel patches for force_discharge and inhibit_charge have freshly landed into Patchwork:

[0] https://patchwork.kernel.org/patch/10396259/
[1] https://patchwork.kernel.org/patch/10396265/
[2] https://patchwork.kernel.org/patch/10396271/

Let's see if the get merged on first try.

linrunner, 2018-06-04:
Thorsten Leemhuis writes about the new interface: https://www.heise.de/ct/artikel/Die-Neuerungen-von-Linux-4-17-4061166.html?seite=3#nav_sicherheit_f%C3%BCr__2

linrunner, 2018-06-13:
@smclt30p : testing the new features with TLP on my T450s produces two problems 

(1) thresholds for BAT1 not writable (vanilla Kernel v4.17)

    echo 42 > BAT1/charge_start_threshold

results in: `bash: echo: write error: No such device`. BAT0 works.

(2) force_discharge BAT0 writes BAT1 too (v4.17 plus above Patches)

    echo 1 > BAT0/force_discharge

results in: BAT0/force_discharge = 1 *and*  BAT1/force_discharge = 1 *and* BAT1 being discharged! 

I guess you need a kernel bug report for (1)?

linrunner, 2018-06-15:
natacpi is now integrated into TLP. Please test. 

Note: because of the above bugs it is still disabled by default (ENABLE_NATACPI=0).

t-8ch, 2018-06-16:
I have the same issue as @linrunner 

@smclt30p the problem is the following:
`tpacpi_battery_probe()` is called for every battery, yet it `memset(0)`s the whole `battery_info` struct, which holds the information about supported operations for *all* batteries.
This means, that the last battery probed will overwrite the support for all batteries.
The struct should only be initialized once, probably in some `__init` function or statically.

Also `tpacpi_battery_store()` returns `-ENODEV` if `what == THRESHOLD_START` and `tpacpi_battery_set()` fails, while it is `-EINVAL` for `what == THRESHOLD_STOP`

linrunner, 2018-06-25:
@t-8ch : thanks for your insight.

t-8ch, 2018-06-25:
@linrunner Unfortunately [the patch](https://sourceforge.net/p/ibm-acpi/mailman/message/36344427/) did not get any feedback yet :-(

smclt30p, 2018-06-25:
> @linrunner <https://github.com/linrunner> Unfortunately the patch
> <https://sourceforge.net/p/ibm-acpi/mailman/message/36344427/> did not
> get any feedback yet :-(

I wanted to post some feedback but I gave up in the end. The process of
patch submissions to the Linux kernel is very volatile. Its based on an
ancient waiting scheme that has no specific assigned person or anything. If
the kernel devs actually used Bugzilla or something similar, the process of
patch submissions would be much more reliable. I myself have a patch
waiting for review and nothing is happening for a good 2 months now.

It sucks.

t-8ch, 2018-06-25:
I have been advised to post it to more lists in the future.
For the current patch I will ping them again next week.

smclt30p, 2018-06-25:
> I have been advised to post it to more lists in the future.
> For the current patch I will ping them again next week.

I dont think that would yeild any results. The whole patch submission
system is a mess.

linrunner, 2018-06-25:
@smclt30p whining doesn't help ;-). 4.17 is a released kernel, so a bug report is fine imho. To advance things I have created a bug report:

https://bugzilla.kernel.org/show_bug.cgi?id=200275

 @t-8ch : I suggest you confirm the bug and attach your patch as a solution.

smclt30p, 2018-06-25:
> @smclt30p <https://github.com/smclt30p> whining doesn't help ;-). 4.17 is
> a released kernel, so a bug report is fine imho. To advance things I have
> created a bug report:
>
> https://bugzilla.kernel.org/show_bug.cgi?id=200275

Nobody reads those. I have a bug open in the fd.o bugzilla for 4 months and
a patch and nothing is happening.

<https://bugzilla.kernel.org/show_bug.cgi?id=200275>

linrunner, 2018-06-26:
@t-8ch : your patch solves problem (1) above, thresholds work now for BAT0 and BAT1. For (2) symptoms remains the same. 

On the X220 (BAT0 only) thresholds and force_discharge work.

Btw: the patch doesn't merge with 4.17.

t-8ch, 2018-06-26:
@linrunner 
For 2) I actually think this may be a limitation of the EC. I will try tomorrow on my box.

Lets see if it gets into 4.18 and then start with 4.17

t-8ch, 2018-06-27:
@linrunner It seems the ACPI call is a little off (at least it is different to `tpacpi-bat` which works correctly:

```sh
# example to explain the debugging stuff:
troy# echo 80 > BAT1/charge_stop_threshold
[ 1299.997985] thinkpad_acpi: th stop: 250 # dmesg, "250 is raw acpi argument
troy# /usr/share/tlp/tpacpi-bat -s SP 2 80
ACPI: BCSS 0x250 # again, 0x250 is the raw acpi argument                 
```
This shows that both methods should use the same arguments (both work)

```sh
troy# /usr/share/tlp/tpacpi-bat -s FD 2 1 
ACPI: BDSS 0x201
troy# /usr/share/tlp/tpacpi-bat -g FD 2  
yes
troy# echo 1 >BAT1/force_discharge
[ 1440.267608] thinkpad_acpi: force: 102
troy# /usr/share/tlp/tpacpi-bat -g FD 2   
yes (break on AC detach)     
```
Observations:
* The values are different
* The native method does not set a plain `force_discharge` but one that breaks on AC.
  (For inhibit charge this is documented in the source)
* tpacpi-bat uses `BDSS` while tpacpi-bat uses `PSBS` for setting the `force_discharge` flag.

Unfortunately tpacpi-bat is licensed under the GPLv3+ while the kernel is GPLv2, so a direct port would be a license violation.

The most likely problem:
All other values are set with the following logic:
```
param = value;
param |= battery << 8;
```

For force_discharge it is:

```
param = battery;
param |= value << 8;
```

So probably a typo.

mcsurly, 2018-07-02:
Hi.
kernel 4.17 is live, and my Fedora 28 has:
/sys/class/power_supply/BAT0/charge_start_threshold
/sys/class/power_supply/BAT0/charge_stop_threshold

Note, with threshold in effect, /sys/class/power_supply/BAT0/status does show as Unknown.

Since this is here, and live, can we (you :D ) go ahead and add support for "native" or kernel_native whatever you want to call it? Whoever did the packaging for Fedora did a masterful job, and it was very easy for me to add the TLP repository, and it builds kmod-acpi_call like a champ.
But it's unnecessary baggage now, for the most essential functions (thresholds).
I know force_discharge and inhibit_charge are still not available without acpi_call, but it would be nice for most users to be able to use thresholds out of the box in the interim.

Many thanks to the three of you. Excellent work.

linrunner, 2018-07-02:
@mcsurly : it's in the master branch already. As long as BAT1 thresholds in kernel 4.17 (and BAT0 force_discharge) do not work, it will not be enabled by default. Use NATACPI_ENABLE=1. 

@t-8ch : 

```
	case FORCE_DISCHARGE:
-		param = battery;
-		param |= value << 8;
		if ACPI_FAILURE(tpacpi_battery_acpi_eval(SET_DISCHARGE, &ret, param)) {
			pr_err("failed to set force dischrage on %d", battery);
			return -ENODEV;
		}
		return 0;

```
does not fix with force_discharge.

smclt30p, 2018-07-02:
> @mcsurly <https://github.com/mcsurly> : it's in the master branch
> already. As long as BAT1 thresholds in kernel 4.17 (and BAT0
> force_discharge) do not work, it will not be enabled by default. Use [...]
force_discharge and inhibit_charge work just fine on my X220 that I wrote
the patch on.

There is a couple of different ACPI methods for force_discharge, some of
which work on my X220 and some of which are documented to work but in
practice they don’t. That is from official Lenovo documentation.

 does not fix with force_discharge.

linrunner, 2018-07-26:
OK, after some experimenting I propose to change the ACPI call for force_discharge to as in tpacpi-bat:

linrunner, 2018-07-26:
I want to avoid such a logic. It will be a royal pain to maintain in the kernel and I'll get flooded with issues for TLP.

Hence my proposal above. The calls in tpacpi-bat simply work with [very few exceptions](https://github.com/teleshoes/tpacpi-bat/blob/master/tpacpi-bat#L394). 

Why reinvent the wheel here?

t-8ch, 2018-07-26:
If it works (I trust your experience here) I am all for it!

linrunner, 2018-07-27:
@t-8ch : does it work on your ThinkPad?

t-8ch, 2018-07-27:
@linrunner I won't be able to test it until tomorrow.

t-8ch, 2018-07-31:
@linrunner Works for me.

t-8ch, 2018-08-25:
The fix for the multi-battery probing will be in 4.19

linrunner, 2018-08-31:
@t-8ch : great :-)

mcsurly, 2018-10-03:
Soooo... @linrunner with the fix for multi-battery probing going in 4.19, and your BAT0 force_discharge patch going in as well.... when 4.19 hits Fedora would you be ready to make a release 1.1.1 or 1.2?
I think you could set NATACPI_ENABLE=1 by default at that point as well, but ... I'm comfortable changing that setting myself!
I think there is a lot of appeal of having TLP able to save battery wear on ThinkPads without requiring you to add another yum repo (and akmod kernel modeul) to pick up acpi_call.

linrunner, 2018-10-04:
> and your BAT0 force_discharge patch going in as well....

I didn't see the force_discharge patches from @smclt30p  being merged. Do you have a reference?

smclt30p, 2018-10-04:
They are not merged. I am sick and tired of re-submitting patches because some picky maintainer does not like my choice of brace and comment placement and style. Feel free to edit them and re-submit them.

pschichtel, 2019-01-15:
Seems like the patch landed in 4.19.

linrunner, 2019-02-21:
I'm trying to get the patch for multi-battery probing into Ubuntu's 4.18 too:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1812099

[74 more replies]

closed 2022-01-25 by linrunner