linrunner/TLP #322: Battery thresholds disabled in config, but still active according to tlp-stat?

alexzeitgeist, 2018-02-10:
### Symptom data

```
--- TLP 1.1 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
TLP_PERSISTENT_DEFAULT=0
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_SCALING_GOVERNOR_ON_AC=powersave
CPU_SCALING_GOVERNOR_ON_BAT=powersave
CPU_HWP_ON_AC=balance_performance
CPU_HWP_ON_BAT=balance_power
CPU_MIN_PERF_ON_AC=0
CPU_MAX_PERF_ON_AC=100
CPU_MIN_PERF_ON_BAT=0
CPU_MAX_PERF_ON_BAT=75
CPU_BOOST_ON_AC=1
CPU_BOOST_ON_BAT=0
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=power
DISK_DEVICES="sda sdb"
SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
AHCI_RUNTIME_PM_TIMEOUT=15
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=on
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_AC=0
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
USB_AUTOSUSPEND=1
USB_BLACKLIST_BTUSB=0
USB_BLACKLIST_PHONE=0
USB_BLACKLIST_PRINTER=0
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = LENOVO ThinkPad T460 20FMXXXXXX
BIOS           = R06ET56W (1.30 )
Release        = Ubuntu 17.10
Kernel         = 4.13.0-32-generic #35-Ubuntu SMP Thu Jan 25 09:13:46 UTC 2018 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.13.0-32-generic root=UUID=66cf4166-5dd3-4362-bf30-ba595b1f8f8e ro quiet splash intel_pstate=enable i915.semaphores=1 elevator=deadline vt.handoff=7
Init system    = systemd v234
Boot mode      = UEFI

+++ TLP Status
State          = enabled
Last run       = 15:46:23,    233 sec(s) ago
[... 207 lines cut ...]

The reason: I want to make use of the dualmode battery firmware, and thus not rely on TLP settings for charging/discharging my batteries.

### Actual behavior

TLP appears to have set some thresholds anyway (see output above). Also:

```
root@thinkpad-linux:/usr/share/tlp# /usr/share/tlp/tpacpi-bat -v -g ST 1
Call    : \_SB.PCI0.LPC.EC.HKEY.BCTG 0x1
Response: 0x34blled
75 (relative percent)
root@thinkpad-linux:/usr/share/tlp# /usr/share/tlp/tpacpi-bat -v -g SP 1
Call    : \_SB.PCI0.LPC.EC.HKEY.BCSG 0x1
Response: 0x360lled
96 (relative percent)
root@thinkpad-linux:/usr/share/tlp# /usr/share/tlp/tpacpi-bat -v -g ST 2
Call    : \_SB.PCI0.LPC.EC.HKEY.BCTG 0x2
Response: 0x34blled
75 (relative percent)
root@thinkpad-linux:/usr/share/tlp# /usr/share/tlp/tpacpi-bat -v -g SP 2
Call    : \_SB.PCI0.LPC.EC.HKEY.BCSG 0x2
Response: 0x300lled
0 (default)
```

linrunner, 2018-02-10:
Hi, please read the [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#how-to-disable-thresholds) before opening an issue. Thanks.

closed 2018-02-10 by linrunner

alexzeitgeist, 2018-02-10:
Thanks, my fault. I swear I did read the FAQ, but must have skipped this point.

linrunner, 2018-02-10:
That's the downside of an extensive FAQ ... ;-)