linrunner/TLP #323: "tlp fullcharge" not working unless I remove and reinsert the battery

Ottaviocr, 2018-02-15:
### Symptom data
The tlp battery thresholds work fine on my Thinkpad e130, but the "fullcharge" option doesn't work straight away. The only workaround is to remove and then reinsert the battery, which is annoying,

1. Does the problem occur on battery or AC or both?
Only applies to AC mode
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.

`tlp-stat` before the "fullcharge" command:
https://gist.github.com/Ottaviocr/a347f3922bfbcc1204ec272b5fb22178

`tlp-stat` after the "fullcharge" command:
https://gist.github.com/Ottaviocr/8b974c88332622613caa7c6395282431

### Expected behavior
Should start charging immediately up to 100% capacity

### Actual behavior
It doesn't appear to do anything

### Steps to reproduce the problem
sudo tlp fullcharge && sleep 20 && sudo tlp stat

linrunner, 2018-02-15:
After `tlp fullcharge` issue 

    tlp discharge

let it run 2 seconds and  terminate it with Ctrl+C

Ottaviocr, 2018-02-15:
I'm not sure what you want me to do. If I guess correctly, you want me to run tlp discharge for two seconds and then try again with tlp fullcharge ?

If this is the case, I've just done it and there is no change. 

``oc@e130 ~ $ history|tail

 1993  sudo tlp discharge
 1994  sudo tlp fullcharge 
 1995  sudo tlp stat -b
 1996  history
 1997  history|tail
oc@e130 ~ $ sudo tlp stat -b
--- TLP 0.9 --------------------------------------------

+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (kernel module 'tp_smapi' not installed)
tpacpi-bat = active

+++ ThinkPad Battery Status: BAT1 (Main / Internal)
/sys/class/power_supply/BAT1/manufacturer                   = SANYO
/sys/class/power_supply/BAT1/model_name                     = 45N1057
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  62160 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  60770 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  34230 [mWh]
/sys/class/power_supply/BAT1/power_now                      =      0 [mW]
/sys/class/power_supply/BAT1/status                         = Unknown (threshold effective)

tpacpi-bat.BAT1.startThreshold                              =     96 [%]
tpacpi-bat.BAT1.stopThreshold                               =    100 [%]
tpacpi-bat.BAT1.forceDischarge                              =      0

Charge                                                      =   56.3 [%]
Capacity                                                    =   97.8 [%]

linrunner, 2018-02-15:
No, the other way round
1. tlp fullcharge
2. tlp discharge # for 2 seconds

Ottaviocr, 2018-02-15:
I have done that. It's still not charging:

/sys/class/power_supply/BAT1/status                         = Unknown (threshold effective)

linrunner, 2018-02-15:
Then I guess you're out of luck, because your problem is caused by Lenovo's embedded controller (EC) firmware that controls charging and discharging as well as enforcing the thresholds.  

Last thing you may try is to update to the newest BIOS.

Ottaviocr, 2018-02-15:
Both BIOS and battery firmware are up to date.

Unplugging and replugging AC does nothing.

The workaround is to manually set charges:

oc@e130 ~ $ sudo tlp setcharge 95 100
Setting temporary charge thresholds for BAT1:
  stop  = 100
  start = 95
oc@e130 ~ $ sudo tlp stat -b
--- TLP 0.9 --------------------------------------------

+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (kernel module 'tp_smapi' not installed)
tpacpi-bat = active

+++ ThinkPad Battery Status: BAT1 (Main / Internal)
/sys/class/power_supply/BAT1/manufacturer                   = SANYO
/sys/class/power_supply/BAT1/model_name                     = 45N1057
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  62160 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  60770 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  35610 [mWh]
/sys/class/power_supply/BAT1/power_now                      =  33507 [mW]
/sys/class/power_supply/BAT1/status                         = Charging

tpacpi-bat.BAT1.startThreshold                              =     95 [%]
tpacpi-bat.BAT1.stopThreshold                               =     95 [%]
tpacpi-bat.BAT1.forceDischarge                              =      0

Charge                                                      =   58.6 [%]
Capacity                                                    =   97.8 [%]

linrunner, 2018-02-15:
I see. tlp fullcharge is just an alias for tlp setcharge 96 100. As you have found a working workaround I consider this issue solved.

Ottaviocr, 2018-02-15:
Alright.

 Incidentally, how do I set my own tlp aliases, apart from using bash builtin alias?

I'd like to have "recalibrate" pointing to full charge, then full discharge, then full charge again.

linrunner, 2018-02-15:
You must edit /usr/sbin/tlp:

https://github.com/linrunner/TLP/blob/master/tlp.in#L290
https://github.com/linrunner/TLP/blob/master/tlp.in#L290

Ottaviocr, 2018-02-15:
Excellent. Thanks.

closed 2018-02-18 by linrunner

thiswillbeyourgithub, 2019-02-21:
Hi,
I thought my story would be a bit useful : 

typing `sudo tlp fullcharge` returns the correct output but no charging happens (tested with battery already charged at threshold)

typing `'sudo tlp setcharge 96 100` returns the same output but starts charging instantly

I am on ubuntu 18.04 wih a thinkpad T480s

linrunner, 2019-02-22:
Interesting, even if wrongly placed here. 

In fact there is no difference between `tlp setcharge 96 100` and `tlp fullcharge`, refer to the [code]([url](url)).

My guess is, that the firmware for the *80 generation doesn't work reliable with the ACPI calls used by tpacpi-bat and the new natacpi kernel interface (see issue #321). For example on my X1C6 once I set a stop threshold < 100, I can't increase it again.

thiswillbeyourgithub, 2019-02-22:
Sorry for posting at the wrong place, I had just seen this thread a while ago.

Just did again some basic testing and did "sudo tlp fullcharge" and it indeed worked directly. Contray to what I observed the other day.

You seem to be right, unreliable firmware.

linrunner, 2019-02-22:
Another explanation may be that Lenovo changed the ACPI calls and the old ones no longer work properly. The future will show ...

grssnbchr, 2019-10-28:
I have a T480s running Ubuntu 18.04 and neither `tlp fullcharge` nor `tlp setcharge 96 100` work. Charging stops at the previously set `80`. The only fix I see is uninstalling tlp.