linrunner/TLP #324: akmod-tp-smapi fails to build on F27, Linux 4.15.3

jurf, 2018-02-17:
Logfile: [0.42-4-for-4.15.3-300.fc27.x86_64.failed.log](https://github.com/linrunner/TLP/files/1733827/0.42-4-for-4.15.3-300.fc27.x86_64.failed.log)

Installed using [these](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#installation) instructions.

linrunner, 2018-02-18:
tp-smapi needs a [patch](https://github.com/evgeni/tp_smapi/commit/76c5120f7be4880cf2c6801f872327e4e70c449f) for 4.15. The package maintainer is informed.

linrunner, 2018-02-19:
Update is available. Please test.

jurf, 2018-02-19:
Hm, didn’t get any update. I tried reinstalling your repo file and updating but nothing. The version I have still fails.

jurf, 2018-02-19:
Update came but still fails, that’s weird…

[0.42-5-for-4.15.3-300.fc27.x86_64.failed.log](https://github.com/linrunner/TLP/files/1737153/0.42-5-for-4.15.3-300.fc27.x86_64.failed.log)

Thorashh, 2018-02-19:
Sorry, my mistake

New Update available 
akmod-tp_smapi-0.42-6.fc27.x86_64.rpm, tp_smapi-0.42-6.fc27.noarch.rpm

Please do ```dnf upgrade --refresh```
Should work now.

jurf, 2018-02-19:
Thanks, works now!

> Please do dnf upgrade --refresh

And that’ll be the fault of me not getting any update, I completely forgot `update` was deprecated, sorry.

BTW, did you manage to get HDAPSD running with tp-smapi? I keep getting a `Could not find a suitable interface` error.

closed 2018-02-19 by jurf