linrunner/TLP #326: RE_TPACPI_ONLY is not inclusive of ThinkPad 25 - tp_smapi load error

exsilium, 2018-02-27:
Hi! 👋  🙇

It would seem that the `RE_TPACPI_ONLY` pattern for egrep does not currently match for ThinkPad 25:

https://github.com/linrunner/TLP/blob/216a77dce72c6564d013b06cddcb0c2ea809fda2/tlp-functions.in#L44

Thus resulting in the `tp_smapi` load error in `tlp-stat -b` due to the unsupported hardware.

System string as reported in `tlp-stat -s` is : `System = LENOVO ThinkPad 25 20K70000MX`

Cheers! 🍻

linrunner, 2018-03-01:
Hi, thanks for your hint. Please test the change and report.

exsilium, 2018-03-05:
Thanks, I'll try to test it this evening. Much appreciated on the quick turnaround! 🙇 🍻

exsilium, 2018-03-06:
Seems to work well for me - `tlp-stat -b` now reports correctly:

```
+++ ThinkPad Battery Features
tp-smapi   = inactive (unsupported hardware)
tpacpi-bat = active
```
Thanks!

linrunner, 2019-03-11:
Released with TLP 1.2.