linrunner/TLP #327: Appdata issues

Mystro256, 2018-02-27:
Taking a quick look, it looks like the name of the appdata xml is incorrect according to the documentation I could find:

https://www.freedesktop.org/software/appstream/docs/chap-Quickstart.html

I assume it should actually be *..appdata.xml not *.metadata.xml.

Perhaps this can explain an issue that a Fedora user reported. I haven't had time to look into this closer.
https://bugzilla.redhat.com/show_bug.cgi?id=1549430

Correct me if I'm wrong.

linrunner, 2018-02-28:
Hi,

could you ask the user if it works with the renamed file? I'll check with the author of the change in the meantime.

Mystro256, 2018-02-28:
Ok noted thanks, I'll try it on a local machine I have too

linrunner, 2018-02-28:
_The file author says:_

The documentation says otherwise:

https://www.freedesktop.org/software/appstream/docs/chap-Metadata.html#sect-Metadata-GenericComponent

⁠2.1.2. Filesystem locations: "Upstream projects can ship one or more
metainfo files in /usr/share/metainfo/%{id}.metainfo.xml, where id is a
unique identifier of this specific component."

and

"Applications are a special case here, because they are usually treated
differently by software centers (and also for historical reasons). If
your metainfo file contains an application, as described in Section 2.2,
“Desktop Applications”, you may want to install it as
/usr/share/metainfo/%{id}.appdata.xml."

From what I understand, appdata.xml is for GUI applications.

Also, still in 2.1.2, /usr/share/metainfo is indeed the right path to
put AppStream data in, /usr/share/appdata is actually the legacy path
(contradictory to what comment #1 says in the bug report).

Also confirmed here:

https://www.freedesktop.org/software/appstream/docs/sect-Metadata-ConsoleApplication.html

⁠2.3.1. Introduction: "In order to enhance the available metadata about
their application, projects shipping a console application can ship one
or more metainfo files in /usr/share/metainfo/%{id}.metainfo.xml."

Regarding the "bug" mentioned earlier, it seems that GNOME Software
hides non-GUI applications (knowing the GNOME "mentality", I'm pretty
sure it's a deliberate choice). Example : AppStream includes a CLI
command (appstreamcli), and provides a file
"/usr/share/metainfo/org.freedesktop.appstream.cli.metainfo.xml"
describing it. Searching for "AppStream" in GNOME software returns nothing.

I think there's no bug here, and that our AppStream file is fine.

Regards,

Mystro256, 2018-02-28:
Ok, sounds good.

I was thinking that myself about GNOME Software but I wasn't sure. It seems you are correct in that appdata is used for GUI applications (or at least used for any application including a *.desktop file if I understand correctly).

I'll close this and report back.

Thanks!

closed 2018-02-28 by Mystro256