linrunner/TLP #329: Not able to set charge thresholds in T470 fedora 27

AkshayIyer12, 2018-03-19:
### Symptom data

1. Does the problem occur on battery or AC or both? both
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
```
[darkJedi@localhost libmap]$ sudo tlp-stat 
--- TLP 1.1 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
TLP_PERSISTENT_DEFAULT=0
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_HWP_ON_AC=balance_performance
CPU_HWP_ON_BAT=balance_power
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=power
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
AHCI_RUNTIME_PM_TIMEOUT=15
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=on
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_AC=0
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
USB_AUTOSUSPEND=1
USB_BLACKLIST_BTUSB=0
USB_BLACKLIST_PHONE=0
USB_BLACKLIST_PRINTER=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = LENOVO ThinkPad T470 20HEA02BIG
BIOS           = N1QET67W (1.42 )
Release        = "Fedora release 27 (Twenty Seven)"
Kernel         = 4.15.8-300.fc27.x86_64 #1 SMP Fri Mar 9 18:11:36 UTC 2018 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-4.15.8-300.fc27.x86_64 root=/dev/mapper/fedora-root ro rd.lvm.lv=fedora/root rd.lvm.lv=fedora/swap rhgb quiet LANG=en_US.UTF-8
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
Last run       = 03:14:42 PM,    338 sec(s) ago
Mode           = AC
Power source   = AC

[... 167 lines cut ...]
/sys/class/power_supply/BAT0/cycle_count                    =     36
/sys/class/power_supply/BAT0/energy_full_design             =  24000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  24610 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  24610 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Full

Charge                                                      =  100.0 [%]
Capacity                                                    =  102.5 [%]

+++ Battery Status
/sys/class/power_supply/BAT1/manufacturer                   = Celxpert
/sys/class/power_supply/BAT1/model_name                     = 01AV424
/sys/class/power_supply/BAT1/cycle_count                    =     25
/sys/class/power_supply/BAT1/energy_full_design             =  24050 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  22900 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  22780 [mWh]
/sys/class/power_supply/BAT1/power_now                      =   2274 [mW]
/sys/class/power_supply/BAT1/status                         = Charging

Charge                                                      =   99.5 [%]
Capacity                                                    =   95.2 [%]

+++ Charge total                                            =   99.7 [%]
```

linrunner, 2018-03-25:
The necessary kernel module cannot be loaded:

> tpacpi-bat = inactive (kernel module 'acpi_call' load error)

Try to load the module by hand and check the error messages. Power off and reboot. Check again.

Eventually uninstall and reinstall **akmod-acpi_call**, check the messages and the build log.

theLastOfCats, 2018-03-30:
I think problem is in Linux kernel, especially in kernel modules. There are no `akmod-acpi_call` kernel module for Linux v4.15.10 and newer

closed 2018-05-22 by linrunner