linrunner/TLP #331: systemd-udevd[396]: Process '/usr/bin/tlp auto' failed with exit code 4

FadeMind, 2018-03-27:
Always after fresh boot I have some mess in journal:
```
systemd-udevd[396]: Process '/usr/bin/tlp auto' failed with exit code 4
```
Anyway
```
[fademind@manjaro ~]$ systemctl status tlp -l
● tlp.service - TLP system startup/shutdown
   Loaded: loaded (/usr/lib/systemd/system/tlp.service; enabled; vendor preset: disabled)
   Active: active (exited) since Tue 2018-03-27 12:22:59 CEST; 7min ago
     Docs: http://linrunner.de/tlp
  Process: 657 ExecStart=/usr/bin/tlp init start (code=exited, status=0/SUCCESS)
 Main PID: 657 (code=exited, status=0/SUCCESS)
```

```
systemd 238.51-1.1
linux 4.16 (linux416 4.16.r180325.g3eb2ce8-1)
tlp-git (r57.216a77d-1)
```

linrunner, 2018-04-03:
Hi, what exactly do you mean by "mess"? Someone stepped on the easter eggs? ;-)

See http://linrunner.de/en/tlp/docs/tlp-faq.html#exitcode4

closed 2018-04-03 by FadeMind

linrunner, 2019-03-11:
Released with TLP 1.2.