linrunner/TLP #332: Parse error when using using zsh as default sh.

brentlintner, 2018-03-31:
First off, thanks for the awesome project.

### Symptom data

1. Does the problem occur on battery or AC or both?

Both.

2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.

Battery: https://gist.github.com/brentlintner/1b745e6ccb4e63ec3da5424a2b4083e5
AC: https://gist.github.com/brentlintner/f2e6af5a8a44e67226a0d47e58adafa8

### Expected behavior

I can use the `tlp` command.

### Actual behavior

Results in a parse error:

```sh
$ /usr/bin/tlp diskid 
/usr/bin/tlp:348: parse error near `}'
```

### Steps to reproduce the problem

In my OS's case:

1. `ln -s /usr/bin/zsh /usr/bin/sh`
2. `tlp diskid`

### Workarounds

1. Switch `sh` to point to `bash`.
2. Moving the `}` on [348](https://github.com/linrunner/TLP/blob/master/tlp.in#L348) to the next line works with both zsh/bash.

linrunner, 2018-04-03:
Done. Thanks for your report.

closed 2018-04-03 by brentlintner

linrunner, 2019-03-11:
Released with TLP 1.2.