linrunner/TLP #334: btusb autosuspend is not disabled by TLP with kernel 4.16.0

demokritos, 2018-04-11:
### Symptom data
1. Does the problem occur on battery or AC or both? AC (not tested on battery)
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
https://gist.github.com/demokritos/decd97e7b1a364a07b3506575eb6f7bc

### Expected behavior
Bluetooth device(btusb) autosuspend is disabled.

### Actual behavior
Bluetooth device(btusb) autosuspend is not disabled by TLP.

### Steps to reproduce the problem
Upgrade Linux kernel version to 4.16.0.
reboot

linrunner, 2018-04-12:
Please try with

    USB_BLACKLIST="8087:07dc"

demokritos, 2018-04-12:
I already tried that, but I got the same result. Autosuspend was not
disabled for the device.

linrunner, 2018-04-13:
I see. Please provide a [full trace](http://linrunner.de/en/tlp/docs/tlp-configuration.html#tracemode) from system startup only.

demokritos, 2018-04-13:
Here is a full trace.
https://gist.github.com/demokritos/4451f4c41d44423182732f3e9ff62693

linrunner, 2018-04-13:
The device seems to be recognized and handled properly by TLP

     L.1028 19:17:56 tlp[977]: udev_usb.on_btusb_black: Bus 1 Dev 4 ID 8087:07dc /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5

I can only guess something else changes to "auto" afterwards. Maybe another tool or udev rule you deployed?

demokritos, 2018-04-13:
With kernel 4.15.15 and below, I don't have this issue.

Very weird.
I have tried turning on udev debug log in /etc/udev/udev.conf.
If udev debug log is on, the log line (tlp[977]: udev_usb.on_btusb_black:
Bus 1 Dev 4 ID 8087:07dc /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5
) does not appear.
Is it possible that tlp's udev rule is not applied but that log message is
printed?
I don't have my own udev rules.

linrunner, 2018-04-17:
In fact the log entry does not reflect whether the write was successful ([see code](https://github.com/linrunner/TLP/blob/master/tlp-usb-udev.in#L237)).

BUT: an error in writing would be due to the kernel.

makitso, 2018-05-17:
I have two 18.04 laptops, one with ubuntu and one with ubuntu budgie. Both were using TLP and the 4.16 kernel. Both laptops were experiencing strange battery problems e.g. not being fully charged while in suspend, etc. So, I reverted back to the standard 4.15 kernel and now I no longer see the problem.

makitso, 2018-05-17:
I picked this item up from the tlp github site - laptop not charging while in AC on deep sleep. 

https://github.com/linrunner/TLP/issues/344

The code in /etc/default/tlp/tlp 
# PCI Express Active State Power Management (PCIe ASPM):
#   default, performance, powersave.
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave

changed to 

# PCI Express Active State Power Management (PCIe ASPM):
#   default, performance, powersave.
# PCIE_ASPM_ON_AC=default
# PCIE_ASPM_ON_BAT=default

linrunner, 2018-05-17:
@makitso : I don't understand how your remark about ASPM is related to this issue?

lazycrypt, 2018-05-21:
I'm having the same problem on Arch. After a while bluetooth devices will lose connection and if I check bluetoothctl the bluetooth adapter ends up being stuck in a constant on/off loop until I manually turn it off and on again.

Though the intervals at which this happens seem to be entirely random? Sometimes it will happen shortly after starting up but then the next time there wouldn't be any issues the whole session.

linrunner, 2018-05-22:
@demokritos : any news?

demokritos, 2018-05-22:
No. The issue is still there. I have worked around the issue in the way arjendekorte mentioned.

lazycrypt, 2018-05-25:
Can't say either using `tlp usb` or the above mentioned conf work for me. Currently on the `4.16.11-1-ARCH` kernel.

arjendekorte, 2018-05-26:
@lazycrypt: What does the output of
`tlp stat -u`
show?

lazycrypt, 2018-05-26:
So

here is what I get with just `USB_BLACKLIST_BTUSB=1`:
```
--- TLP 1.1 --------------------------------------------

+++ USB
Autosuspend         = enabled
Device whitelist    = (not configured)
Device blacklist    = (not configured)
Bluetooth blacklist = enabled
Phone blacklist     = disabled
WWAN blacklist      = enabled

Bus 006 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 006 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 005 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 004 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 005 ID 04f2:b2ea control = auto, autosuspend_delay_ms =  2000 -- Chicony Electronics Co., Ltd Integrated Camera [ThinkPad] (uvcvideo)
Bus 001 Device 004 ID 0a5c:21e6 control = auto, autosuspend_delay_ms =  2000 -- Broadcom Corp. BCM20702 Bluetooth 4.0 [ThinkPad] (btusb)
Bus 001 Device 003 ID 147e:2020 control = auto, autosuspend_delay_ms =  2000 -- Upek TouchChip Fingerprint Coprocessor (WBF advanced mode) (no driver)
Bus 001 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 003 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
```

And here is what I get if I also add the Bluetooth adapter to `USB_BLACKLIST`:
```
--- TLP 1.1 --------------------------------------------

+++ USB
Autosuspend         = enabled
Device whitelist    = (not configured)
Device blacklist    = 0a5c:21e6
Bluetooth blacklist = enabled
Phone blacklist     = disabled
WWAN blacklist      = enabled

Bus 006 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 006 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 005 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 004 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 005 ID 04f2:b2ea control = auto, autosuspend_delay_ms =  2000 -- Chicony Electronics Co., Ltd Integrated Camera [ThinkPad] (uvcvideo)
Bus 001 Device 004 ID 0a5c:21e6 control = on,   autosuspend_delay_ms =  2000 -- Broadcom Corp. BCM20702 Bluetooth 4.0 [ThinkPad] (btusb)
Bus 001 Device 003 ID 147e:2020 control = auto, autosuspend_delay_ms =  2000 -- Upek TouchChip Fingerprint Coprocessor (WBF advanced mode) (no driver)
Bus 001 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 003 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
```

Now with just `USB_BLACKLIST_BTUSB=1` I will always get random disconnects, even a minute after logging in. If I also add the adapter to the blacklist it seems to be more stable at the moment? I remember having regular disconnects with that as well but I'll test it out some more.

lazycrypt, 2018-05-29:
So it seems manually blacklisting the Bluetooth adapter frpm autosuspending with the combinanation of what arjendekorte posted my mouse and gamepad no longer lose connections at random.

There really must be something up with the kernel.

closed 2019-04-04 by linrunner

Feichtmeier, 2019-04-27:
@linrunner I am having the exact same issue with my dell xps 13 9380 and kernel 4.18.0-18-generic installed via 
```bash
sudo apt install --install-recommends linux-generic-hwe-18.04 xserver-xorg-hwe-18.04
```

Using
```bash
sudo tlp usb
```
fixes it until next reboot

PaulVanSchayck, 2020-06-22:
I'm having also the exact same issues now on Ubuntu 20.04 (5.4.0-7634-generic). Should I open a new issue?

ValdikSS, 2020-12-15:
Guess TLP should re-enable USB autosuspend rules upon suspend
https://github.com/linrunner/TLP/issues/545

linrunner, 2020-12-16:
On my ThinkPad X220 with Ubuntu 20.04, Kernel 5.4  i see  in TLP's trace during boot:

> usb_suspend_device.udev.**on**_btusb_deny.**no_change**: Bus 1 Dev 5 ID 0a5c:217f /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4 []

But after boot the state has changed to `auto`:
> Bus 001 Device 005 ID 0a5c:217f control = **auto**, autosuspend_delay_ms = 2000 -- Broadcom Corp. BCM2045B (BDC-2.1) (btusb)

This means that:

1. TLP finds the device in status `on` and leaves it like this because of `USB_BLACKLIST_BTUSB=1`
2. Something else changes to **auto** _afterwards_

I have no idea what "something" is and suggest you look for it together, because the problem cannot be solved effectively in TLP without knowing the problem source, if at all.

linrunner, 2020-12-16:
ps. of course one can try to change to batch mode on boot with:

     X_TLP_USB_MODE=1

But that doesn't solve the possibly related problem with suspend/resume #545.