linrunner/TLP #338: Setting lower discharge limit on dual battery systems impossible

tbart, 2018-04-24:
I have a dual battery system, see https://gist.github.com/tbart/2f5b8daaaf5d8df0d6299b8ea648ab6f for a tlp-stat output.
Setting the charge thresholds works as expected, but I'd also like to set the *discharge* thresholds.
That is, the second battery only gets discharged once the first one is at 4%. That's not really the way I'd expect it to be as it unnecessarily stresses the first battery.

It would be nice to have the first battery discharge until START_CHARGE_THRESH_BAT0 and then discharge the second one. That way both batteries mostly operate in the middle range between the limits set. I am not really sure what should happen after both batteries hit their lower limit, but getting to this point would already be a great achievement.
As TLP is no persistent service it might be tricky to do the following (but in the mean time, cron would probably do), but would it be possible to switch discharging to the fuller battery every few minutes? Or just discharge both batteries at the same time?

Is it technically possible to force the second battery to discharge?
I can see "tpacpi-bat.BAT1.forceDischarge" which sounds promising, but can TLP control this flag? Or do I need to use the tpacpi-bat perl tool for it?

Thanks in advance!

linrunner, 2018-04-25:
Hi, the lenovo firmware offers no way to do this.

closed 2018-04-25 by linrunner

linrunner, 2018-04-25:
Reason added to the FAQ.

bhack, 2018-05-04:
@linrunner "newer ThinkPad models may not need charge thresholds due to dualmode battery firmware" what it really mean for tlp use?