linrunner/TLP #339: tlp blocks suspend on ubuntu 18.04

damageboy, 2018-04-25:
### Symptom data
When I attempt suspend after installing TLP I **sometimes** get:

```
[Wed Apr 25 18:26:24 2018] Freezing of tasks failed after 20.006 seconds (1 tasks refusing to freeze, wq_busy=0):
[Wed Apr 25 18:26:24 2018] tlp             D    0 18109      1 0x00000104
[Wed Apr 25 18:26:24 2018] Call Trace:
[Wed Apr 25 18:26:24 2018]  __schedule+0x297/0x8b0
[Wed Apr 25 18:26:24 2018]  schedule+0x2c/0x80
[Wed Apr 25 18:26:24 2018]  schedule_preempt_disabled+0xe/0x10
[Wed Apr 25 18:26:24 2018]  __mutex_lock.isra.2+0x18c/0x4d0
[Wed Apr 25 18:26:24 2018]  __mutex_lock_slowpath+0x13/0x20
[Wed Apr 25 18:26:24 2018]  ? __mutex_lock_slowpath+0x13/0x20
[Wed Apr 25 18:26:24 2018]  mutex_lock+0x2f/0x40
[Wed Apr 25 18:26:24 2018]  control_store+0x4b/0xd0
[Wed Apr 25 18:26:24 2018]  dev_attr_store+0x1b/0x30
[Wed Apr 25 18:26:24 2018]  sysfs_kf_write+0x3c/0x50
[Wed Apr 25 18:26:24 2018]  kernfs_fop_write+0x125/0x1a0
[Wed Apr 25 18:26:24 2018]  __vfs_write+0x1b/0x40
[Wed Apr 25 18:26:24 2018]  vfs_write+0xb1/0x1a0
[Wed Apr 25 18:26:24 2018]  SyS_write+0x55/0xc0
[Wed Apr 25 18:26:24 2018]  do_syscall_64+0x73/0x130
[Wed Apr 25 18:26:24 2018]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
[Wed Apr 25 18:26:24 2018] RIP: 0033:0x7fc12b5c3154
[Wed Apr 25 18:26:24 2018] RSP: 002b:00007ffd6aa96ef8 EFLAGS: 00000246 ORIG_RAX: 0000000000000001
[Wed Apr 25 18:26:24 2018] RAX: ffffffffffffffda RBX: 0000564ec38a6cd0 RCX: 00007fc12b5c3154
[Wed Apr 25 18:26:24 2018] RDX: 0000000000000005 RSI: 0000564ec38a6cd0 RDI: 0000000000000001
[Wed Apr 25 18:26:24 2018] RBP: 0000000000000005 R08: 0000564ec38aa188 R09: 0000000000000004
[Wed Apr 25 18:26:24 2018] R10: 00000000fffffffc R11: 0000000000000246 R12: 0000000000000001
[Wed Apr 25 18:26:24 2018] R13: 0000000000000005 R14: 7fffffffffffffff R15: 0000564ec35550a5
[Wed Apr 25 18:26:24 2018] OOM killer enabled.
[Wed Apr 25 18:26:24 2018] Restarting tasks ... done.
```

### Expected behavior
Suspend should work

### Actual behavior
Suspend doesn't work

### Steps to reproduce the problem
Install tlp. at some point this starts happening (after a couple of suspend/resume cycles)

damageboy, 2018-04-25:
FWIW
I can't kill neither normally with SIGTERM nor with SIGKILL

For now I guess I will disable it from startup

damageboy, 2018-04-25:
I've update the original issue report with the link to the output of tlp-stat...

linrunner, 2018-05-07:
Your output shows that TLP is not run upon system start and during suspend / resume (tlp.service and tlp-sleep.service disabled). I guess this was for testing purposes.

Any result from implementing the troubleshooting guide?

rathboma, 2018-06-05:
I'm also having this issue. It is inconsistent and only happens sometimes. I've only tested it on battery mode as that is when I typically sleep my laptop.

Here is my tlp-stat:

https://gist.github.com/rathboma/e988bbed0735bdcbdfd15b04e1352e0f

I've disabled many of the power management features of my laptop already. In particular PCIe which was causing issues with missing peripherals when reconnecting to a dock.

I'll try to work through a bit of the troubleshooting guide, any ideas in the meantime?

rathboma, 2018-06-06:
I believe this may be the culprit: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1768852

@damageboy do you use a thunderbolt dock?

rathboma, 2018-06-11:
The thunderbolt dock fix just got released today and my TLP issues are now gone. TLP refusing to exit wasn't really the culprit, rather a misleading error message related to this thunderbolt undocking kernel panic.

@damageboy I'd recommend a software update today and see if that helps you at all?

closed 2018-07-14 by linrunner