linrunner/TLP #340: TLP creates various problems at intel network adapter drivers

xzj8b3, 2018-05-02:
TLP causes problems and blocks to intel network adapter drivers

In Fedora 28 the problem occurred when you reboot after installing the module, but the problem is not resolved even by deleting the form!
Updated that module or suggested how to reset its settings at defaulth is quite a many users are going crazy to understand the cause of problems!

Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller driver: r8169
Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter driver: ath9k

https://serverfault.com/questions/616485/e1000e-reset-adapter-unexpectedly-detected-hardware-unit-hang?newreg=14f08e6d4ecf415e9d288d5fae2c485b

Informazioni addizionali:
giornalista: libreport-2.9.3
cmdline: BOOT_IMAGE = / vmlinuz-4.15.10-300.fc27.x86_64 root = / dev / mapper / fedora_x555lp-root ro rd.lvm.lv = fedora_x555lp / root rd.lvm.lv = fedora_x555lp / swap rhgb quiet LANG = it_IT .UTF-8
crash_function: dev_graft_qdisc
kernel: 4.15.10-300.fc27.x86_64
runlevel: N 5
tipo: Kerneloops

Backtrace troncato:
NETDEV WATCHDOG: enp2s0 (r8169): coda di trasmissione 0 scaduta
ATTENZIONE: CPU: 0 PID: 0 alla rete / sched / sch_generic.c: 323 dev_watchdog + 0x1f3 / 0x200
Moduli collegati in: rfcomm fusibile cmac nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 xt_conntrack ip_set nfnetlink ebtable_nat ebtable_broute ponte stp llc ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack libcrc32c iptable_mangle iptable_raw iptable_security ebtable_filter ebtables ip6table_filter ip6_tables bnep sunrpc grasso vfat ath3k btusb uvcvideo btrtl btbcm btintel rtsx_usb_ms bluetooth videobuf2_vmalloc memstick videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media ecdh_generic arc4 ath9k intel_rapl ath9k_common x86_pkg_temp_thermal ath9k_hw intel_powerclamp coretemp kvm_intel mac80211 kvm iTCO_wdt
 iTCO_vendor_support asus_nb_wmi asus_wmi sparse_keymap irqbypass crct10dif_pclmul crc32_pclmul ATH ghash_clmulni_intel intel_cstate cfg80211 intel_uncore snd_hda_codec_realtek snd_hda_codec_hdmi intel_rapl_perf snd_hda_codec_generic snd_hda_intel snd_hda_codec joydev snd_hda_core snd_hwdep snd_seq i2c_i801 snd_seq_device intel_pch_thermal snd_pcm rfkill lpc_ich snd_timer snd mei_me wmı mei soundcore shpchp processor_thermal_device intel_soc_dts_iosf int3400_thermal int3402_thermal acpi_thermal_rel asus_wireless int340x_thermal_zone amdgpu chash rtsx_usb_sdmmc mmc_core rtsx_usb radeon i915 i2c_algo_bit drm_kms_helper crc32c_intel ttm drm serio_raw r8169 mii video
CPU: 0 PID: 0 Comm: swapper / 0 Non contaminato 4.15.10-300.fc27.x86_64 # 1
Nome hardware: ASUSTeK COMPUTER INC. X555LP / X555LP, BIOS X555LP.203 11/06/2014
RIP: 0010: dev_watchdog + 0x1f3 / 0x200
RSP: 0018: ffff91db2ee03e90 EFLAGS: 00010282
RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000 RCX: 0000000000000000
RDX: 0000000000040400 RSI: 00000000000000f6 RDI: 0000000000000300
RBP: ffff91db256d4478 R08: 0000000000000412 R09: 0000000000000004
R10: ffff91db2ee03d00 R11: 0000000000000001 R12: ffff91db256d4000
R13: 0000000000000000 R14: ffff91db2ee03ee0 R15: 0000000000000000
Vendo: 0000000000000000 (0000) GS: ffff91db2ee00000 (0000) knlGS: 0000000000000000
CS: 0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: 000055fcd3f15130 CR3: 00000001fa20a005 CR4: 00000000001606f0
Call Trace:
 <IRQ>
 ? dev_graft_qdisc + 0x70 / 0x70
 call_timer_fn + 0x2B / 0x120
 run_timer_softirq + 0x3cc / 0x420
 ? tick_sched_handle + 0x22 / 0x70
 ? sched_clock + 0x5 / 0x10
 __do_softirq + 0xe7 / 0x2cb
 irq_exit + 0xf1 / 0x100
 smp_apic_timer_interrupt + 0x6C / 0x120
 apic_timer_interrupt + 0x87 / 0x90
 </ IRQ>
RIP: 0010: cpuidle_enter_state + 0xa2 / 0x2e0
RSP: 0018: ffffffff95203e88 EFLAGS: 00000246 ORIG_RAX: ffffffffffffff11
RAX: ffff91db2ee20a80 RBX: 00000010cd0233ff RCX: 000000000000001f
RDX: 00000010cd0233ff RSI: fffffffea9afe7fe RDI: 0000000000000000
RBP: ffff91db2ee2a400 R08: 0000000000000941 R09: 0000000000000f63
R10: ffffffff95203e68 R11: 0000000000000336 R12: 0000000000000007
R13: ffffffff952d1558 R14: 0000000000000000 R15: 00000010cc1e1437
 ? cpuidle_enter_state + 0x92 / 0x2e0
 do_idle + 0x17b / 0x1d0
 cpu_startup_entry + 0x6F / 0x80
 start_kernel + 0x4e0 / 0x500
 secondary_startup_64 + 0xA5 / 0XB0
Codice: 48 00 48 63 4d e8 eb 93 4c 89 e7 c6 05 b4 31 bb 00 01 e8 21 50 fd 89 d9 48 89 c2 4c 89 e6 48 c7 c7 58 f6 13 95 e8 ed 78 92 ff <0f> 0b eb c0 66 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 48 c7 47

linrunner, 2018-05-03:
Hi, 

I do not understand what you're trying to say. In fact you *don't* have *intel* network adapters. I guess this is a kernel issue.

I suggest you read [TLP Troubleshooting](http://linrunner.de/en/tlp/docs/tlp-troubleshooting.html) and try to blacklist the ethernet adapter (r8169). 

Btw: no further support without the required tlp-stat outputs.

xzj8b3, 2018-05-03:
The problem of blocking the drivers is caused by TLP.... and it occurred
again just reinstalled the module after a clean installation of Fedora
28... ...

As long as the module was not present error did not show up.... just
installed it appeared....

I have reinstalled notebook without TLP.... No problem reported!!!!

Create TLP problems

Seguimi su Facebook <https://www.facebook.com/IvanDeFelice>  Seguimi su
Twitter <https://twitter.com/xzj8b3>  Seguimi su Istagram
<https://www.instagram.com/defelice.ivan/> Seguimi su Pinterest
<https://it.pinterest.com/xzj8b3/>

                            Seguimi su Google +
<https://plus.google.com/+IvanDeFelice>     Seguimi su Flipboard
<https://flipboard.com/profile>

rathann, 2018-05-07:
The issue title seems to be inaccurate, but the problem looks real.

linrunner, 2018-05-07:
@xzj8b3 : did you follow the troubleshooting guide and excluded (blacklisted) the ethernet adapter from runtime pm?

xzj8b3, 2018-05-08:
From Fedora 24 to 27 to create those problems is TLP

It did even in Fedora 28 and immediately after the installation of TLP the
problem was born at the next reboot

In doubt I reinstalled without TLP.... problem disappeared

Generates too many problems that form!!!!

Seguimi su Facebook <https://www.facebook.com/IvanDeFelice>  Seguimi su
Twitter <https://twitter.com/xzj8b3>  Seguimi su Istagram
<https://www.instagram.com/defelice.ivan/> Seguimi su Pinterest
<https://it.pinterest.com/xzj8b3/>

                            Seguimi su Google +
<https://plus.google.com/+IvanDeFelice>     Seguimi su Flipboard
<https://flipboard.com/profile>

closed 2018-05-22 by linrunner