linrunner/TLP #344: Default TLP configuration prevents entering deeper sleep states on AC

superm1, 2018-05-15:
### Symptom data

1. Does the problem occur on battery or AC or both?
AC

### Expected behavior
CPU can enter into C6 or lower on both battery and AC.

### Actual behavior
CPU can't enter past C5 on AC.

### Steps to reproduce the problem
* Look at powertop/turbostat output on AC.
* Remove power plug.
* Look at powertop/turbostat output on battery.

As part of enabling better support for suspend-to-idle one of the issues that we identified was that when using TLP the default configuration will override the kernel's ASPM policies (which for most distributions they follow the "default" ASPM policy (IE keep BIOS ASPM defaults).

TLP however configures on AC to change this.
```
# PCI Express Active State Power Management (PCIe ASPM):
#   default, performance, powersave.
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
```

This leads to higher power consumption on AC, slower charging.  On some CPU packages it can (perhaps counterintuitively?) lead to worse performance because the package will overheat sooner and need to throttle.

I'd like to propose that for your defaults you should set both of those to the word `default` and comment them out.  If someone knows they want to change the performance or `powersave `specifically they can.  This will allow the general purpose scenarios to work more correctly.

linrunner, 2018-05-15:
OK, thanks for your hint.

CottonEaster, 2018-10-02:
It seems to me that the default TLP config file should contain a note about this matter, i.e. to the effect that using the 'performance' setting will disable deep sleep.

linrunner, 2018-10-02:
Not in the file but in the docs. Having every detail in the file makes it confusing.

Besides, isn't it obvious that "performance" may lead to sleep disorders? ;-)

CottonEaster, 2018-10-02:
> Besides, isn't it obvious that "performance" may lead to sleep disorders? ;-)

It was not obvious to me. But, still, if the 'sleep disorder' is no longer the default . . actually, no; I do think you'll have people turning on the performance mode and being suprised that what they regard as separate, basic functionality - sleep - doesn't work anymore. But, still, it's up to you, of course.

koponomarenko, 2019-01-30:
I second that. I would/want expect sleep to be a separate functionality.

linrunner, 2019-03-11:
Sorry guys for not meeting your expectations, but the kernel interface – e.g. /sys/module/pcie_aspm/parameters/policy – offers no possibility of influencing the sleep states directly. Btw the same is true for all other power saving sysfiles. You have to deal with it.

For good measure I added a note to the [config docs](https://linrunner.de/en/tlp/docs/tlp-configuration.html#aspm).

closed 2019-03-11 by linrunner