linrunner/TLP #345: Different profiles for different power supplies?

jbaiter, 2018-05-16:
I have a Thinkpad T480s which uses USB-C for charging. Depending on the circumstances, I either use a 65W (at home), 45W (on the road) or 30W (powerbank) power supply.
It would be great if I could declare a profile for each of these power supplies and adjust CPU frequencies and charging thresholds (e.g. no charging on powerbank) with TLP.

On my T480s, there is an ACPI method to read the wattage the USB-C power supply declares: `"\_SB.PCI0.LPCB.EC.HWAT`, other USB-C powered Thinkpads should have something similar, since it's also displayed in Lenovo Vantage on Windows.

linrunner, 2018-05-18:
Hi,

your proposal for additional profiles has advantages but also a major downside:
* Configuration (and related documentation)  would grow significantly, i.e. SETTING_ON_BAT, SETTING_ON_AC1, SETTING_ON_AC2, SETTING_ON_AC3, probably confusing some users
* Extending the configuration like that will break third party tools like TLPUI
* Code would grow strongly to reflect the many case distinctions

I guess the core of your requirement is to ignore specific power supplies (e.g. the powerbank) and apply BAT settings for them.

A simpler approach could be to have a POWER_SUPPLY_BLACKLIST.

linrunner, 2018-05-18:
ps. I won't implement something like "stop charging". That would be in the same category as "battery balancing", see [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#featuresexcluded).

jbaiter, 2018-05-18:
Thank you for the feedback! You're right about the `POWER_SUPPLY_BLACKLIST`, this would basically solve the use case. But I guess that's a pretty narrow use case to introduce a new configuration variable that's not exactly self-explanatory.

I implemented the behavior now with a small script that  puts TLP into battery mode and sets the powerbank-specific charging thresholds, triggered by an udev rule.

closed 2018-05-18 by jbaiter