linrunner/TLP #347: Thunderbolt controller disappears after enabling TLP

Venemo, 2018-05-25:
### Symptom data

After enabling TLP, the Thunderbolt controller in my Dell XPS 13 9370 disappers from the output of `lspci`.

1. Does the problem occur on battery or AC or both?  
Both
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.  
https://gist.github.com/Venemo/60c9cdcca18b9b5e5ff96c4ed41a6ecc (taken after enabling TLP)

### Expected behavior

I would expect the Thunderbolt controller to stay there.

### Actual behavior

The Thunderbolt controller disappears. I do not have an actual Thunderbolt peripheral yet, so could not try that, but the ports can still be used for charging and connecting regular USB peripherals.

### Steps to reproduce the problem

I'm running on Fedora 28. Here is what I did:

```
[root@timur-xps ~]# lspci | grep Thunder
03:00.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
04:00.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
04:01.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
04:02.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
04:04.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
05:00.0 System peripheral: Intel Corporation JHL6540 Thunderbolt 3 NHI (C step) [Alpine Ridge 4C 2016] (rev 02)
39:00.0 USB controller: Intel Corporation JHL6540 Thunderbolt 3 USB Controller (C step) [Alpine Ridge 4C 2016] (rev 02)
[root@timur-xps ~]# systemctl enable tlp
Created symlink /etc/systemd/system/multi-user.target.wants/tlp.service → /usr/lib/systemd/system/tlp.service.
[root@timur-xps ~]# systemctl start tlp
[root@timur-xps ~]# lspci | grep Thunder
[root@timur-xps ~]# 
```

linrunner, 2018-05-26:
I suggest to blacklist the controller, see http://linrunner.de/en/tlp/docs/tlp-troubleshooting.html

Venemo, 2018-05-26:
@linrunner Thanks! In the meantime I have determined that this is probably **not** a TLP issue. Not 100% sure, though. The Thunderbolt controller disappears from `lspci` shortly after boot, even when TLP is disabled.

Venemo, 2018-05-30:
It seems that the Thunderbolt hardware is simply shut down after login (with or without tlp), there are even some journal messages about it:

```
[   45.251062] thunderbolt 0000:05:00.0: stopping RX ring 0
[   45.251067] thunderbolt 0000:05:00.0: disabling interrupt at register 0x38200 bit 12 (0xffffffff -> 0xffffefff)
[   45.251074] thunderbolt 0000:05:00.0: stopping TX ring 0
[   45.251078] thunderbolt 0000:05:00.0: disabling interrupt at register 0x38200 bit 0 (0xffffffff -> 0xfffffffe)
[   45.251081] thunderbolt 0000:05:00.0: control channel stopped
[   45.251131] thunderbolt 0000:05:00.0: freeing RX ring 0
[   45.251136] thunderbolt 0000:05:00.0: freeing TX ring 0
[   45.251142] thunderbolt 0000:05:00.0: shutdown
```

For now, I don't have any actual Thunderbolt device to test with, but I'm 99.9% sure this has nothing to do with tlp. Sorry for the noise. I'll close this issue now.

closed 2018-05-30 by Venemo

linrunner, 2018-05-31:
Thanks for your feedback