linrunner/TLP #349: Mode enforcing does not hold

jtourille, 2018-05-30:
With power supply disconnected (thus on battery mode), if I force the AC mode via `sudo tlp ac`, TLP switches back automatically to the BAT mode within roughly 30 seconds.
I am running Ubuntu 18.04. I am a TLP user for many years now, installing it on Ubuntu 14.04 and 16.04. I have never seen something like that. I am clueless, maybe something is interfering with tlp?

Here is my tlp-stat output: https://gist.github.com/jtourille/52645f5fe7cf3951f0643197061a9e22

linrunner, 2018-05-31:
Hi,

in fact `tlp bat` is not designed as an *enforcement*. 

I guess your laptop issues battery udev events every ~30 secs that switch tlp back to battery mode. You can look for them with

    tlp --pev

To have AC settings all the time you could change your settings to

    TLP_DEFAULT_MODE=AC
    TLP_PERSISTENT_DEFAULT=1

but this may not fit your use case.

jtourille, 2018-05-31:
That's weird, I used to switch back to AC mode when I wanted to do something CPU-intensive and the command was acting as an enforcement. In fact, it is written in the documentation that `tlp ac` or `tlp bat` _will ignore the actual power source_.
I checked the `tlp-stat --pev` and I can see the udev events, thanks for the tip.

linrunner, 2018-07-29:
Hi, 

I've finally implemented a solution for this. The documentation will call it "manual" mode.

Example: after issuing `tlp bat` on AC power  `tlp-stat -s` will show 

    +++ TLP Status
    Mode           = battery (manual)
    Power source   = AC

Would be nice if you could test on your laptop and report back!

jtourille, 2018-07-30:
I have installed the last version (27a87750ab328598a9744a4b67f11acfe0c8c70c) and the manual mode seems to work on my laptop. This is great, thank you.

linrunner, 2018-07-30:
Glad to hear that. Thanks for your feedback!

linrunner, 2019-03-11:
Released with TLP 1.2.