linrunner/TLP #350: PCIE_ASPM enabled by tlp, instead of manually

waldo99, 2018-06-04:
### Symptom data
This may be unique to certain firmware / models, but I'm finding that one of my devices isn't enabling L0s for ASPM, even though it's capable of it and PCIE_ASPM_ON_BAT=powersave and pcie_aspm=force is set for cmdline boot.

1. Does the problem occur on battery or AC or both?
Bat
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
https://gist.github.com/waldo99/c566e1aec8e33b8df5ee5b4d7ae3b265
### Expected behavior
All capable devices set to L1 L0s for ASPM.
### Actual behaver
One device needs to be forced - As you can see from lspci -vv | grep ASPM:
````
		LnkCap:	Port #6, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <512ns, L1 <16us
			ClockPM- Surprise- LLActRep+ BwNot+ ASPMOptComp-
		LnkCtl:	ASPM L0s L1 Enabled; RCB 64 bytes Disabled- CommClk+
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
		LnkCap:	Port #3, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <512ns, L1 <16us
			ClockPM- Surprise- LLActRep+ BwNot+ ASPMOptComp-
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
		LnkCap:	Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s unlimited, L1 <64us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM L0s L1 Enabled; RCB 64 bytes Disabled- CommClk+
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
		LnkCap:	Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <32us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
```
### Steps to reproduce the problem
Change from/to ac/bat. I fix it using this script: 
http://drvbp1.linux-foundation.org/~mcgrof/scripts/enable-aspm

Could this be incorporated into TLP?

linrunner, 2018-06-04:
Hi, how (W) much does it save?

baybal, 2018-09-03:
Confirming this: I can force all devices to both L0 and L1 myself, but automatic switch only turns on L1.

`baybal@xps ~ $ sudo lspci -vv | grep   'abled;'
Password: 
		LnkCtl:	ASPM L0s L1 Enabled; RCB 64 bytes Disabled- CommClk-
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+`
`

Possibly a kernel issue, and we need to manually switch L0 on every device

Dell 9360 rev 1.2

That should save around 1wh per hour

baybal, 2018-09-03:
@linrunner this is what I am using on Dell 9360 https://gist.github.com/baybal/b499fc5811a7073df0c03ab8da4be904

baybal, 2018-09-11:
That's unlikely as it would involve whitelisting of every device which which supports L0, but don't declare its support. This is better be handled in userspace

baybal, 2018-11-10:
@linrunner what do you think? Kernel devs obviously don't want to touch "as intended" bugged devices that don't advertise L0. My bet, laptop makers intentionally resort to quick and ugly hack of overriding L0 advertisement on some hardware simply because they don't want to deal with driver bugs on windows.

It's a very easy to implement function (see my gist). A single digit number of devices on the market that can't deal with L0 despite PCIE standard requiring it, can be easily blacklisted.

khfeng, 2018-11-12:
> @linrunner what do you think? Kernel devs obviously don't want to touch "as intended" bugged devices that don't advertise L1. My bet, laptop makers intentionally resort to quick and ugly hack of overriding L1 advertisement on some hardware simply because they don't want to deal with driver bugs on windows.

ASPM link state transition happens at hardware level, if ASPM causes bugs on Windows, it will likely cause bugs on Linux.

> 
> My understating how kernel handles it now is: 1. get firmware advertised ASPM states, 2. and apply its own blacklist on top of it. And there is no "whitelist" there to force enable ASPM on bugged firmwares.

For a good reason: ODMs enable or disable ASPM based on their own lab testing results, or suggested by silicon vendors.

> 
> It's a very easy to implement function (see my gist). A single digit number of devices on the market that can't deal with L1 despite PCIE standard requiring it, can be easily blacklisted (or simply reuse one from the kernel).

Instead of doing this, we should understand why your XPS 9360 doesn't have optimal power optimization instead. Let's start from SoC Package C-State.
What's the deepest Package C-State when screen is on and when screen is off, respectively?

baybal, 2018-11-15:
C8 in both modes, individual cores go down to c10

khfeng, 2018-11-16:
When the screen is off, does Package C-State hit C10 if L1 is enabled for all PCI devices?

baybal, 2018-11-16:
No, even when L1 is enabled for all devices

baybal, 2018-11-16:
Tlp-stat output https://gist.github.com/baybal/f948bff092fa9588fc0fb3821b59c0d8

linrunner, 2020-01-20:
This has to be solved in the kernel driver.

closed 2020-01-20 by linrunner