linrunner/TLP #351: Reboots instead of shutdown

Dartan, 2018-06-10:
### Symptom data
I'm on Ubuntu 18.04. After installing TLP (issue also occurs with powertop --auto-tune) my laptop always reboots instead of shutting down. Removing TLP resolves the issue. I've verified that wake-on-lan is disabled in the BIOS.

1. Does the problem occur on battery or AC or both?
Both
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
--- TLP 1.1 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
TLP_PERSISTENT_DEFAULT=0
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_HWP_ON_AC=balance_performance
CPU_HWP_ON_BAT=balance_power
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=power
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
AHCI_RUNTIME_PM_TIMEOUT=15
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=on
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_AC=0
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
USB_AUTOSUSPEND=1
USB_BLACKLIST_BTUSB=0
USB_BLACKLIST_PHONE=0
USB_BLACKLIST_PRINTER=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = Dell Inc. Precision 5510
BIOS           = 1.7.0
Release        = Ubuntu 18.04 LTS
Kernel         = 4.15.0-22-generic #24-Ubuntu SMP Wed May 16 12:15:17 UTC 2018 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-4.15.0-22-generic root=/dev/mapper/ubuntu--vg-root ro quiet splash vt.handoff=1
Init system    = systemd v237
Boot mode      = BIOS (CSM, Legacy)

+++ TLP Status
State          = enabled
Last run       = 10:47:32 AM,    767 sec(s) ago
Mode           = battery
Power source   = battery

[... 190 lines cut ...]

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = DELL 1P6KD5A
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   7368 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   7000 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   6323 [mAh]
/sys/class/power_supply/BAT0/current_now                    =    651 [mA]
/sys/class/power_supply/BAT0/status                         = Discharging

Charge                                                      =   90.3 [%]
Capacity                                                    =   95.0 [%]

+++ Suggestions
* Install smartmontools for disk drive health info

### Expected behavior
Laptop should shutdown.

### Actual behavior
Laptop reboots.

### Steps to reproduce the problem
Attempt to shutdown laptop.

krieghof, 2018-06-10:
I have the same problem reboot instead of shutdown on Ubuntu and Fedora. For me It happened every time on battery but only sometimes on AC. On Solus it did not reboot but the shutdown took over 3 minutes. Now I'm on Manjaro with no problems.
Thinkpad L480

linrunner, 2018-06-11:
Hi, 

there's a [FAQ section](http://linrunner.de/en/tlp/docs/tlp-faq.html#system) on the subject and I suggest you read [TLP Troubleshooting](http://linrunner.de/en/tlp/docs/tlp-troubleshooting.html) to isolate the cause.

linrunner, 2018-06-25:
Please report if your problem was solved and close the issue.

Dartan, 2018-06-25:
Recently got a kernel update which has resolved the issue.

closed 2018-06-25 by Dartan