linrunner/TLP #356: natacpi is unable to lower charge thresholds in most cases

tanriol, 2018-07-16:
### Symptom data

With Linux 4.17.6 and `natacpi` thresholds support, `tlp fullcharge` successfully increases the thresholds, but `tlp setcharge` cannot reduce them back.

```
# rg CHARGE_THRESH /etc/tlp.conf
272:# START_CHARGE_THRESH value and stops when exceeding the STOP_CHARGE_THRESH value.
274:START_CHARGE_THRESH_BAT0=75
275:STOP_CHARGE_THRESH_BAT0=80
# tlp stat -b
[...]
+++ Battery Features
natacpi    = active (data, thresholds)
tpacpi-bat = active (discharge)
tp-smapi   = inactive (ThinkPad not supported)
[...]
/sys/class/power_supply/BAT0/charge_start_threshold         =     75 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     80 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0
# tlp fullcharge
Setting temporary charge thresholds for BAT0:
  stop  = 100
  start = 96
# tlp setcharge
Setting temporary charge thresholds for BAT0:
  stop  => Error: cannot set threshold. Aborting.
```

### Probable reason

The in-kernel threshold implementation rejects new limit unless `start <= stop`. `tlp setcharge` tries to set the stop threshold first, which is rejected as it is below the configured (old) start limit.

linrunner, 2018-07-17:
Thanks for your report :-)

linrunner, 2018-10-04:
Small cause, big effect. A lot of code had to be restructured. Thanks for your patience.

Please test!

tanriol, 2018-10-11:
Works with the current master.

closed 2018-10-11 by tanriol

linrunner, 2019-03-11:
Released with TLP 1.2.