linrunner/TLP #357: After recalibration strange charging behaviour

julled, 2018-08-01:
### Symptom data
I just installed TLP for my T430 and did not change any configuration before i started recalib.
I started to recalibrate my battery. It discharged until 0% but then i remained at 0%. As it did not charge over time, i tried to plug and unplug the AC, but it did not help. Then i restarted and it charged again, but only until 67%.
Why does it only charge until 67% ? I also tried to uninstall tlp to reset that behaviour, but it did not help. I also booted into win10, but there it also only charges until 67%. Did i broke the battery somehow?

If i try to run: sudo tlp chargeonce BAT0
Error: no start charge threshold configured for BAT0.

The battery status is:
/sys/class/power_supply/BAT0/status                         = Unknown (threshold effective)

1. Does the problem occur on battery or AC or both?
On AC
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
https://gist.github.com/julled/4254cc54040752380db302235a85f4d4

### Expected behavior
Recharge the battery until stop-threshold , as long as current charge is below start-threshold
### Actual behavior
Stops charging at 67% , which is below my set start-threshold
### Steps to reproduce the problem
See above..,

linrunner, 2018-08-01:
Looks like your battery is broken. If a battery behaves like this during recalibration, it was already defective before. It had lost already 33% of it's design capacity and I guess that's the reason you tried to recalibrate it.

The issue tracker is for TLP bugs only, please ask further questions in a hardware oriented forum.

closed 2018-08-01 by linrunner

julled, 2018-08-01:
Thx linerunner for looking after the error ! :) It wasnt my intention to blame you for the issue, as it may sounded like that.
Yes, i wanted to get a better indication of the filling of my battery and i knew before from TLP that it lost some of its capacity. Before it charged without problem to 100% and had like real 2,5h left when unplugging. Now its showing 1,5h after unplugging, but i havent tested the real duration.
There is a discrepancy in the values which get shown in TLP. If i look at:
/sys/class/power_supply/BAT0/energy_full_design             =  56160 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  37680 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  25420 [mWh]
It indicates me, that there are some mWh which could be charged, but dont get charged. If the values shown are correct, it would indicate, that it not lost just 33%, but more than 50% of its original value, as energy_now is less then half the design value.
Please correct me if i missinterpret some values. thx :)

linrunner, 2018-08-01:
No objection from my part. Charging blocked at 67% (25420 / 37860) without a corresponding stop threshold suggests the battery is defective. The capacity (e.g. battery wear level) being 67 % too (37680 / 56160), seems pure coincidence to me.

MJimitater, 2019-10-24:
I have the same **Error: no start charge threshold configured for BAT0.**, eventhough I did set the thresholds and recalibrate the battery. How come?

I attached my tlp-stat here:
https://gist.github.com/MJimitater/dbc4b0e5e340b09911f50edbc2a01fdb

Im grateful for any advise

linrunner, 2019-10-24:
Exactly what the error message says. You somehow _set_ the thresholds – with `tlp setcharge` i guess. But you need to _configure_ them in **/etc/default/tlp** -->. https://linrunner.de/en/tlp/docs/tlp-configuration.html#chargethresholds

MJimitater, 2019-10-24:
Thanks @linrunner , I missed that! Works like magic now :)