linrunner/TLP #359: DEVICES_TO_ENABLE_ON_LAN_DISCONNECT doesn't enable WiFi and WWAN on detach from USB 3.1 dock with Ehternet adapter

RussianNeuroMancer, 2018-08-29:
### Symptom data

1. On AC, as I attach tablet to dock station 
2. [tlp-stat](https://gist.github.com/RussianNeuroMancer/72392b9e14464b62f6925c7e8e927f8a)

### Expected behavior

DEVICES_TO_ENABLE_ON_LAN_DISCONNECT enable listed devices when I detach tablet from USB 3.1 dock station that provide Ethernet connection.

### Actual behavior

DEVICES_TO_ENABLE_ON_LAN_DISCONNECT does not enable WiFi and WWAN when I detach tablet from dock station. 
I also can't use DEVICES_TO_ENABLE_ON_DOCK because modern USB 3.1 dock stations is detected just as USB hubs, so there is no docking/undocking event.

### Steps to reproduce the problem

Fill DEVICES_TO_DISABLE_ON_LAN_DISCONNECT and DEVICES_TO_ENABLE_ON_LAN_DISCONNECT with "wifi wwan" and then try to attach tablet or laptop to USB 3.1 dock that provide LAN connection.

I tested it with Dell Latitude 7285 and Belkin USB-C Express Dock 3.1 HD F4U093.

RussianNeuroMancer, 2018-09-03:
@linrunner 

> I don't own your hardware so I'm not able to reproduce this issue. 

I guess USB-to-Ethernet adapter would do same thing, maybe you have one?

> or at least provide a trace of the event

I enabled trace mode with `TLP_DEBUG="bat disk lock nm path pm rf run sysfs udev usb"` but there is no events in journalctl log of tlp.service on cable attach/detach. I missed some option?

linrunner, 2018-09-10:
I don't have an USB-to-Ethernet adapter but I know it works with in principle --> https://github.com/linrunner/TLP/commit/9f5f6866a3485a10a63344ea3e5fed6eb44f98d7

So must be about your particular hardware. 

Configuring TLP_DEBUG is sufficient, must be something else in your system setup. 

You didn't answer my question which of your USB devices are dock related. I suggest you unplug all USB devices from the dock and the compare tlp-stat -u docked and undocked.

RussianNeuroMancer, 2018-09-16:
> Did you use tlp-stat -T to show trace data?

I see, I looked for log in the wrong place (in tlp.service journal). 

[tlp-stat -T](https://gist.github.com/RussianNeuroMancer/30f56c3f02c8c44e0bdb099c467f094c) output with enabled debug.

> You didn't answer my question which of your USB devices are dock related. 

Sorry, I missed this. Here is tablet's own USB devices: 
Bus 002 Device 002: ID 413c:81b6 Dell Computer Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 002: ID 044e:1218 Alps Electric Co., Ltd 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Everything else you see in [tlp-stat](https://gist.github.com/RussianNeuroMancer/72392b9e14464b62f6925c7e8e927f8a) is either belong to dock station or to display USB hub that attached to dock station.

linrunner, 2018-09-19:
Thanks for the trace. tlp-rdw in fact detects the change:

     14 20:01:17 tablet1 tlp[3005]: +++ rdw_nm(enx58ef68a8892b).down

Enabling wifi fails because of a hardware radio switch or some other issue related to your hardware or kernel

    20:01:17 tablet1 tlp[3013]: device_switch(wifi, on).hard_blocked: rc=2

You can see this in the tlp-stat output too:

    +++ Wireless
    bluetooth = on
    wifi      = off (hardware)
    wwan = off (software)

wwan is switched on successfully 

    20:01:17 tablet1 tlp[3043]: invoke_nmcli(wwan, on).radio: rc=0
    20:01:17 tablet1 tlp[3045]: get_devs(wwan) = 1
    20:01:17 tablet1 tlp[3046]: device_switch(wwan, on).ok: rc=0

Maybe NetworkManager fails to detect the change?

linrunner, 2018-09-19:
ps. in the aftermath I may add 

    Bus 002 Device 002: ID 413c:81b6 Dell Computer Corp.

to tlp-rdw.rules so the dock will be detected as such.

RussianNeuroMancer, 2018-09-21:
> Enabling wifi fails because of a hardware radio switch or some other issue related to your hardware or kernel

phy0 get hardware block after automatic disabling by tlp. Manual phy0 blocking by running "rfkill block 0" result in soft block, not hard block. However, blocking dell-wifi result in phy0 hard block, that seems like can not be undone by tlp automatically (only manually or by Enabling in NetworkManager settings)

```
~$ rfkill
ID TYPE      DEVICE              SOFT      HARD
 0 wlan      phy0           unblocked unblocked
 1 wlan      dell-wifi      unblocked unblocked
 2 bluetooth dell-bluetooth unblocked unblocked
 3 wwan      dell-wwan      unblocked unblocked
 4 bluetooth hci0           unblocked unblocked
~$ rfkill block 1
~$ rfkill
ID TYPE      DEVICE              SOFT      HARD
 0 wlan      phy0           unblocked   blocked
 1 wlan      dell-wifi        blocked unblocked
 2 bluetooth dell-bluetooth unblocked unblocked
 3 wwan      dell-wwan      unblocked unblocked
 4 bluetooth hci0           unblocked unblocked
 ```

> Maybe NetworkManager fails to detect the change?

Maybe, I'll keen an eye on rfkill output and NetworkManager logs for cases when establishing WWAN connection fail.

> I may add the proper device to tlp-rdw.rules so the dock will be detected as such. Unfortunately you misunderstood my suggestion: I need the USB devices of the bare dock only, not the devices of the tablet.

Sure, here is list of "Belkin USB-C Express Dock 3.1 HD F4U093" own USB devices:

```
Bus 004 Device 004: ID 0bda:8153 Realtek Semiconductor Corp. 
Bus 004 Device 003: ID 050d:093a Belkin Components 
Bus 004 Device 002: ID 050d:093a Belkin Components 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 08bb:2912 Texas Instruments 
Bus 003 Device 003: ID 050d:093b Belkin Components 
Bus 003 Device 002: ID 050d:093b Belkin Components 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

linrunner, 2018-09-21:
> phy0 get hardware block after automatic disabling by tlp. Manual phy0 blocking by running "rfkill block 0" result in soft block, not hard block.
The hard block is the result of what rfkill and the kernel do. TLP just issues the command

    rfkill block wifi

([see](https://github.com/linrunner/TLP/blob/master/tlp-rf-func#L228)). 

I guess the second rfkill wlan devices "dell-wifi" causes this. Try to unload the dell specific module. 

TLP can't fix this, rfkill / kernel must do.

RussianNeuroMancer, 2018-09-23:
> The hard block is the result of what rfkill and the kernel do. TLP just issues the command

Yes, `rfkill block wifi` result in same state: 

```
ID TYPE      DEVICE              SOFT      HARD
 0 wlan      phy0             blocked   blocked
 1 wlan      dell-wifi        blocked unblocked
 2 bluetooth dell-bluetooth unblocked unblocked
 3 wwan      dell-wwan      unblocked unblocked
 4 bluetooth hci0           unblocked unblocked
```

However, `rfkill unblock wifi` unblock both of phy0 and dell-wifi, so phy0 is no longer hardblocked.

> Try to unload the dell specific module.

There is few such modules: dell_wmi, dell_rbtn, dell_laptop, dell_smbios, dell_wmi_descriptor, dell_smm_hwmon. How I can find out which one create additional devices in rfkill list?

> TLP can't fix this, rfkill / kernel must do.

I not sure my bug report will be accepted there, since `rfkill unblock wifi` did what it intended to do.

linrunner, 2018-09-24:
> I not sure my bug report will be accepted there, since rfkill unblock wifi did what it intended to do.

Objection: if `rfkill unblock wifi` would work reliable, then tlp-rdw could enable wifi as configured. Without a properly working rfkill tlp-rdw can't work reliable.

In fact there are two bug symptoms:

1. two rfkill instances (phy0, dell-wifi) for the same wifi device
1. rfkill block causing hard blocks (only soft blocks allowed by design)

RussianNeuroMancer, 2018-11-01:
I re-verified original issue on two another devices (Dell Venue 8 Pro 5855 and HP Elite x2 1013 G3) that doesn't have dell-wifi rfkill entry, and find that tlp switch networks reliably (so I close this bugreport). Therefore, this issue should be reported to kernel then, but unfortunately I doesn't have devices with dell-wifi rfkill entry (I had to return Dell Latitude 7285 due to hardware issues) so I fill fill kernel bugreport as soon as I get such device again.

closed 2018-11-01 by RussianNeuroMancer

linrunner, 2018-11-01:
Thanks for your feedback.