linrunner/TLP #362: Mode won't switch to ac

yaronsumel, 2018-09-04:
### Symptom data

1. Does the problem occur on battery or AC or both?
AC
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
https://gist.github.com/yaronsumel/3b3194b1ffb08f8d8425566a653a2131

### Expected behavior

when laptop is plugged in again to AC the mode stays battery

### Actual behavior

Should have switched to AC

### Steps to reproduce the problem

plug the laptop to ac

yaronsumel, 2018-09-04:
after reboot 
https://gist.github.com/yaronsumel/1ce5fe4d03d52b8de2125a5e328fa5ce

linrunner, 2018-09-10:
Maybe your charger isn't detected properly. Please show the output of

    tlp-stat --psup

with AC connected. 

Then show the relevant output of

    tlp-stat --pev

when

1. disconnecting AC
2. connecting AC

yaronsumel, 2018-09-14:
@linrunner It was reproduce-able really easy about week ago.. at this moment after updating to kernel 4.18.7 it seems fine.. will update the ticket with more info if its back. so far its ok :) thanks for tlp ..great work.

yaronsumel, 2018-10-22:
@linrunner the problem is happening again. 

battery -> ac

```bash
$ sudo tlp-stat --pev
--- TLP 1.1 --------------------------------------------

+++ Monitor power supply events -- cancel with ^C

monitor will print the received events for:
UDEV - the event which udev sends out after rule processing

UDEV  [1948.314120] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=73
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=7112000
POWER_SUPPLY_CHARGE_FULL_DESIGN=7368000
POWER_SUPPLY_CHARGE_NOW=5260000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=SMP
POWER_SUPPLY_MODEL_NAME=DELL 1P6KD5A
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER=795
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=12836000
SEQNUM=20254
SUBSYSTEM=power_supply
USEC_INITIALIZED=1948272839

UDEV  [1956.920625] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=73
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=7112000
POWER_SUPPLY_CHARGE_FULL_DESIGN=7368000
POWER_SUPPLY_CHARGE_NOW=5259000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=SMP
POWER_SUPPLY_MODEL_NAME=DELL 1P6KD5A
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER=795
POWER_SUPPLY_STATUS=Unknown
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=12229000
SEQNUM=20287
SUBSYSTEM=power_supply
USEC_INITIALIZED=1956886422

UDEV  [1959.933309] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=73
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=7112000
POWER_SUPPLY_CHARGE_FULL_DESIGN=7368000
POWER_SUPPLY_CHARGE_NOW=5259000
POWER_SUPPLY_CURRENT_NOW=1000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=SMP
POWER_SUPPLY_MODEL_NAME=DELL 1P6KD5A
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER=795
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=12221000
SEQNUM=20320
SUBSYSTEM=power_supply
USEC_INITIALIZED=1959908302

```

ac -> battery -> ac

```bash
UDEV  [2084.790358] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=75
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=7112000
POWER_SUPPLY_CHARGE_FULL_DESIGN=7368000
POWER_SUPPLY_CHARGE_NOW=5394000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=SMP
POWER_SUPPLY_MODEL_NAME=DELL 1P6KD5A
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER=795
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=12914000
SEQNUM=21089
SUBSYSTEM=power_supply
USEC_INITIALIZED=2084748861

UDEV  [2090.666023] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=75
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=7112000
POWER_SUPPLY_CHARGE_FULL_DESIGN=7368000
POWER_SUPPLY_CHARGE_NOW=5395000
POWER_SUPPLY_CURRENT_NOW=1000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=SMP
POWER_SUPPLY_MODEL_NAME=DELL 1P6KD5A
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER=795
POWER_SUPPLY_STATUS=Unknown
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=12322000
SEQNUM=21122
SUBSYSTEM=power_supply
USEC_INITIALIZED=2090641228

UDEV  [2093.714228] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=75
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=7112000
POWER_SUPPLY_CHARGE_FULL_DESIGN=7368000
POWER_SUPPLY_CHARGE_NOW=5395000
POWER_SUPPLY_CURRENT_NOW=1000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=SMP
POWER_SUPPLY_MODEL_NAME=DELL 1P6KD5A
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER=795
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=12312000
SEQNUM=21123
SUBSYSTEM=power_supply
USEC_INITIALIZED=2093687758

```

yaronsumel, 2018-10-22:
And after reboot

```bash
--- TLP 1.1 --------------------------------------------

+++ Monitor power supply events -- cancel with ^C

monitor will print the received events for:
UDEV - the event which udev sends out after rule processing

UDEV  [30.328578] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=80
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=7112000
POWER_SUPPLY_CHARGE_FULL_DESIGN=7368000
POWER_SUPPLY_CHARGE_NOW=5698000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=SMP
POWER_SUPPLY_MODEL_NAME=DELL 1P6KD5A
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER=795
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=12960000
SEQNUM=6954
SUBSYSTEM=power_supply
USEC_INITIALIZED=30279084

UDEV  [30.768434] change   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/AC (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/AC
POWER_SUPPLY_NAME=AC
POWER_SUPPLY_ONLINE=0
SEQNUM=6953
SUBSYSTEM=power_supply
USEC_INITIALIZED=30237181

UDEV  [42.130443] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=80
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=7112000
POWER_SUPPLY_CHARGE_FULL_DESIGN=7368000
POWER_SUPPLY_CHARGE_NOW=5697000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=SMP
POWER_SUPPLY_MODEL_NAME=DELL 1P6KD5A
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER=795
POWER_SUPPLY_STATUS=Unknown
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=12344000
SEQNUM=7091
SUBSYSTEM=power_supply
USEC_INITIALIZED=42114737

UDEV  [42.553632] change   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/AC (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/AC
POWER_SUPPLY_NAME=AC
POWER_SUPPLY_ONLINE=1
SEQNUM=7090
SUBSYSTEM=power_supply
USEC_INITIALIZED=42093006

UDEV  [45.002837] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=80
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=7112000
POWER_SUPPLY_CHARGE_FULL_DESIGN=7368000
POWER_SUPPLY_CHARGE_NOW=5697000
POWER_SUPPLY_CURRENT_NOW=1000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=SMP
POWER_SUPPLY_MODEL_NAME=DELL 1P6KD5A
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER=795
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=12418000
SEQNUM=7092
SUBSYSTEM=power_supply
USEC_INITIALIZED=44992987

```

yaronsumel, 2018-10-23:
@linrunner so I have been managed to reproduce that time after time now. If laptop booted-up with AC connected it will work correctly and if laptop booted-up while on battery the mode will always stays battery

linrunner, 2018-10-24:
TLP cant't work properly without AC events 

    UDEV  [42.553632] change   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/AC (power_supply)
    ACTION=change
   
Maybe a BIOS issue. Is it at the latest version?

linrunner, 2018-10-24:
Batteries are not always charging when AC is plugged in. So this is not enough to distinguish between AC and BAT safely.

yaronsumel, 2018-10-24:
IMO charged status should have AC mode as well ... How about manual fallback when unrecognized  ?

yaronsumel, 2018-10-24:
@linrunner ended up with these rules.. seems to work. anything else to add here ?
```bash
ACTION=="change", SUBSYSTEM=="power_supply", ATTR{status}=="Unknown", RUN+="/usr/sbin/tlp ac"
ACTION=="change", SUBSYSTEM=="power_supply", ATTR{status}=="Full", RUN+="/usr/sbin/tlp ac"
ACTION=="change", SUBSYSTEM=="power_supply", ATTR{status}=="Charging", RUN+="/usr/sbin/tlp ac"
ACTION=="change", SUBSYSTEM=="power_supply", ATTR{status}=="Discharging", RUN+="/usr/sbin/tlp bat"
```

linrunner, 2018-10-27:
So I guess AC is present, only the event is missing for your laptop. To be sure, please show _after booting on BAT and AC plugged afterwards_:

    tlp-stat --pev

yaronsumel, 2018-10-28:
@linrunner [booting on ac](https://github.com/linrunner/TLP/issues/362#issuecomment-431990756) [booting on battery ](https://github.com/linrunner/TLP/issues/362#issuecomment-431989151) thanks

linrunner, 2018-10-28:
Sorry, I meant

    tlp-stat --psup

yaronsumel, 2018-10-28:
booted with battery
--- TLP 1.1 --------------------------------------------

+++ Power supply diagnostic
/sys/class/power_supply/BAT0/type:Battery
/sys/class/power_supply/BAT0/present:1
/sys/class/power_supply/BAT0/device/path:\_SB_.BAT0

booted with ac
--- TLP 1.1 --------------------------------------------

+++ Power supply diagnostic
/sys/class/power_supply/AC/type:Mains
/sys/class/power_supply/AC/online:1
/sys/class/power_supply/AC/device/path:\_SB_.AC__
/sys/class/power_supply/BAT0/type:Battery
/sys/class/power_supply/BAT0/present:1
/sys/class/power_supply/BAT0/device/path:\_SB_.BAT0

rducom, 2018-11-05:
I have the same symptoms, sometimes it works, sometimes not. Same missing AC udev event when it doesn't works. (plug the battery, and an BAT event). But it "seems" resolved since I `sudo apt install acpitool`.

yaronsumel, 2018-11-06:
@rducom already installed.. but thanks :)

rducom, 2018-11-07:
Problem is back on my side... so sorry for the noise, installing acpitool didn't resolve the pb

yaronsumel, 2018-11-07:
@linrunner sure.. anything that I've missed in those rules ? and @rducom well .. just apply [these rules](https://github.com/linrunner/TLP/issues/362#issuecomment-432781427)

linrunner, 2018-11-12:
@yaronsumel : your rules are ok, but they don't cover all cases. Resume after Suspend (`tlp resume`), System startup (`tlp init start`) and a simple `tlp start` require a reliable AC detection without udev power supply events.

linrunner, 2018-11-12:
ps. I suggest you file a kernel bug about the missing events.

linrunner, 2019-01-18:
Hi @yaronsumel , 

I guess I have a proper solution now. Could you please test with the master branch? Without your custom rules of course.

linrunner, 2019-02-05:
Your feedback would still be welcome :-)

linrunner, 2019-03-05:
@yaronsumel : i'd still like to hear about your test results.

linrunner, 2019-03-05:
Beta packages [here](https://download.linrunner.de/packages/).

yaronsumel, 2019-03-05:
@linrunner seems like its working fine with the beta :)

linrunner, 2019-03-05:
Great :-)

closed 2019-03-05 by linrunner

linrunner, 2019-03-11:
Released with TLP 1.2.

yaronsumel, 2019-03-12:
@linrunner I'm on 1.2 .. seems fine .. thanks for the work