linrunner/TLP #365: Card reader consumes power even when usb autosuspend is on (plus more on pcie_aspm)

ableabhinav, 2018-09-19:
### Symptom data
I use powertop to analyse battery consumption (not --autotune). The following is a snippet:
```
The battery reports a discharge rate of 10.3 W 
The estimated remaining time is 0 hours, 47 minutes

Summary: 3406.4 wakeups/second,  0.0 GPU ops/seconds, 0.0 VFS ops/sec and 81.0% CPU use

Power est.              Usage       Events/s    Category       Description
  6.97 W    100.0%                      Device         USB device: USB2.0-CRW (Generic)
  1.49 W    277.2 ms/s     2119.9       Process        /usr/lib/firefox/firefox -contentproc -childID 4 -isForBrowser -prefsLen 14265 -schedulerPrefs 0001,2 -parentBuildID 2018091010
  729 mW      8.7 pkts/s                Device         Network interface: wlp6s0 (iwlwifi)
  524 mW    100.0%                      Device         Radio device: btusb
  370 mW    391.1 ms/s     216.2        Process        /usr/lib/firefox/firefox -contentproc -childID 7 -isForBrowser -prefsLen 14265 -schedulerPrefs 0001,2 -parentBuildID 2018091010
  330 mW     58.6 ms/s     472.7        Process        /usr/lib/firefox/firefox
  154 mW     23.5 ms/s     224.1        Process        /usr/bin/pulseaudio --start --log-target=syslog
  105 mW      3.7 ms/s     164.1        Process        powertop
 80.1 mW     41.5 ms/s      88.5        Process        /usr/lib/firefox/firefox -contentproc -childID 2 -isForBrowser -prefsLen 13626 -schedulerPrefs 0001,2 -parentBuildID 2018091010
 23.5 mW      1.4 ms/s      36.4        Interrupt      [7] sched(softirq)
 13.0 mW      2.3 ms/s      18.7        Timer          tick_sched_timer
 11.1 mW    197.8 us/s      17.7        Process        [rcu_sched]
 7.55 mW    330.0 us/s      11.8        Process        [irq/48-iwlwifi]
 4.87 mW      1.0 ms/s       6.9        Process        /opt/google/chrome/chrome /home/abh98/powerreport.html
 3.55 mW      0.8 ms/s       4.9        Process        /opt/google/chrome/chrome --type=renderer --field-trial-handle=1469706573660522489,7270713664347220981,131072 --service-pipe-to
 2.78 mW    544.6 us/s       3.9        Process        /usr/lib/firefox/firefox -contentproc -childID 3 -isForBrowser -prefsLen 14265 -schedulerPrefs 0001,2 -parentBuildID 2018091010
 2.60 mW    240.4 us/s       3.9        Interrupt      [9] acpi
 2.34 mW      0.8 ms/s       2.9        Process        /usr/bin/konsole
 2.34 mW      3.9 ms/s      0.00        Timer          hrtimer_wakeup
 1.26 mW     56.5 us/s       2.0        Process        /opt/google/chrome/chrome --type=renderer --field-trial-handle=1469706573660522489,7270713664347220981,131072 --service-pipe-to
 1.23 mW     10.9 us/s       2.0        kWork          pm_runtime_work
  767 uW    256.7 us/s       1.0        Process        /usr/sbin/thermald --no-daemon --dbus-enable
  623 uW     16.7 us/s       1.0        Process        /usr/sbin/minidlnad -f /etc/minidlna.conf -P /run/minidlna/minidlna.pid
  618 uW      9.3 us/s       1.0        kWork          flush_to_ldisc
  617 uW      7.8 us/s       1.0        Process        [ksoftirqd/2]
  617 uW      7.3 us/s       1.0        kWork          pci_pme_list_scan
  615 uW      4.6 us/s       1.0        kWork          vmstat_shepherd
  615 uW      4.1 us/s       1.0        kWork          intel_edp_drrs_downclock_work
  615 uW      3.5 us/s       1.0        Timer          sched_rt_period_timer
  614 uW      2.8 us/s       1.0        kWork          blk_timeout_work
  613 uW      0.0 us/s       1.0        kWork          mmc_rescan
  443 uW      0.7 ms/s      0.00        Interrupt      [1] timer(softirq)
  318 uW    529.5 us/s      0.00        Interrupt      [9] RCU(softirq)
  234 uW    388.6 us/s      0.00        Process        /usr/bin/plasma-browser-integration-host /usr/lib/mozilla/native-messaging-hosts/org.kde.plasma.browser_integration.json plasma
  107 uW    177.9 us/s      0.00        Timer          process_timeout
```
1) As seen from the first entry a usb device is powered on 100% of the time and consumes more power than display backlight and wireless.
2) Searching on the internet, came across a similar issue (on a Dell XPS 13) that suggests its a card reader.
3) Later I came across PCIe ASPM i.e power management of the peripherals.
4) I found out about this regression with kernel 2.6.38 `https://www.phoronix.com/scan.php?page=article&item=linux_aspm_solution&num=1`
and it was subsequently dealt with in kernel 3.2 `https://www.phoronix.com/scan.php?page=news_item&px=MTA0MTM`

Now using cat /sys/module/pcie_aspm/parameters/policy shows:
```[default] performance powersave powersupersave```

My questions are as follows:
1) Is card reader not controlled by usb autosuspend in tlp configuration? Since it shows up as a usb device. 
2) Does this show aspm is not set to the powersave policy? PCIE_ASPM_ON_BAT is set to powersave in the configurations. Furthermore, does this show the regression is still present?
3) Could I use pcie_aspm=force kernel parameter while boot-up while using tlp?

My device: Dell Inspiron 3558
[tlp-stat](https://gist.github.com/ableabhinav/ffced850be90cd9ef27cc4b9e579d491)

linrunner, 2018-09-20:
(1) autosuspend is active for the card reader as is obvious from the tlp-stat output

    Bus 002 Device 004 ID 0bda:0129 control = auto, autosuspend_delay_ms = 2000 -- Realtek Semiconductor Corp. RTS5129 Card Reader Controller (rtsx_usb)

Hint: what powertop tells you about power consumption of *single* components is complete bullshit (see the TLP FAQ)

(2,3) The bios tells the kernel not to touch ASPM

     /sys/module/pcie_aspm/parameters/policy = default (using bios preferences)

 If you insist on changing it, you must "force". See Matthew Garrets numeous blogpost on that subject.

Please don't use the issue tracker as a "support forum", it is for bug reports only. Thank you and have a nice day.

closed 2018-09-20 by linrunner

ableabhinav, 2018-09-20:
> The card reader's driver, rtsx_usb, needs to continuously wake up the device to check if there's a card in its slot. If the USB controller is also suspended, it also gets woken up. If the USB controller is suspended, it also gets woken up. If it's an Intel SoC, the SoC also leave from deeper C-State.
> 
> So the high power usage is actually three things combined. I am working on a patch to leave card reader in suspended state, which can avoid this situation. Hopefully I can finish the work an make it merged in v4.19 cycle.

I do not use the card reader, so would this stop happening if I simply unload the module? Furthermore, how about a script that allows the user to activate/deactivate the card reader? Just curious if its possible.

khfeng, 2018-09-21:
> I do not use the card reader, so would this stop happening if I simply unload the module? Furthermore, how about a script that allows the user to activate/deactivate the card reader? Just curious if its possible.

Yes, just unload rtsx_usb_ms, rtsx_usb_sdmmc and rtsx_usb.
Or wait for my patch which can keep it calm :)

ableabhinav, 2018-09-21:
Thanks for the advice and your work with the patch! Hopeful to see some battery life improvements with the patch :)

ableabhinav, 2018-11-22:
> The card reader's driver, rtsx_usb, needs to continuously wake up the device to check if there's a card in its slot. If the USB controller is also suspended, it also gets woken up. If the USB [...]

Did your patch got merged with kernel 4.19?

khfeng, 2018-11-23:
> Did your patch got merged with kernel 4.19?

No not really. Hopefully it'll be merged in 4.20.

khfeng, 2018-11-23:
FWIW this is the series:
https://patchwork.kernel.org/project/linux-usb/list/?series=38305