linrunner/TLP #367: TLP Recalibrate unexpected behaviour

web1bastler, 2018-10-01:
### Symptom data

The symptom can only occur on AC.
I am trying to recalibrate the original battery for my X230 Tablet running the latest BIOS version (2.71) and running Ubuntu 18.04 with the 4.18.11-041811-generic Kernel. The tlp version is the one supplied by apt.

tlp-stat -v -b output:

--- TLP 1.1 --------------------------------------------

+++ ThinkPad Battery Features
tp-smapi   = inactive (unsupported hardware)
tpacpi-bat = active

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 45N1079
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  62160 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  44630 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  24850 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown (threshold effective)

/sys/class/power_supply/BAT0/voltage_min_design             =  11100 [mV]
/sys/class/power_supply/BAT0/voltage_now                    =  11733 [mV]

tpacpi-bat.BAT0.startThreshold                              =     96 [%]
tpacpi-bat.BAT0.stopThreshold                               =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   55.7 [%]
Capacity                                                    =   71.8 [%]

### Expected behavior
Fullcharging the battery and force-discharging the battery once full

### Actual behavior
Fullcharge and force-discharge won't complete and tell me that they're done midway

Console output:

Setting temporary charge thresholds for BAT0:
  stop  = 100
  start = 96

Currently discharging battery BAT0:
voltage            =  11749 [mV]
remaining capacity =  24850 [mWh]
remaining percent  =     55 [%]
remaining time     = not discharging [min]
state              = Discharging
force discharge    = 0
Press Ctrl+C to cancel.

Done: battery BAT0 was completely discharged.
      The battery starts charging now. For a complete recalibration
      keep AC connected until the battery is fully charged.
(Real time elapsed: 0:07.19)

I would love to get some feedback whether I'm doing something wrong or if it's a genuine issue

linrunner, 2018-10-02:
Hi,

the battery controller – not TLP – determines when the discharge ends. So I guess your battery has a problem. I'd look for a replacement before it finally fails.

web1bastler, 2018-10-02:
Thank you very much for the Info. I planned to disassemble the original battery anyways in order to reverse engineer the protection/id board to make my own easily re-programmable version for a 12+ Cell battery. So now I don't have anything holding me back now!

closed 2018-10-02 by web1bastler