linrunner/TLP #368: ThinkPad P50 detection does not match regexp

mcsurly, 2018-10-03:
**_TL;DR_ - Somebody with a ThinkPad P50 please respond to this issue**, 
reporting the output of `cat /sys/class/dmi/id/product_version`

### Symptom data
1. Does the problem occur on battery or AC or both?
Symptom occurs during any non-manual run of tlp, such as tlp init start, or AC plugin/unplug.

### Expected behavior
My ThinkPad P50 matches the regular expression patterns that drive logic/behavior appropriate for a ThinkPad P50

### Actual behavior
check_thinkpad () function bails out on me, because this check fails:
```
        # check DMI product string for occurrence of "ThinkPad"
        if printf '%s' "$pv" | grep -q "ThinkPad"; then
```
on MY ThinkPad P50 (which did have to get its motherboard replaced, so the technician DID have to use modify NVS to manually set the serial number -- and possibly the model name!) I have the following DMI info:
/sys/class/dmi/id/product_version --> Think**p**ad P50   **_note the capitalization_**
(the same is true throughout DMI. It's in the modalias, product_family, product_sku, and uevent)

So EITHER we need a patch here (egrep -q "Think[Pp]ad") OR you can tell me that the support guys screwed me over and I have the only Think**p**ad P50 on earth, while everybody else has a Think**P**ad P50.
If that's the case, please reject this issue. I'll hack my own copy of TLP. (I also have to patch a few spots in udev rules for the same reason)

### Steps to reproduce the problem
execute `TLP_DEBUG=bat tlp init start` and see that it reports the warning `check_thinkpad.not_a_thinkpad` on a Think**p**ad P50

linrunner, 2018-10-04:
No problem to consider the deviating string in the code. Why another issue?

ps. I removed your attachment (docx) for safety reasons.

linrunner, 2018-10-05:
Just for the record: please provide a full tlp-stat output (via Gist). I don't have one for the P50 yet.

> I also have to patch a few spots in udev rules for the same reason
TLP's udev rules d not contain "ThinkPad" strings. If you have additions for a dock, e.g. concerning **tlp-rdw.rules**, please make a pull request.

Lenovo's BIOS updates do have a function to change the model code. Did you try that?

efosao, 2018-11-21:
This was from my P51. _(Not a P50)_

```sh
$ cat /sys/class/dmi/id/product_version
ThinkPad P51
```

closed 2019-01-05 by linrunner

linrunner, 2019-03-11:
Released with TLP 1.2.