linrunner/TLP #369: TLP sets wrong values for charge_stop_threshold and charge_start_threshold and displays wrong with natacpi

samcv, 2018-10-20:
### Symptom data

1. Does the problem occur on battery or AC or both?
* Both
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
* Battery: https://gist.github.com/samcv/bc5c177691b5374ebf6becfe920dbdba
* AC: https://gist.github.com/samcv/c094f557d94cb66e93caecbc9ef178ff

### Expected behavior
Expected the TLP config to set `/sys` to the same values I set them to:
```
for i in /sys/class/power_supply/BAT{0,1}/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done
/sys/class/power_supply/BAT0/charge_start_threshold = 90 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold = 95 [%]
/sys/class/power_supply/BAT1/charge_start_threshold = 90 [%]
/sys/class/power_supply/BAT1/charge_stop_threshold = 95 [%]
```
### Actual behavior
This is what they are actually set to, note how start is at 95, and stop is at 90 which that is the opposite of how the config file has it set.
```
for i in /sys/class/power_supply/BAT{0,1}/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done
/sys/class/power_supply/BAT0/charge_start_threshold = 95 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold = 90 [%]
/sys/class/power_supply/BAT1/charge_start_threshold = 95 [%]
/sys/class/power_supply/BAT1/charge_stop_threshold = 90 [%]
```
The output from tlp stat -b shows all values at 90% which is also incorrect as well, since those are not
the values which the `/sys` system shows.
```
tlp stat -b | grep _thresh
/sys/class/power_supply/BAT0/charge_start_threshold         =     90 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     90 [%]
/sys/class/power_supply/BAT1/charge_start_threshold         =     90 [%]
/sys/class/power_supply/BAT1/charge_stop_threshold          =     90 [%]
```
### Steps to reproduce the problem
For the config, do:
```
START_CHARGE_THRESH_BAT0=90
STOP_CHARGE_THRESH_BAT0=95
START_CHARGE_THRESH_BAT1=90
STOP_CHARGE_THRESH_BAT1=95
```
Restart tlp and then check the output of `tlp stat -b | grep _thresh` and the output of `for i in /sys/class/power_supply/BAT{0,1}/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done`

dywisor, 2018-10-21:
Thanks for your report.

To locate the culprit more easily, please rerun some commands with a modified config in trace mode:

1. Edit your config as follows:

```sh
START_CHARGE_THRESH_BAT0=90
STOP_CHARGE_THRESH_BAT0=95
START_CHARGE_THRESH_BAT1=80
STOP_CHARGE_THRESH_BAT1=85

TLP_DEBUG="run bat"
```

2. And then run these two commands (regardless of AC/BAT mode):

```sh

tlp start

tlp stat -b | grep _thresh
```

3. Post the logfile (``journalctl | grep -i tlp`` or so) plus the output of above tlp-stat command.

samcv, 2018-10-21:
@linrunner yeah I figured there may be something like that going on. Though also it seemed I'd better report it since TLP also shows the wrong output of `tlp stat -b` comparred to what `/sys` shows.

linrunner, 2018-10-23:
@self: don't read bug reports on smartphone ;-)

@samcv : it's strange that reading the sysfiles produces results differing from tlp-stat -b because the latter just reads the same sysfiles. I think we use the suggestion of @dywisor with some modification:

Configure trace:

     TLP_DEBUG="run bat"

Don't change thresholds in config. 

Comment tlp-stat Line 519  to

    #nodebug=1

Change thresholds to some other values

   tlp setcharge 40 60 BAT0

Check sysfiles

    for i in /sys/class/power_supply/BAT{0,1}/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done

Check with tlp-stat

    tlp-stat -b

Post all terminal outputs. Post trace by

   tlp-stat -T

samcv, 2018-10-23:
Okay, I did what you asked: https://gist.github.com/samcv/f09ccf288b78bc50233ad7e313b2bb35
Hopefully this is helpful.

dywisor, 2018-10-24:
Observations taken from the logs you've provided plus some code insights:

Writing thresholds works, but start/stop are flipped:

> ~/g/linux ❯❯❯ s tlp setcharge 40 60 BAT0
> [...]
> ~/g/linux ❯❯❯ for i in /sys/class/power_supply/BAT{0,1}/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done
> /sys/class/power_supply/BAT0/charge_start_threshold = 60 [%]
> /sys/class/power_supply/BAT0/charge_stop_threshold = 40 [%]

This happens in [set_thresholds()](https://github.com/linrunner/TLP/blob/b6dac68eb414a1f5cb38641893aa6fea8c2e4332/tlp-functions.in#L2468-L2474).
The log suggests that TLP appears to have written the thresholds to the correct files:

> Oct 23 00:48:24 sabayon tlp[99324]: +++ mode=setcharge (1.1.902+git-master-20181020) ++++++++++++++++++++++++++++++++++++++++
> [...]
> Oct 23 00:48:24 sabayon tlp[99389]: set_thresholds(BAT0, 40, 60, 1).**start**.write: old=95; **new=40**; steprc=0
> Oct 23 00:48:24 sabayon tlp[99390]: set_thresholds(BAT0, 40, 60, 1).**stop**.write: old=95; **new=60**; steprc=0

This would imply that the control file variables are flipped, but that's *probably* [not the case](https://github.com/linrunner/TLP/blob/b6dac68eb414a1f5cb38641893aa6fea8c2e4332/tlp-functions.in#L2313-L2315).

Reading, however, always returns ``charge_stop_threshold``. This occurs in [get_treshold()](https://github.com/linrunner/TLP/blob/b6dac68eb414a1f5cb38641893aa6fea8c2e4332/tlp-functions.in#L2335-L2393):

> Oct 23 00:48:26 sabayon tlp[99507]: +++ mode=auto (1.1.902+git-master-20181020) ++++++++++++++++++++++++++++++++++++++++
> [...]
> Oct 23 00:48:41 sabayon tlp[99735]: battery_present(BAT0): bm_read=natacpi; bat_str=BAT0; bat_idx=1; bd_read=/sys/class/power_supply/BAT0; **bf_start**=/sys/class/power_supply/BAT0/**charge_start_threshold**; **bf_stop**=/sys/class/power_supply/BAT0/**charge_stop_threshold**; bf_dischg=; rc=0
> Oct 23 00:48:41 sabayon tlp[99781]: get_threshold(start): bm_thresh=natacpi; bat_idx=1; thresh=**40**
> Oct 23 00:48:41 sabayon tlp[99784]: get_threshold(stop): bm_thresh=natacpi; bat_idx=1; thresh=**40**
> Oct 23 00:48:41 sabayon tlp[99785]: get_force_discharge(BAT0): bm_dischg=none; bat_idx=1; rc=255

I have yet to figure out why ;-)

- What is your ``/bin/sh`` linked to?

On a side note, [line 109 - 160](https://gist.github.com/samcv/f09ccf288b78bc50233ad7e313b2bb35#file-gistfile1-txt-L109-L160) looks suspicious - setcharge repeatedly failing due to no *thresh* method found, but succeeding later on.

samcv, 2018-10-24:
Also I compiled the acpi_call and uncommented the `NATACPI_ENABLE=1` line. If I set start to 90 and stop to 95, I get back 90 for start and 90 for stop when doing `tlp stat -b`.

linrunner, 2018-10-25:
@André : thanks for your analysis. I also haven't managed to find a bug in the code yet ... ;-)

@samcv :  do you mean you have tested with the tpacpi(-bat ) method? Could you provide a trace and output of

    setcharge 40 60 
    tlp-stat -b 

for it?

To see what the kernel really does and if read results are repeatable I need a test with sysfiles only – traces and outputs please:

    for i in /sys/class/power_supply/BAT0/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done

    echo 75 >  /sys/class/power_supply/BAT0/charge_start_threshold 

    for i in /sys/class/power_supply/BAT0/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done
    for i in /sys/class/power_supply/BAT0/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done

    echo 85 >  /sys/class/power_supply/BAT0/charge_stop_threshold 

    for i in /sys/class/power_supply/BAT0/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done
    for i in /sys/class/power_supply/BAT0/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done

samcv, 2018-10-25:
@linrunner OK. This is really weird. This is using the builtin kernel API I'm assuming (that is the only thing exposed with /sys right?)

```
sabayon /home/samantha # for i in /sys/class/power_supply/BAT0/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done
/sys/class/power_supply/BAT0/charge_start_threshold = 95 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold = 95 [%]
sabayon /home/samantha # sleeptime1=10; sleeptime2=0; echo 75 >  /sys/class/power_supply/BAT0/charge_start_threshold; sleep $sleeptime1; for i in /sys/class/power_supply/BAT0/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done; sleep $sleeptime2; for i in /sys/class/power_supply/BAT0/charge_{start,stop}_threshold; do printf "%s = %s [%%]\n" "$i" "$(cat "$i")"; done
/sys/class/power_supply/BAT0/charge_start_threshold = 75 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold = 75 [%]
/sys/class/power_supply/BAT0/charge_start_threshold = 95 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold = 75 [%]
```
I tried with varying length of sleep's. It seems very consistent, the first read shows both start and stop the same, but when it is read again, the values are different. Even if you sleep for 10 seconds after writing, you will still read out identical start and stop thresholds.

tomaszmajcherczyk, 2019-03-14:
Hey,
Any news on this one? I have the same problem it always set both thresholds to the same value.

linrunner, 2019-03-15:
@tomaszmajcherczyk : does it just show the same values or does *indeed* charging stop at the value shown? Because wrong values showing is a well known issue for L and E series already --> https://linrunner.de/en/tlp/docs/tlp-faq.html#wrong-threshold-values

tomaszmajcherczyk, 2019-03-15:
@linrunner  Actually both, btw I'm on thinkpad E480, I'm using:

`sudo tlp setcharge 70 80 BAT0`, got info:

Setting temporary charge thresholds for BAT0:
  stop  =  80
  start =  70
If I type: `sudo tlp stat` got this:

```
tpacpi-bat.BAT0.startThreshold                              =      80 [%]
tpacpi-bat.BAT0.stopThreshold                               =     80 [%]
```

And it charges to 80% as expected (still charging after, but won't go above), so the problem is "start charging value" which for some reason always set up the same as the other one.

linrunner, 2019-04-04:
@samcv : any news on this?

GHolk, 2020-04-27:
Hi, I face the same threshold of charging start and charging stop issue on thinkpad 13 2nd generation.

After I set following config and run `tlp start` in ac mode,
```
# Main / Internal battery (values in %)
START_CHARGE_THRESH_BAT0=75
STOP_CHARGE_THRESH_BAT0=81
# Ultrabay / Slice / Replaceable battery (values in %)
START_CHARGE_THRESH_BAT1=79
STOP_CHARGE_THRESH_BAT1=85
```
I got following result.
```
Linux Foundation 2.0 root hub (hub)     

+++ ThinkPad Battery Features           
tp-smapi   = inactive (unsupported hardware)                                    
tpacpi-bat = active 

+++ ThinkPad Battery Status: BAT1 (Main / Internal)                             
/sys/class/power_supply/BAT1/manufacturer                   = SMP               
/sys/class/power_supply/BAT1/model_name                     = LNV-01AV436       
/sys/class/power_supply/BAT1/cycle_count                    =    257            
/sys/class/power_supply/BAT1/energy_full_design             =  42010 [mWh]      
/sys/class/power_supply/BAT1/energy_full                    =  40210 [mWh]      
/sys/class/power_supply/BAT1/energy_now                     =  40210 [mWh]      
/sys/class/power_supply/BAT1/power_now                      =      0 [mW]       
/sys/class/power_supply/BAT1/status                         = Full              

tpacpi-bat.BAT1.startThreshold                              =     79 [%]        
tpacpi-bat.BAT1.stopThreshold                               =     79 [%]        
tpacpi-bat.BAT1.forceDischarge                              =      0            

Charge                                                      =  100.0 [%]        
Capacity                                                    =   95.7 [%]        

```

I also tried modifying `/sys` directly, and got following result:
```
/sys/class/power_supply/BAT1 $ head charge_st*                                  
==> charge_start_threshold <==          
79                  

==> charge_stop_threshold <==           
79                  
/sys/class/power_supply/BAT1 $ echo 80 | sudo tee charge_stop_threshold         
80                  
/sys/class/power_supply/BAT1 $ head charge_st*                                  
==> charge_start_threshold <==          
80                  

==> charge_stop_threshold <==           
80                  
/sys/class/power_supply/BAT1 $ echo 75 | sudo tee charge_start_threshold        
75                  
/sys/class/power_supply/BAT1 $ head charge_st*                                  
==> charge_start_threshold <==          
75                  

==> charge_stop_threshold <==           
75                  
/sys/class/power_supply/BAT1 $          
```

It seem that it is the issue from driver?

samcv, 2020-10-22:
If anyone else is still following this issue, I have come up with a patch that fixes this issue for me. It appears as if reading these sysfs values will not return what you asked for, but whatever the previous value you checked.
Example:
```bash
$ echo 50 | sudo tee /sys/class/power_supply/BAT1/charge_start_threshold
$ echo 95 | sudo tee /sys/class/power_supply/BAT1/charge_stop_threshold
$ cat /sys/class/power_supply/BAT1/charge_start_threshold
95 # wrong. but the last thing we wrote was 95 to the stop threshold...
$ cat /sys/class/power_supply/BAT1/charge_stop_threshold
50 # also wrong... but the last thing we read was the start threshold
$ cat /sys/class/power_supply/BAT1/charge_start_threshold
95 # also wrong... but the last thing we read was the stop threshold
```
Quite annoying, but reading the same file a second time gives you the correct data:
```bash
$ echo 50 | sudo tee /sys/class/power_supply/BAT1/charge_start_threshold
$ echo 95 | sudo tee /sys/class/power_supply/BAT1/charge_stop_threshold
$ cat /sys/class/power_supply/BAT1/charge_start_threshold
95 # wrong
$ cat /sys/class/power_supply/BAT1/charge_start_threshold
50 # yay this one is correct!
```
Here is a patch which fixes the problem for me. If you apply this patch, `tlp-stat -b` will give the correct information.
```bash
diff --git a/func.d/35-tlp-func-batt b/func.d/35-tlp-func-batt
index 6a020be..9ed4ded 100644
--- a/func.d/35-tlp-func-batt
+++ b/func.d/35-tlp-func-batt
@@ -350,6 +350,8 @@ read_threshold () { # read and echo charge threshold
                 stop)  bsys=$_bf_stop ;;
             esac
             # get effective threshold, "none" if not readable/non-existent
+            # read it twice due to a bug in natacpi. The first read may get the last read value from natacpi, not value we are trying to fetch
+            rt=$(read_sysf $bsys "none")
             rt=$(read_sysf $bsys "none")
             ;; # natacpi, tpsmapi
 
```

GHolk, 2020-10-23:
> If anyone else is still following this issue, I have come up with a patch that fixes this issue for me. It appears as if reading these sysfs values will not return what you asked for, but whatever the previous value you checked.

I can not reproduce your situation. No matter how many times I try, `cat /sys/class/power_supply/BAT1/charge_start_threshold` always  show same value.

linrunner, 2020-10-23:
@samcv : Great, i would not have thought of it.

1. What's the result when you read `charge_start_threshold` a 3rd time?
2. When you write `charge_stop_threshold` first and `charge_start_threshold` second, does this also apply to `charge_stop_threshold`?

samcv, 2020-10-23:
@linrunner 
1. If you read the same value three times in a row, the second and third reads will both be correct
2. If you first write to `charge_stop_threshold`, and secondly write `charge_start_threshold`, when you try and do a read from `charge_stop_threshold` you will actually get the value you just wrote to `charge_start_threshold`.

The behavior I observe can be summed up with this pseudocode here. Hope this makes sense. As you can see, writes and reads both seem to share some global register. Both BAT0 and BAT1 share the same global state. Both thresholds as well as read and writes share the same $global_register in the case of this pseudocode. So doesn't matter which battery, which threshold or if you do a read or a write. It will still affect the next read you get. 
```
var $global_register;
sub read_value($sysfs_path) {
  var $to_return = $global_register;
  $global_register = sysfs_to_hardware($sysfs_path).get_actual_value();
  return $to_return;
}
sub write_value($sysfs_path, $value) {
  $global_register = $value;
  sysfs_to_hardware($sysfs_path).write_actual_value($global_register);
}

```

samcv, 2020-10-23:
@GHolk can you try and tell me the output?
```
echo 50 | sudo tee /sys/class/power_supply/BAT1/charge_start_threshold;  echo 95 | sudo tee /sys/class/power_supply/BAT1/charge_stop_threshold; cat /sys/class/power_supply/BAT1/charge_start_threshold; cat /sys/class/power_supply/BAT1/charge_start_threshold
```
Please give me the exact output you get running that. We make sure they happen immediately after each other without any delay. But it's also possible that since we have different laptops the issue is not identical in this aspect.

linrunner, 2020-10-25:
@samcv : Change has been included. Thank you very much.

@ALL : Effectiveness on the E/L/S series still needs to be checked.

linrunner, 2020-10-25:
Leaving the issue open to see what comes out with @GHolk 's ThinkPad 13 2nd gen.

GHolk, 2020-10-25:
@linrunner sorry for late reply, the output is:
```
$ echo 50 | sudo tee /sys/class/power_supply/BAT1/charge_start_threshold
50
$ echo 95 | sudo tee /sys/class/power_supply/BAT1/charge_stop_threshold
95
$ cat /sys/class/power_supply/BAT1/charge_start_threshold
95
$ cat /sys/class/power_supply/BAT1/charge_start_threshold
95
```

samcv, 2020-10-29:
@GHolk I saw your output from `tlp-stat -b` does not mention `natacpi`. What kernel are you running? Though I guess maybe you are using a new enough kernel if you have those sysfs files in the correct places. Though even if you have an old TLP, sysfs is still responding with buggy information. I see a commit to the Linux kernel adding a battery quirk for your model here, https://github.com/torvalds/linux/commit/6640ee6289b9974334e36d8283fc07f37f3b235d So knowing the kernel version will help a lot.

linrunner, 2020-10-29:
@samcv : natacpi is basically a kernel implementation of tpacpi-bat. The acpi calls used are identical and i suppose tpapbi-bat gives the same false results. 

Btw: interesting patch, but I am not sure if it is related to @GHolk 's problem.

samcv, 2020-10-29:
@linrunner yeah, i'm not sure that it has anything to do with the problem. I think if we can repeat the same results for `BAT0` as for `BAT1` the patch is not related. Though the patch seems to imply that it's `BAT1` that should work, and before the patch `BAT0` is broken. And the ACPI call looks pretty dead simple, aside from those quirks.

The next thing I would try is just running:
```bash
echo 50 | sudo tee /sys/class/power_supply/BAT1/charge_start_threshold
echo 95 | sudo tee /sys/class/power_supply/BAT1/charge_stop_threshold
cat /sys/class/power_supply/BAT*/*
cat /sys/class/power_supply/BAT1/charge_start_threshold
```
 and see if reading a bunch of different things will somehow allow it to read the correct charge threshold. As well as trying the same as above but replacing `BAT1` with `BAT0`. If that fails both times, then only other thing I would try is updating the BIOS or something. And I don't currently have any other ideas...

GHolk, 2020-10-30:
@samcv thanks for your help, my kernel version is 4.19.0
```term
~ $ uname -a
Linux dt13 4.19.0-12-amd64 #1 SMP Debian 4.19.152-1 (2020-10-18) x86_64 GNU/Linux
```

By the way, there is only ACAD and BAT1 in my `/sys/class/power_supply/` , no BAT0 .

linrunner, 2021-04-09:
Happens on X280 too (see tpacpi-bat issue).

closed 2021-09-24 by linrunner

cdarken, 2022-12-10:
This happens  on a L14 Gen2 AMD, basically the charge_start_threshold can't be set lower than charge_stop_threshold. Or at least that's what the sysfs contents suggests.
If you try setting either one of the values, the other one is set to the same value.
Even if my current battery level was below the threshold, them being the same value means that the charge will not start.
The only way it seems to be working is using `tlp setcharge`. Then the status is changed to 'Charging' and it will reach the stop threshold.
PS: Sorry for commenting in this closed issue, it's the only one that I've found discussing this bug, just wanted to add info for others that could be searching for ideas.

linrunner, 2022-12-10:
@cdarken the/your symptom is a long standing issue of Lenovo's firmware for (among others) the L-Series. 

See [FAQ](https://linrunner.de/tlp/faq/battery.html#charge-thresholds-shown-by-tlp-stat-b-do-not-correspond-to-the-configured-ones.)

> Nonetheless the charge thresholds work as configured.

linrunner, 2022-12-11:
You may try at [Lenovo Forums](https://forums.lenovo.com/t5/Linux-Operating-Systems/ct-p/lx_en).