linrunner/TLP #375: Can't detected power source correctly

zaxoavoki, 2018-12-09:
Hello. 
When my laptop is plugged in I write `sudo tlp stat -s` in console and get such result:

```
--- TLP 1.1 --------------------------------------------

+++ System Info
System         = LENOVO Lenovo ideapad 320S-13IKB 81AK
BIOS           = 5ZCN27WW
Release        = Ubuntu 18.04.1 LTS
Kernel         = 4.15.0-42-generic #45-Ubuntu SMP Thu Nov 15 19:32:57 UTC 2018 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.15.0-42-generic root=UUID=2558c7f7-0fbb-4de3-b949-5286112f340f ro quiet splash vt.handoff=1
Init system    = systemd v237
Boot mode      = UEFI

+++ TLP Status
State          = enabled
Last run       = 12:51:14,    107 sec(s) ago
Mode           = battery
Power source   = battery
```
And we can see `Power source = battery`, why? I just tried to change to **AC mode** writting `sudo tlp ac`, but it changes only for a few minutes. 

I tried to see changes if my laptop is chargind in isn't charging via `sudo tlp-stat --pev`:
```
--- TLP 1.1 --------------------------------------------

+++ Monitor power supply events -- cancel with ^C

monitor will print the received events for:
UDEV - the event which udev sends out after rule processing

UDEV  [3415.261804] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=98
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_ENERGY_FULL=33850000
POWER_SUPPLY_ENERGY_FULL_DESIGN=36000000
POWER_SUPPLY_ENERGY_NOW=33380000
POWER_SUPPLY_MANUFACTURER=LENOVO
POWER_SUPPLY_MODEL_NAME=BASE-BAT
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_POWER_NOW=862175000
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER=12345678
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11520000
POWER_SUPPLY_VOLTAGE_NOW=13194000
SEQNUM=4070
SUBSYSTEM=power_supply
USEC_INITIALIZED=3415207932

UDEV  [3418.753118] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=98
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_ENERGY_FULL=33850000
POWER_SUPPLY_ENERGY_FULL_DESIGN=36000000
POWER_SUPPLY_ENERGY_NOW=33380000
POWER_SUPPLY_MANUFACTURER=LENOVO
POWER_SUPPLY_MODEL_NAME=BASE-BAT
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_POWER_NOW=840390000
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER=12345678
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11520000
POWER_SUPPLY_VOLTAGE_NOW=12976000
SEQNUM=4071
SUBSYSTEM=power_supply
USEC_INITIALIZED=3418700061
```
As we can see, `POWER_SUPPLY_STATUS` changes. However it can't detect power source correctly. What should I do?

linrunner, 2018-12-09:
With AC connected please show:

    tlp-stat --psup

linrunner, 2018-12-09:
There is no AC device recognized by the kernel. Does this happen when you boot on battery only or on AC too? Then it would be a duplicate of #223 . See the [FAQ](https://linrunner.de/en/tlp/docs/tlp-faq.html#acquirk).

zaxoavoki, 2018-12-09:
This happens in any way (even if I run a laptop on AC mode).

linrunner, 2018-12-09:
This can't be fixed in TLP. I'd check if a BIOS update is available. 

TLP 1.2 will modify the behaviour of `tlp ac` so the ac settings will persist. You may use this as a workaround then.

zaxoavoki, 2018-12-09:
Unfortunately. I have an issue with `tlp ac`. I mean when I write `tlp ac` it changes to AC mode for 5-6 seconds. How can I leave AC mode enabled?

linrunner, 2018-12-09:
See my above comment. You'll have to wait for TLP 1.2.

closed 2018-12-09 by linrunner

zaxoavoki, 2018-12-09:
Okay. Thank you.

zaxoavoki, 2018-12-15:
@linrunner I have one more question. The mode is constantly switched to BAT when the charge percentage of the device changes. Can I turn off this switching?

linrunner, 2019-01-18:
Hi @zaxoavoki ,

I guess I have a proper solution to detect AC now. Could you please test with the master branch?

zaxoavoki, 2019-01-19:
@linrunner, wow, yeah! It detects power source correctly now. 
But I still have a small trouble. It doesn't change mode (**AC** to **BAT** or **BAT** to **AC**) automatically. And it can't detect `Last run` field (unknown). But I think it's because I didn't install it correctly from Github repository (because I have a lot of problems with installing it manually).

linrunner, 2019-01-19:
Just build yourself a Debian package, see https://linrunner.de/en/tlp/docs/tlp-developer-documentation.html

zaxoavoki, 2019-01-19:
Where I need to move binary files after [5th step](https://linrunner.de/en/tlp/docs/tlp-developer-documentation.html)?

linrunner, 2019-01-19:
You're welcome.