linrunner/TLP #379: advanced battery features for dell devices

hyperfekt, 2019-01-09:
My understanding is that currently only ThinkPad devices have advanced battery features available, due to the existence of tp_smapi.
For Dell devices there exists the 'Dell Command Configure' which appears to provide similar functionality in regards to battery management, for example [running off AC power](https://www.dell.com/support/manuals/us/en/04/command-configure-v4.1/dcc_cli_4.1.0/-peakshiftcfg?guid=guid-9c69fbdc-fbe0-457f-a18e-4e0f350e5923&lang=en-us) and [express charging](https://www.dell.com/support/manuals/us/en/04/command-configure-v4.1/dcc_cli_4.1.0/-advbatterychargecfg?guid=guid-44a3b124-7215-47b3-aafa-5a2f0bca8c11&lang=en-us), even though the interface is rather suboptimal.
That's all the more reason to build a tool for it, but the question is if it should be this one.

hyperfekt, 2019-01-15:
I'm afraid you are mistaken, there are Linux binaries available as well: https://www.dell.com/support/article/us/en/04/sln311302/dell-command-configure?lang=en

linrunner, 2019-01-20:
I see. 

--PrimaryBattChargeCfg seems do to charge thresholds. Did you spot something like force_discharge (for recalibration)?

I didn't see source downloads and there are binary packages for Ubuntu 16.04 and RHEL 6/7 only. 

I won't call a closed source tool from inside TLP and would still prefer using the generic kernel interface mentioned above.

linrunner, 2019-03-07:
Yesterday I stumbled over smbios-utils: https://github.com/dell/libsmbios/ Any volunteers to check this out?

spockfish, 2019-04-03:
My daily machine is a Dell XPS running openSUSE with TLP. So what do you exactly need?

linrunner, 2019-04-03:
General question: are Dell battery features similar enough to the ThinkPad approach so one can integrate them into the existing TLP commands (setcharge/discharge/recalibrate):

* What are the exact commands (with binary path) to set the start and stop threshold?
* Valid ranges for the threshold values/parameters?
* What is the exact command to check Dell battery features are actually supported on a laptop?
* Is there a command to discharge a battery while AC is connected?
* How to select primary / secondary battery in the above commands on Dell laptops (if dual batteries models exist)?

spockfish, 2019-04-04:
Ok. I've got the latest smbios installed. It's shipped with openSUSE, so that's the good news. The bad news is that I get nothing out of it. For example, a 'sudo smbios-battery-ctl --battery-charge' results in this:

```
Libsmbios version : 2.4.2
smbios-battery-ctl version : 2.4.2
 
 Supported battery charging features: 
         NIL
 
 Battery charging Status: 
         NIL

```

No difference what so ever with AC plugged in or not. So I'm not sure if this libsmbios is actually bringing something to the table...

linrunner, 2019-04-05:
Dead end? Just for the record: could you post the output of

    tlp-stat -s

marmistrz, 2019-05-26:
The relevant option is described in the manual [https://www.dell.com/support/manuals/us/en/04/command-configure-v4.2/dcc_cli_4.2/-primarybattchargecfg?guid=guid-681d4efe-eed0-4d0f-b290-afdd74e81765&lang=en-us](here)
As for the last two points, I can't find anything that would allow it.

I'm also getting NILs from smbios, I'm going to report it in the libsmbios repository.

In any case, smbios-battery-ctl is pretty low-level could do the trick, but on the other hand, I'm not sure if it wouldn't conflict with the BIOS settings.

linrunner, 2019-05-28:
Just for the record:

> cctk --PrimaryBattChargeCfg=Custom:50-70

marmistrz, 2019-05-28:
On the other hand, I have second thoughts about introducing a proprietary opt-dependency to tlp. Maybe we could ask the Dell developers to expose the API via dell/smbios? (or ask if it's already possible, `smbios-token-ctl` seems pretty low-level)

While I'll probably won't be able to contribute code to this one, I can promise testing the changes related to power-management on Dell laptops.

For reference, here's the issue about `smbios-battery-ctl`: https://github.com/dell/libsmbios/issues/71

linrunner, 2019-05-29:
I'm not going to include code that calls a proprietary tool.

marmistrz, 2019-05-29:
>> On the other hand, I have second thoughts about introducing a proprietary opt-dependency to tlp.
> However, i'm not going to include code that calls a proprietary tool.
> ps. and i would definitely prefer a kernel solution that uses the natacpi framework laid out in #321.
What I meant was that `cctk` is proprietary and that I think it may be a bad idea to depend on it ;) It's great that we agree on this. :)

So we can either use libsmbios (GPL2) or natacpi. It looks like I've already solved the major obstacles, so I think I can extend `smbios-battery-ctl` to support PrimaryBatteryChargeCfg during the weekend. 

I won't be able to help with natacpi in any other way than testing. You may consider contacting the Dell developers. (I'm not affiliated with Dell in any way, I'm just using a laptop produced by the company)

marmistrz, 2019-05-31:
@linrunner let me know if you're ok with [this CLI](https://github.com/dell/libsmbios/pull/73#issuecomment-497766350)

marmistrz, 2019-06-03:
@linrunner the pull request to dell/libsmbios has been merged. You can now use it to implement `tlp fullcharge` and `tlp setcharge` on Dell laptops.

linrunner, 2019-06-12:
OK, then let's get started. 

I'll need a bit more than just the description of the CLI because the routines in [func.d/35-tlp-func-batt](https://github.com/linrunner/TLP/blob/master/func.d/35-tlp-func-batt#L29) do a lot of checks before actually writing anything - and I don't have the hardware of course ...

I need the exact commands and their corresponding outputs / return codes for the following: 

(1) Is **libsmbios** installed and is it supporting the new CLI?

OK, i'll guess we just have to check for **smbios-battery-ctl** in the PATH.

(2) Does the laptop actually support thresholds?

(3) Reading the thresholds

(4) Writing the thresholds (+ what happens with invalid param values?)

(5) Checking for batteries: are Dell laptops  BAT0 only or are there models with two batteries?

For reference: I'll call the feature **dellsm** in TLP.

awehrfritz, 2020-01-15:
> For reference: I'll call the feature **dellsm** in TLP.

@linrunner: did you start implementing this somewhere? For instance, is there a branch that contains the initial pluming for using smbios rather than one of the kernel modules?

I have a Dell Latitude laptop that supports setting the thresholds in the bios or via the `cctk` utility and I would be interested in getting this to work in TLP.

marmistrz, 2020-01-15:
Sorry, I completely forgot about it

> OK, then let's get started.
>
> I'll need a bit more than just the description of the CLI because the routines in [...]

You can probably grep `smbios-battery-ctl --help`

> (2) Does the laptop actually support thresholds?

I don't know, but I guess that `smbios-battery-ctl` would return an error

> (3) Reading the thresholds

```
smbios-battery-ctl --get-charging-cfg
```

> (4) Writing the thresholds (+ what happens with invalid param values?)

```
smbios-battery-ctl --set-charging-mode X
smbios-battery-ctl --set-custom-charge-interval low high
```
I don't know what happens if params are invalid.

> 
> (5) Checking for batteries: are Dell laptops BAT0 only or are there models with two batteries?

No idea. For all  those things I don't know about it's probably best to ask in dell/libsmbios (e.g. through github issues)

See also this fragment from `--help`.
```
  --get-charging-cfg    Get the current Primary Battery Charge Configuration
  --set-charging-mode=<MODE>
                        Set the current Primary Battery Charge Configuration.
                        Valid choices are: ['primarily_ac', 'adaptive',
                        'custom', 'standard', 'express']
  --set-custom-charge-interval=<START> <END>
                        Set the percentage bounds for custom charge. Both must
                        be integers. START must lie in the range [50, 95], END
                        must lie in the range [55, 100], END must be at least
                        (START + 5)
```

linrunner, 2020-01-15:
I have no plans to implement this myself. Reasons are:

1. It's hopeless without a Dell laptop at hand for try and error
2. I've not enough time

cheng3100, 2020-10-02:
Hi guys, I'm confused by your discussion. I'm thinking of buying a dell laptop. My question is simple. Is the latest version now available to set battery threshold of dell laptops?  I only have a thinkpad at hand. Is there anyone who has tested command like `smbios-battery-ctl --set-custom-charge-interval low high` on a dell machine?

raduburla, 2020-10-02:
Hi,

With TLP i don't think something was implemented but you can use dell tools and smbios to change thermal and charge thresholds, I created the above script that I use to change them and it works great on Dell XPS 7590

```
#!/bin/bash

if [ $# -eq 0 ]
then
 echo "\$0 charge/thermal"
 echo "\$1 value"

 exit 1
fi

if [ $1 = 'charge' ]
then
 if [ $# -eq 1 ]
 then
        echo "a value of: 
                        PrimAcUse, Custom:50-70, 
                        Adaptive, Express, Standard"
        exit 1
 fi

 read -sp "Setup Password:" password

 /opt/dell/dcc/cctk --ValSetupPwd=$password --PrimaryBattChargeCfg=$2
 exit 0
fi

if [ $1 = 'thermal' ]
then
 if [ $# -eq 1 ]
 then
        echo "a value of: 
                        balanced, cool-bottom,
                        quiet, performance"
        /usr/sbin/smbios-thermal-ctl -g
                        exit 1
 fi

 /usr/sbin/smbios-thermal-ctl --set-thermal-mode=$2
 /usr/sbin/smbios-thermal-ctl -g
 exit 0
```

/opt/dell/dcc/cctk is Dell Command Configure util developed by Dell, you can download a .deb from their website

awehrfritz, 2020-10-02:
I have only used the `cctk` utility a few times and never got to test the [libsmbios](https://github.com/dell/libsmbios) utilities properly since the version on my system is too old. Also, I don't really need it as I set the thresholds once and use them ever since without modification.

I reckon this will become all much easier once there is support for this in the kernel which is currently work-in-progress, see: https://lkml.org/lkml/2020/7/29/87. There is also an open [issue](https://github.com/dell/libsmbios/issues/84) in libsmbios which has some further information, but no progress on the libsmbios side yet since the kernel interface is still missing obviously.

cheng3100, 2020-10-02:
@raduburla Wow, this is very good. Thank you! I think I will get a dell laptop!

cheng3100, 2020-10-02:
@awehrfritz I see it now. Thank you for your link. After some research. I found that the battery threshold is closely related to battery firmware which is much more closesource compared to the host driver. Maybe that is reason why it is so hard to implement.

marmistrz, 2020-10-02:
`smbios-battery-ctl --set-custom-charge-interval low high` was implemented by me and works on my Dell laptop :P
You also need `smbios-battery-ctl --set-charging-mode custom`

But I'm using it standalone, without tlp.

cheng3100, 2020-10-07:
@marmistrz Thanks, I will try it after my new dell laptop arrive.

bllooi, 2021-03-27:
> Hi,
>
> With TLP i don't think something was implemented but you can use dell tools and smbios to change thermal and charge thresholds, I created the above script that I use to change them and it works [...]

This is actually a super cool script! Would it be possible for you to explain it?

raduburla, 2021-03-27:
Hi,

Sure, to use it save it as a .sh file then `chmod +x` that file. For the bellow examples I use the name of the sh file as dell.sh

The script supports 2 main options that you want to change:
1. `charge` - where you set charge thresholds like PrimAcUse, Custom:50-70, Adaptive, Express or Standard
For example if you want to set the laptop to Adaptive charge you run it like this:

`sudo bash dell.sh charge Adaptive`

If you have set a Bios password it will ask for the password. I did not test without a password, if it does not work try replace the following line in the script `/opt/dell/dcc/cctk --ValSetupPwd=$password --PrimaryBattChargeCfg=$2` with `/opt/dell/dcc/cctk  --PrimaryBattChargeCfg=$2` and remove `read -sp "Setup Password:" password`

2. `thermal` - to change the thermal settings of the laptop, like balanced, cool-bottom, quiet or performance
For example if you want it to performance you run it like this:

`sudo bash dell.sh thermal performance`

You will need cctk (https://www.dell.com/support/kbdoc/ro-ro/000178000/dell-command-configure) and smbios (https://github.com/dell/libsmbios) installed

If you get errors you might need to change some Bios configurations. I had to disable some security features from the Bios for the charge settings to work, I can't remember exactly what settings. Can post a list of my actual settings if needed.

Hope this help

linrunner, 2021-04-21:
Current state of things is: I would love to support Dell, but not by means of an external tool (or external kernel module). I had my fair share of experience with that in > 10 years on ThinkPads and i think 

So the prerequisite is mainline kernel support for

```
/sys/class/power_supply/BAT0/charge_control_start_threshold
/sys/class/power_supply/BAT0/charge_control_end_threshold

```
So let's see what happens here: https://lkml.org/lkml/2020/7/29/87

zalaps, 2021-12-17:
> Sorry, I completely forgot about it. Unfortunately, **smbios didn't have a new release after my changes accepted**, so you need to patch `smbios-battery-ctl` manually.

Thanks!!!

I've dell vostro originally with Win11. Changed to ubuntu 20.04 LTS with UEFI. TLP is not supported. Following commands worked:

```
sudo smbios-battery-ctl --set-charging-mode=custom
sudo smbios-battery-ctl --set-custom-charge-interval=50 55
sudo smbios-battery-ctl --get-charging-cfg 
```

tlvu, 2022-02-13:
```
$ sudo smbios-battery-ctl --get-charging-cfg
Charging mode: custom
Charging interval: (70, 80)
```
Unplug charger, then the config is lost !
```
$ sudo smbios-battery-ctl --get-charging-cfg
Charging mode: express
```

I have `smbios-battery-ctl 2.4.3`, latest Ubuntu 20.04.3 LTS and I am on a Dell Latitude E6230.

Any idea why the config did not persist?

linrunner, 2022-02-15:
@tlvu , @zalaps : Be so kind and stay on topic. The discussion about smbios-battery-ctl does not belong here. Thank you.

zalaps, 2022-02-15:
> @tlvu , @zalaps : Be so kind and stay on topic. The discussion about smbios-battery-ctl does not belong here. Thank you.

@linrunner 
Frankly, We found references to smbios-battery-ctl from this thread only. Also, there weren't any request to exclude smbios stuff earlier. 
Let me know, I would gladly delete the post if required. :)

linrunner, 2022-02-15:
@zalaps : no problem, no need to delete. I only wanted to sharpen the topic with my hint :-)

MicroJoe, 2022-05-08:
I have asked Dell an update about kernel support on https://github.com/dell/libsmbios/issues/84. FYI the employees working on the driver seem to have left the company 5 months ago, so support for a proper kernel interface might still take a while.

As of today, one has to use `smbios-battery-ctl` and not rely on TLP for setting battery wear control for Dell devices.

linrunner, 2024-10-01:
Released with 6.11 https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/platform/x86/dell?id=ab58016c68cc5b8c3622e38b7db64994e4833d9f

Anybody ready to test if I supply a plugin?

AkechiShiro, 2024-10-01:
Will see if I can test before end of week

linrunner, 2024-10-04:
@AkechiShiro I would need the output of `tlp-stat -s -b`. Thanks.

AkechiShiro, 2024-10-04:
Under Linux 6.11's kernel right ?

linrunner, 2024-10-04:
@AkechiShiro if your already run 6.11 then I'd prefer

     grep . /sys/class/power_supply/BAT?/*

Otherwise 

    tlp-stat -s -b

suffices atm.

linrunner, 2024-10-04:
@AkechiShiro You shall/need not build anything atm. Just show the output requested above, works with any recent TLP version. I just need some facts about Dell laptops. Thanks.

linrunner, 2024-10-04:
**The wait is finally over: an initial version of the Dell plugin is ready to test.**

Kernel 6.11 is required.

For the test you either need TLP 1.7.0, supplemented by the download of **[/usr/share/tlp/bat.d/65-dell](https://raw.githubusercontent.com/linrunner/TLP/refs/heads/feature/bat.d/dell/bat.d/65-dell)**, or you can build a package for your distribution based on Branch [feature/bat.d/dell](https://github.com/linrunner/TLP/tree/feature/bat.d/dell/bat.d).

I need at least the output of 

    sudo tlp-stat -s -b 

Please also run the automatic test script (with sudo or as root) and show the output:
**[unit-tests/charge-thresholds_dell](https://raw.githubusercontent.com/linrunner/TLP/refs/heads/feature/bat.d/dell/unit-tests/charge-thresholds_dell)**

It requires the tool **[clitest](https://github.com/aureliojargas/clitest)** to be installed. Should be available as a package in most distributions. 

@hyperfekt @cheng3100 @gagath @marmistrz @awehrfritz @zalaps @spockfish @raduburla @AkechiShiro

linrunner, 2024-10-06:
@AkechiShiro Is something keeping you from testing?

serycjon, 2024-10-06:
Hi, here are my outputs.

<details><summary>tlp-stat results</summary>

```
--- TLP 1.7.0 --------------------------------------------

+++ System Info
System         = Dell Inc.  Latitude 5491
BIOS           = 1.15.0
OS Release     = Arch Linux
Kernel         = 6.11.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Mon, 30 Sep 2024 23:49:50 +0000 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=d8c139f4-1622-47b9-8e7c-f927e363cc0a rw loglevel=3 quiet
Init system    = systemd 
Boot mode      = UEFI
Suspend mode   = s2idle [deep]

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 02:17:27 PM, 308 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: dell
Supported features: none available
Driver usage:
* natacpi (dell_laptop) = inactive (laptop not supported)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = DELL GD1JP65
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   8947 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   7284 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   5629 [mAh]
/sys/class/power_supply/BAT0/current_now                    =      1 [mA]
/sys/class/power_supply/BAT0/status                         = Not charging

Charge                                                      =   77.3 [%]
Capacity                                                    =   81.4 [%]

```

</details>

<details><summary>unit test results</summary>

```
#1	# +++ Dell laptops +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#2	#
#3	# --- tlp start
#4	sudo tlp start -- START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0= START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#5	sudo tlp start -- START_CHARGE_THRESH_BAT0="60" STOP_CHARGE_THRESH_BAT0="100" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#6	sudo tlp start -- START_CHARGE_THRESH_BAT0="100" STOP_CHARGE_THRESH_BAT0="100" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #6, line 16] sudo tlp start -- START_CHARGE_THRESH_BAT0="100" STOP_CHARGE_THRESH_BAT0="100" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
@@ -1,2 +1 @@
-Error in configuration at START_CHARGE_THRESH_BAT0="100": not specified, invalid or out of range (50..95). Battery skipped.
 TLP started in AC mode (auto).
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#7	sudo tlp start -- START_CHARGE_THRESH_BAT0="50" STOP_CHARGE_THRESH_BAT0="0" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #7, line 19] sudo tlp start -- START_CHARGE_THRESH_BAT0="50" STOP_CHARGE_THRESH_BAT0="0" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
@@ -1,2 +1 @@
-Error in configuration at STOP_CHARGE_THRESH_BAT0="0": not specified, invalid or out of range (55..100). Battery skipped.
 TLP started in AC mode (auto).
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#8	sudo tlp start -- START_CHARGE_THRESH_BAT0="50" STOP_CHARGE_THRESH_BAT0="101" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
[... 191 lines cut ...]
#40	#
#41	# --- tlp-stat
#42	sudo tlp-stat -b | grep -E 'charge_(control|behaviour)'
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #42, line 102] sudo tlp-stat -b | grep -E 'charge_(control|behaviour)'
@@ -1,2 +0,0 @@
-/sys/class/power_supply/BAT0/charge_control_start_threshold =     95 [%]
-/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#43	sudo tlp-stat -b -- X_THRESH_SIMULATE_READERR=1 | grep -E 'charge_(control|behaviour)'
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------ [... cut ...]
-/sys/class/power_supply/BAT0/charge_control_end_threshold   = (not available) [%]
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#44	#
#45	# --- Reset test machine to configured thresholds
#46	sudo tlp setcharge BAT0  > /dev/null 2>&1
#47	#

FAIL: 28 of 47 tests failed
```

</details>

linrunner, 2024-10-06:
@serycjon thanks for testings.

Something fundamental seems to be missing. The [sources of the driver](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/platform/x86/dell?id=ab58016c68cc5b8c3622e38b7db64994e4833d9f) say something about an attribute `charge_types` and custom charging mode. Please show:

    grep . /sys/class/power_supply/BAT0/* 2> /dev/null | grep -v uevent

serycjon, 2024-10-06:
I didn't do anything except what was mentioned in https://github.com/linrunner/TLP/issues/379#issuecomment-2394182769 
Maybe I need to first enable some driver?

Output of `grep . /sys/class/power_supply/BAT0/* 2> /dev/null | grep -v uevent`:
```
/sys/class/power_supply/BAT0/alarm:0
/sys/class/power_supply/BAT0/capacity:77
/sys/class/power_supply/BAT0/capacity_level:Normal
/sys/class/power_supply/BAT0/charge_full:7284000
/sys/class/power_supply/BAT0/charge_full_design:8947000
/sys/class/power_supply/BAT0/charge_now:5627000
/sys/class/power_supply/BAT0/current_now:1000
/sys/class/power_supply/BAT0/cycle_count:0
/sys/class/power_supply/BAT0/manufacturer:SMP
/sys/class/power_supply/BAT0/model_name:DELL GD1JP65
/sys/class/power_supply/BAT0/present:1
/sys/class/power_supply/BAT0/serial_number:3356
/sys/class/power_supply/BAT0/status:Not charging
/sys/class/power_supply/BAT0/technology:Li-poly
/sys/class/power_supply/BAT0/type:Battery
/sys/class/power_supply/BAT0/voltage_min_design:7600000
/sys/class/power_supply/BAT0/voltage_now:8206000
```

linrunner, 2024-10-06:
The necessary module _dell_laptop_ is loaded, otherwise the plugin would not be activated at all. I will email the kernel dev.

linrunner, 2024-10-06:
Oops. I think I was wrong about the kernel version in which the feature is merged. Seems to be coming with 6.12:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/platform/x86/dell?id=84bbfe6b6435658132df2880258d34babe46d3e0

Then we'll just have to wait. Unless you want to install a 6.12-rc.

AkechiShiro, 2024-10-06:
Sorry @linrunner, I just couldn't find the time to test yet

serycjon, 2024-10-07:
@linrunner I have never tweaked with kernel, so I will probably wait. The 6.12 will be probably released quite soon - maybe in a month or so? Judging only from the historical data (https://en.wikipedia.org/wiki/Linux_kernel_version_history#Releases_6.x.y).

langemar, 2024-10-27:
Hi,
as fedora rawhide provides a suitable kernel for testing I gave it a shot. Please let me know if I can provide further information.

best regrads
markus

<details>

<summary># tlp-stat -s -b</summary>

```
root@localhost-live:~/TLP-1.7.0# tlp-stat -s -b
--- TLP 1.7.0 --------------------------------------------

+++ System Info
System         = Dell Inc.  Latitude 3450
BIOS           = 1.8.0
EC Firmware    = 1.5
OS Release     = Fedora Linux 42 (Workstation Edition Prerelease)
Kernel         = 6.12.0-0.rc4.20241025gitae90f6a6170d.42.fc42.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Oct 25 15:16:46 UTC 2024 x86_64
/proc/cmdline  = BOOT_IMAGE=/images/pxeboot/vmlinuz root=live:CDLABEL=Fedora-WS-Live-rawh-20241027-n-0 rd.live.image rd.live.check quiet
Init system    = systemd 
Boot mode      = UEFI
Suspend mode   = [s2idle]
SELinux status = enabled (enforcing)

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = unknown
Mode           = unknown
Power source   = AC

Error: TLP's power saving will not apply on boot because tlp.service is not enabled --> Invoke 'systemctl enable tlp.service' to ensure the full functionality of TLP.

+++ Battery Care
Plugin: dell
Supported features: charge thresholds
Driver usage:
* natacpi (dell_laptop) = active (charge thresholds)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  50..95(default)
* STOP_CHARGE_THRESH_BAT0/1:   55..100(default)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = BYD
/sys/class/power_supply/BAT0/model_name                     = DELL 7RDGP45O
/sys/class/power_supply/BAT0/cycle_count                    =      2
/sys/class/power_supply/BAT0/charge_full_design             =   3685 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   3685 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   2926 [mAh]
/sys/class/power_supply/BAT0/current_now                    =      1 [mA]
/sys/class/power_supply/BAT0/status                         = Not charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     60 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     80 [%]
/sys/class/power_supply/BAT0/charge_types                   = Trickle Fast Standard Adaptive [Custom] [%]

Charge                                                      =   79.4 [%]
Capacity                                                    =  100.0 [%]
```

</details>

<details>

<summary># grep . /sys/class/power_supply/BAT0/* 2> /dev/null | grep -v ueven</summary>

```
root@localhost-live:~/TLP-1.7.0# grep . /sys/class/power_supply/BAT0/* 2> /dev/null | grep -v ueven
[... 157 lines cut ...]
#33	sudo tlp setcharge 90 95 -- X_BAT_PLUGIN_SIMULATE=dell
#34	sudo tlp setcharge 90 95 -- X_THRESH_SIMULATE_READERR="1"
#35	sudo tlp setcharge 90 95 -- X_BAT_PLUGIN_SIMULATE=dell
#36	sudo tlp setcharge DEF DEF -- X_BAT_PLUGIN_SIMULATE=dell
#37	sudo tlp setcharge BAT1 -- X_BAT_PLUGIN_SIMULATE=dell
#38	sudo tlp setcharge 0 3 BAT1 -- X_BAT_PLUGIN_SIMULATE=dell
#39	sudo tlp setcharge XYZZY ABCDE BAT1 -- X_BAT_PLUGIN_SIMULATE=dell
#40	#
#41	# --- tlp-stat
#42	sudo tlp-stat -b | grep -E 'charge_(control|behaviour)'
#43	sudo tlp-stat -b -- X_THRESH_SIMULATE_READERR=1 | grep -E 'charge_(control|behaviour)'
#44	#
#45	# --- Reset test machine to configured thresholds
#46	sudo tlp setcharge BAT0  > /dev/null 2>&1
#47	#
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #47, line 111] #
@@ -1 +0,0 @@
-
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

FAIL: 10 of 47 tests failed
```

</details>

linrunner, 2024-10-27:
@langemar looks promising. Would ypu please enable the service

    sudo systemctl enable tlp.service

the run the automatic test again

    unit-tests/charge-thresholds_dell

langemar, 2024-10-27:
Of course:

<details>

<summary># unit-tests/charge-thresholds_dell</summary>

```
root@localhost-live:~/TLP-1.7.0# unit-tests/charge-thresholds_dell
#1	# +++ Dell laptops +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#2	#
#3	# --- tlp start
#4	sudo tlp start -- START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0= START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#5	sudo tlp start -- START_CHARGE_THRESH_BAT0="60" STOP_CHARGE_THRESH_BAT0="100" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#6	sudo tlp start -- START_CHARGE_THRESH_BAT0="100" STOP_CHARGE_THRESH_BAT0="100" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#7	sudo tlp start -- START_CHARGE_THRESH_BAT0="50" STOP_CHARGE_THRESH_BAT0="0" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#8	sudo tlp start -- START_CHARGE_THRESH_BAT0="50" STOP_CHARGE_THRESH_BAT0="101" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#9	sudo tlp start -- START_CHARGE_THRESH_BAT0="90" STOP_CHARGE_THRESH_BAT0="91" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#10	sudo tlp start -- START_CHARGE_THRESH_BAT0="90" STOP_CHARGE_THRESH_BAT0="95" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#11	sudo tlp start -- START_CHARGE_THRESH_BAT0="DEF" STOP_CHARGE_THRESH_BAT0="DEF" START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1=
#12	sudo tlp start -- NATACPI_ENABLE=0 START_CHARGE_THRESH_BAT0="DEF" STOP_CHARGE_THRESH_BAT0="DEF"
#13	#
#14	# --- tlp setcharge w/o arguments
#15	sudo tlp setcharge -- START_CHARGE_THRESH_BAT0="60" STOP_CHARGE_THRESH_BAT0="100" X_SOC_CHECK=0
#16	sudo tlp setcharge -- START_CHARGE_THRESH_BAT0="100" STOP_CHARGE_THRESH_BAT0="100"
#17	sudo tlp setcharge -- START_CHARGE_THRESH_BAT0="50" STOP_CHARGE_THRESH_BAT0="0"
#18	sudo tlp setcharge -- START_CHARGE_THRESH_BAT0="50" STOP_CHARGE_THRESH_BAT0="101"
#19	sudo tlp setcharge -- START_CHARGE_THRESH_BAT0="90" STOP_CHARGE_THRESH_BAT0="91"
#20	sudo tlp setcharge -- START_CHARGE_THRESH_BAT0="90" STOP_CHARGE_THRESH_BAT0="95"
#21	sudo tlp setcharge -- START_CHARGE_THRESH_BAT0="90" STOP_CHARGE_THRESH_BAT0="95"
#22	sudo tlp setcharge -- START_CHARGE_THRESH_BAT0="DEF" STOP_CHARGE_THRESH_BAT0="DEF"
#23	sudo tlp setcharge -- NATACPI_ENABLE=0 START_CHARGE_THRESH_BAT0="DEF" STOP_CHARGE_THRESH_BAT0="DEF"
#24	#
#25	# --- tlp setcharge w/ arguments
#26	sudo tlp setcharge 60 100 -- X_SOC_CHECK=0
#27	sudo tlp setcharge 100 100 -- X_BAT_PLUGIN_SIMULATE=dell
#28	sudo tlp setcharge 50 0 -- X_BAT_PLUGIN_SIMULATE=dell
#29	sudo tlp setcharge 50 101 -- X_BAT_PLUGIN_SIMULATE=dell
#30	sudo tlp setcharge XYZZY 0 -- X_BAT_PLUGIN_SIMULATE=dell
#31	sudo tlp setcharge 50 XYZZY -- X_BAT_PLUGIN_SIMULATE=dell
#32	sudo tlp setcharge 90 91 -- X_BAT_PLUGIN_SIMULATE=dell
#33	sudo tlp setcharge 90 95 -- X_BAT_PLUGIN_SIMULATE=dell
#34	sudo tlp setcharge 90 95 -- X_THRESH_SIMULATE_READERR="1"
#35	sudo tlp setcharge 90 95 -- X_BAT_PLUGIN_SIMULATE=dell
#36	sudo tlp setcharge DEF DEF -- X_BAT_PLUGIN_SIMULATE=dell
#37	sudo tlp setcharge BAT1 -- X_BAT_PLUGIN_SIMULATE=dell
#38	sudo tlp setcharge 0 3 BAT1 -- X_BAT_PLUGIN_SIMULATE=dell
#39	sudo tlp setcharge XYZZY ABCDE BAT1 -- X_BAT_PLUGIN_SIMULATE=dell
#40	#
#41	# --- tlp-stat
#42	sudo tlp-stat -b | grep -E 'charge_(control|behaviour)'
#43	sudo tlp-stat -b -- X_THRESH_SIMULATE_READERR=1 | grep -E 'charge_(control|behaviour)'
#44	#
#45	# --- Reset test machine to configured thresholds
#46	sudo tlp setcharge BAT0  > /dev/null 2>&1
#47	#
OK: 47 of 47 tests passed
```

</details>

linrunner, 2024-10-27:
@langemar Looks perfect now. Thank your very much for testing.

Finally, you could check whether the thresholds work as set.

ps. The service did not have to be started. It was just a matter of activating it so that the warning message does not interfere with the target/actual comparison of the output. Of course, it must also be activated for regular operation of TLP.

langemar, 2024-10-27:
I made a quick test using `tlp set charge x y BAT0` and `smbios-battery-ctl - - get-charging-cfg`. Once I set a value smbios shows the set values.

The values shown by smbios are followed, I'm using it for some weeks now. Therefore I expect tlp is working now too. 

Sorry I already left office so I can't provide a log until tomorrow.

Let me know if I can provide further information. And thank you very much for your great work :) 

Am 27. Oktober 2024 16:42:49 UTC schrieb linrunner ***@***.***>:
> @langemar Looks perfect now. Thank your very much for testing.
>
> Finally, you could check whether the thresholds work as set.
>
> ps. The service did not have to be started. It was just a matter of [...]

linrunner, 2024-10-27:
> Once I set a value smbios shows the set values.

@langemar Please don't bother any further, that information is enough for me.

langemar, 2024-11-05:
> Oddly enough, my Dell system is broken after the latest release, not sure if it's related.

Given that there was no release since any change was done to the master branch concerning this issue it might be possible that the rising sun this morning could be the reason your laptop is broken. (Sorry for beeing sarcastic here, but I need to continue...)

But it might help if you would give some details in what way your laptop is broken and if you assume tlp to be the reason you might add information on the relevant versions you are using. But given what I have said already you might open another issue, or even better, try to find some cause of your problem first and provide information for remote debugging your problem. Such information might atleast include the tlp version in use, the ´tlp-stat -s´ output, your kernel version (and distribution in general) and an description of your exact problem / symptoms.

best regards

linrunner, 2024-11-06:
@vkhodygo Topic here is setting the charge thresholds for Dell laptops. This requires test results as described in https://github.com/linrunner/TLP/issues/379#issuecomment-2394182769. Your post does not contain any test results. 

How you broke your system is offtopic here.

linrunner, 2024-11-24:
Linux 6.12 is out.

@serycjon it's available in the Arch repos right now. So nothing is stopping you from testing :-).

@AkechiShiro What about NixOS?

AkechiShiro, 2024-11-24:
@linrunner 6.12 is available in NixOS as well (in unstable), I'm just waiting for it to land in a new stable release (was planned the 22 November but currently delayed due to some regression in curl)

okarin69, 2024-11-26:
@linrunner with the arrival linux kernel 6.12, in my asus laptop the /sys/class/power_supply/BAT_0/charge_control_end_threshold file has disappered and charging threshold does not work. But when I switch to linux-lts kernel it reappers and the charging threshold works properly. I hope a fix will be available the overall issue instead of just Dell laptops one.

[17 more replies]

closed 2025-02-13 by linrunner