linrunner/TLP #380: Release new verion

bogdanr, 2019-01-14:
It would be nice to tag TLP 1.2 because distros often know when to update a piece of software based on the version.

linrunner, 2019-01-15:
It's not ready yet. Thanks for your patience.

bubbleguuum, 2019-02-06:
Anything particular that is blocking to release 1.2 ? It's been one year 1.1 has been released now and distros will no pick anything else than an official release.

linrunner, 2019-02-28:
Good morning guys, 

you've asked, so it's about time for you to help finish the release.

Fresh beta packages are available [here](https://download.linrunner.de/packages/) (or use the master branch). Please refer to the [changelog](https://github.com/linrunner/TLP/blob/master/changelog) und [dev docs](https://linrunner.de/en/tlp/docs/tlp-developer-documentation.html) too.

I'm expecting your outputs of

    sudo tlp-stat

on **battery and AC** (via https://gist.github.com/ please). Go ahead!

bubbleguuum, 2019-02-28:
Thanks, will test and report.

linrunner, 2019-02-28:
You're welcome to recruit more testers from your environment.

bubbleguuum, 2019-02-28:
Here's some very initial results testing on my ThinkPad P72 running openSUSE Tumbleweed, kernel 4.20.10, using stock tlp config, both tlp.service and tlp-sleep.service started.

TLP installed from master using make install with proper parameter for openSUSE (looked at current RPM package .spec to make sure I had these correct).

- 'tlp-rdw on|off' works 
- 'tlp setcharge' (natacpi) works. Battery was 77% and range was 75-80. Changing it to
80-90 made the battery start charging
- 'tlp-stat -d' lists NVMe drive as expected
- suspend-to-RAM works 100% fine (it did previously)

tlp-stat output on AC: https://gist.github.com/bubbleguuum/c8f9198a834dc6a8d07a63b8bbadf3d8
tlp-stat output on BAT: https://gist.github.com/bubbleguuum/206c363b308dfccad83005d4c3edb9f7

linrunner, 2019-02-28:
@bubbleguuum :  great, thanks. If you have the possibility to provide RPMs, I would be happy to make them available.

bubbleguuum, 2019-02-28:
I will see but no promise as I'm not super familiar with rpm building and it's not entirely trivial.

What about making a post on /r/archlinux for a call to beta testers ? I'm certain a lot of folks there would be happy to test!

linrunner, 2019-02-28:
No objection, but I don't have the technical expertise nor the time do support in many forums. You're aware of someone who could handle this for /r/archlinux?

bubbleguuum, 2019-02-28:
I can (with your permission) make a post on /r/archlinux about the beta test, directing users to this issue.

linrunner, 2019-02-28:
OK, let's try that.

njoyard, 2019-02-28:
Test results on Arch on a Thinkpad x270, with possibly some config remaining from 1.1 (gist contains 2 files, output from AC and from BAT): https://gist.github.com/njoyard/622fddca13d417e62865a2db740cec21

linrunner, 2019-03-01:
[Thread on Reddit /r/archlinux](https://www.reddit.com/r/archlinux/comments/avtsvw/tlp_12_linux_advanced_power_management_is_in_beta/)

pablostebler, 2019-03-01:
Results on my Latitude E7470 on Arch Linux: [AC](https://gist.github.com/pablostebler/3deae32c33fba2ab2ddde1ca57c6a846), [battery](https://gist.github.com/pablostebler/95e60cd6cb4c426b80adb4cc1a4d1643).

zauster, 2019-03-01:
> Fresh beta packages are available [here](https://download.linrunner.de/packages/) (or use the master branch). Please refer to the [changelog](https://github.com/linrunner/TLP/blob/master/changelog) und [dev docs](https://linrunner.de/en/tlp/docs/tlp-developer-documentation.html) too.
> 

The Arch AUR packages seem to be a few years out-of-date, last updated in 2016...

linrunner, 2019-03-01:
@zauster: right, but: the referenced AUR packages install the master branch.

zauster, 2019-03-01:
Yeah, I realized when reading the PKGBUILDs... 
Here are my results: [AC](https://gist.github.com/zauster/a09f6900b9b029e0c6b6891a36a51d3e), [Battery](https://gist.github.com/zauster/50a9e7bfea98b115d707a007d6ce5234)

linrunner, 2019-03-01:
Thanks @all for your feedback.

@zauster : enabling tlp.service and tlp-sleep.service is [mandatory](https://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#arch). Really.

KJoke70, 2019-03-01:
[AC](https://gist.github.com/KJoke70/5d52a9f584542e3f1828793f417eeee0#file-ac), [Battery](https://gist.github.com/KJoke70/5d52a9f584542e3f1828793f417eeee0#file-bat).

linrunner, 2019-03-02:
Fedora packages are [available](https://download.linrunner.de/packages/) now.

ykgmfq, 2019-03-05:
Beta 4:
```
+++ Recommendations
* Install acpi-call kernel module for ThinkPad battery features\n
```
Is this recommendation useful when natacpi is active? Also the printed linebreak
```
> tlp stat
Error: 'tlp stat' no longer supported, use 'tlp-stat' instead.
> tlp
Usage: tlp start|true|bat|false|ac|usb|bayoff|discharge|setcharge|fullcharge|recalibrate|stat|diskid
```
Stat should not be listed if deprecated

linrunner, 2019-03-05:
@ykgmfq :
tpacpi-bat/acpi_call is still needed for forced discharge, natacpi implements only thresholds so far.

> Stat should not be listed if deprecated
Fixed. Thanks.

ykgmfq, 2019-03-05:
https://gist.github.com/ykgmfq/61441199245f8e1fc00f684a6b817883
https://gist.github.com/ykgmfq/ec85f659106541b418cd69aee94785d3

> tpacpi-bat/acpi_call is still needed for forced discharge, natacpi implements only thresholds so far.

Would love to see that in the recommendation. I think quite a lot of people are fine with only the thresholds – like me – and would be spared messing with extra kernel modules.

Off-topic: If the battery config were in a seperate file, say `/etc/default/tlp-bat`, the package manager could give me the new `/etc/default/tlp` without conflict resolution since the battery thesholds are the only things I change.

linrunner, 2019-03-05:
@ykgmfq :
> Would love to see that in the recommendation.

Makes sense. Done.

The single config file is a fundamental feature of TLP. You may always keep your file:

* TLP is always kept backward compatible with older configs
* apt/dpkg prompt you about it
* see [FAQ](https://linrunner.de/en/tlp/docs/tlp-faq.html)

linrunner, 2019-03-06:
**@all** : uploaded version 1.1.909. 

Bugfixes:
* Wrong charge thresholds for BAT0 fixed

Please retest.

zauster, 2019-03-08:
[AC](https://gist.github.com/zauster/fc4635fc9efa7add34a4272f0aa5d19e), [Battery](https://gist.github.com/zauster/ffb30c967fd12909adbe40621cd03537)

KJoke70, 2019-03-08:
[AC](https://gist.github.com/KJoke70/97d9ec78443ed3e658f4e66d7d3087df#file-ac), [Battery](https://gist.github.com/KJoke70/97d9ec78443ed3e658f4e66d7d3087df#file-bat). 

Since there's support for nvme drives now, is [btrfs+tlp](https://github.com/linrunner/TLP/issues/128) safe to use on nvme drives?

linrunner, 2019-03-08:
@KJoke70 : issue #84 says it's some Crucial SSD exposing the problems. It's not clear to me if it is solved for that hardware..

KJoke70, 2019-03-08:
> ps. Issue #84 is ALPM-related and NVMe disks don't use SATA links and ALPM.

So I *shouldn't* encounter the same problem but I guess I'll find out. Either way, thanks!

linrunner, 2019-03-11:
Thanks for your help guys, very much appreciated.

TLP 1.2 was released this morning.

closed 2019-03-11 by linrunner

linrunner, 2019-05-01:
Hi @all: you're welcome to [test 1.2.2 beta](https://github.com/linrunner/TLP/issues/408) too.

linrunner, 2020-01-08:
1.3 Beta phase has started. Please help: https://github.com/linrunner/TLP/issues/449

linrunner, 2021-09-10:
1.4 Beta 2 is [out](https://linrunner.de/tlp/news.html#news-top-1). Enjoy!

linrunner, 2021-12-21:
TLP 1.5 Beta 1 is out. Testers welcome -> https://github.com/linrunner/TLP/issues/609

linrunner, 2023-07-22:
Hi @ALL : TLP 1.6 Beta 1 is out. Testers wanted -> https://github.com/linrunner/TLP/issues/700